ansys/pyaedt #3834: Bug located in connect_to_component in connecting to port set to microwave port.

cmesibov, 2023-11-01:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When scripting for connecting a Circuit microwave port and using the connect_to_component() pyaedt command with use_wire=False, the resultant pageport is not attached to the port pin. 

### Steps To Reproduce

portObj = circuit.modeler.components.create_interface_port(connectorPinNetName, PortLocation, angle=PortSidePortAngle)
Change portObj to type 1 port.
portObj.pins[0].connect_to_component(cable.pins[portSideIdx]) #or whatever you are connecting to.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

(.venv) PS D:\Automation\RickAndMorty\EMI_automation> python -m pip freeze
attrs==23.1.0
certifi==2023.7.22
cffi==1.16.0
charset-normalizer==3.3.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.1
cycler==0.12.0
et-xmlfile==1.1.0
exceptiongroup==1.1.3
Fiona==1.9.4.post1
fonttools==4.43.0
geopandas==0.14.0
idna==3.4
imageio==2.31.5
iniconfig==2.0.0
kiwisolver==1.4.5
matplotlib==3.8.0
networkx==3.1
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.6.0
packaging==23.2
pandas==2.1.1
Pillow==10.0.1
platformdirs==3.11.0
pluggy==1.3.0
scooby==0.7.4
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
tomli==2.0.1
tzdata==2023.3
urllib3==2.0.6
utm==0.7.0
vtk==9.2.6

Samuelopez-ansys, 2023-11-02:
Hi @cmesibov ,

This is a limitation of AEDT API. When you modify the port symbol, the size of the port changes, the problem is that there is not an AEDT method (or we did not find it) to extract the port pin information. We need the new location of the pin. 

Please, if you find this method let us know, if not, you can submit an internal enhancement request to Ansys development.

I have tested without changing the symbol, and it worked.

Thanks.

cmesibov, 2023-11-02:
Hi @Samuelopez-ansys ,
Yes, I agree. Currently, I am moving the port 2 boxes to the left to align the page port.  If the pin location does not update (given that it exists) after the type changes then its a AEDT API bug.  I'll take a look into it.

Samuelopez-ansys, 2023-11-02:
@cmesibov It is not a bug because the port in the UI changes correctly, the problem is that the pin information of the port can not be retrieved from any pin AEDT API command , they only work for other component type. 

Hopefully, you could find an AEDT API method to get the pin port location, then we will be able to add it to pyaedt.

Samuelopez-ansys, 2023-11-07:
@cmesibov I will close this issue. Please if you find a way to get the pin position of a port let us know, if not please submit a defect in Ansys internal system.

closed 2023-11-07 by Samuelopez-ansys