ansys/pyaedt #3837: EDB API: Creating PEC ports

askoeho, 2023-11-02:
### Description of the feature

EDB API is missing the feature how to make a PEC port. Layout 3d supports these kind of ports as in the following screenshot:

[image]

svandenb-dev, 2023-12-04:
Hi  @askoeho,
we will add this feature, thanks for your feedback.

svandenb-dev, 2023-12-05:
@askoeho this was added please grab the latest version when it is available.
you can create PEC boundaries from 2 command now

edbapp.components.create_port_on_pins(refdes="U1", pins="AU38", reference_pins="AU37", pec_boundary=True)
edbapp.components.deactivate_rlc_component(component="C5", create_circuit_port=True, pec_boundary=True)
edbapp.components.add_port_on_rlc_component(component="C65", circuit_ports=False, pec_boundary=True)

please note that the first command only accept 2 pins (one for positive and one for reference) as PEC boundary is not supported with Pingroup.

first command
[image]

2 last command  
[image]

closed 2023-12-05 by svandenb-dev

askoeho, 2023-12-11:
What if I need to use PEC ports in vertical orientation? I'm currently doing this with the legacy API with the following code.

```
oEditor = h3d.oeditor
oEditor.CreatePortsOnComponentsByNet(
	[
		"NAME:Components", 
		"U1"
	], 
	[
		"NAME:Nets", 
		"GND"
	], "PEC", "0", "0", "0")
```