ansys/pyaedt #3838: EDB API: deactivate_rlc_component doesn't make a proper gap port.

askoeho, 2023-11-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

As far as I have understood correctly, the following line of code should make a gap port for a RLC component. But for some reason that is not working.

`edb.components.deactivate_rlc_component(component=smd, create_circuit_port=False)`

### Steps To Reproduce

`edb.components.deactivate_rlc_component(component=smd, create_circuit_port=False)`

-> No gap ports are generated.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

accelerate==0.23.0
adodbapi==2.6.1.3
affine==2.3.1
aiofiles==22.1.0
aiohttp==3.8.3
aiosignal==1.2.0
aiosqlite==0.17.0
alabaster==0.7.12
alembic==1.11.1
algopy==0.5.7
altair==5.1.1
altair-data-server==0.4.1
altair-transform==0.2.0
aniso8601==9.0.1
annotated-types==0.5.0
ansi2html==1.6.0
ansiwrap==0.8.4
ansys-pythonnet==3.1.0rc3
anyio==3.7.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.2
arviz==0.15.1
asciitree==0.3.3
asgi-csrf==0.9
asgiref==3.5.2
asn1crypto==1.4.0
asteval==0.9.27
astroid==2.12.12
astropy==5.3.2
asttokens==2.1.0
async-lru==2.0.4
async-timeout==4.0.3
atomicwrites==1.4.0
attrs==22.1.0
autopep8==1.7.0
azure-core==1.29.2
azure-cosmos==4.5.0
azure-identity==1.14.0
Babel==2.10.3
backcall==0.2.0
baresql==0.8.0
[... 551 lines cut ...]
webcolors==1.12
webencodings==0.5.1
websocket-client==1.6.1
websockets==10.3
Werkzeug==2.2.2
whatthepatch==1.0.2
widgetsnbextension==4.0.9
winpython==7.0.20230923
wordcloud==1.8.1
wrapt==1.14.1
wsproto==1.2.0
xarray==2023.8.0
xarray-einstats==0.5.1
xgboost==1.6.1
XlsxWriter==3.1.2
xlwings==0.24.7
xmltodict==0.13.0
xyzservices==2022.6.0
yapf==0.40.1
yarl==1.7.2
yt-dlp==2023.7.6
zarr==2.11.3
zict==2.2.0
zipp==3.9.0
zstandard==0.21.0

ring630, 2023-11-06:
@svandenb-dev can you have a look at this issue?

svandenb-dev, 2023-11-10:
@askoeho Yes this seems to be a bug. However I think setting create_circuit_port=True should work ? 
This is a recommended setting to prevent shorts in case the port intersect metal. Unless you want to export to HFSS 3D modeler.
Will have a look. Thanks for your feedback

askoeho, 2023-11-10:
create_circuit_port=True works as intended, circuit ports are created. But these are not 1:1 with gap ports and at the same time I would like to have the export to HFSS 3D modeler working.

svandenb-dev, 2023-11-10:
@askoeho bug is fixed please grab the latest release when it's available.

closed 2023-11-10 by svandenb-dev