ansys/pyaedt #3846: Bug located in EDB primities_data.py @layer_name.setter

pyAnsysUser, 2023-11-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The layer name setter seems to have a layer called "TOP" hard coded in lines 126 - 131.

However, it is not clear (to me) that setting the layer_name property on a primitive would achieve the desired effect of changing the layer in the database.

### Steps To Reproduce

set layer_name on EDB primitive

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
cffi==1.16.0     
clr-loader==0.2.6
numpy==1.26.1
plumbum==1.8.2
psutil==5.9.6
pyaedt==0.7.3
pycparser==2.21
pywin32==306
rpyc==5.3.1

pyAnsysUser, 2023-11-04:
I expect there is a way to link a pull request?

SMoraisAnsys, 2023-11-04:
Hi, thanks for the bug report. I'll let the assessment of your PR to @svandenb-dev  and @ring630. 
If you want to link this issue to your PR you can, for example, add "Closes #3847" in your PR description.
So, if the PR is accepted, this issue will be closed automatically (in addition to being linked).

ring630, 2023-11-06:
@pyAnsysUser , if you look at the right side of this page. 
1, Click the gear symbol. 
2, select ansys/pyaedt
3, select your PR

[image]

closed 2023-11-06 by Samuelopez-ansys

ring630, 2023-11-06:
@pyAnsysUser I am not sure why you don't see the gear icon. I linked this issue to the PR. Thans for your contribution.