ansys/pyaedt #3864: Bug located in running TwinBuilder analysis via pyaedt.application.analysis.analyze_setup

steve-biggs-fox, 2023-11-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

If I use PyAEDT to open Ansys TwinBuilder, load a model and run the analysis, this works from an interactive session. If I put the same commands in a script and run the script from the command line, I would expect this to work as well. However, the script is able to open TwinBuilder, open the model, and tries to start the analysis, but then I get a modal error popup that says "SimplorerSolver: Encountered an improper argument".

I get the same error for my own model and one of the Ansys example models, which suggests it's not a problem with the model.

I have contacted Ansys support and they have provided a script that does the same job but using the Ansys ScriptEnv. This works when run from the command line. This suggests there is nothing inherently wrong with the underlying analyze function that is part of the ScriptEnv.

I also made a hybrid version where I use PyAEDT to open TwinBuilder and load the model, then attach ScriptEnv to the same session and run the analysis using the ScriptEnv. That also works from the command line. This suggests that there is nothing wrong with the way PyAEDT opens TwinBuilder or loads the model. Therefore, the problem occurs somewhere within the pyaedt.application.analysis.analyze_setup function. I have debugged this as far as line 1802 of pyaedt.application.analysis.py:

`self.odesign.Analyze(name)`

I think this calls the underlying ScriptEnv function because I cannot step into that. In any case, since I cannot step into that function, I cannot debug any further.

I have also tried running PyAEDT in COM mode, rather than gRPC mode, but this still fails.

I am still awaiting a response from Ansys support about what the error message means (Which argument is improper? In what way is it improper?) and what could cause it to be triggered within that function. Of course, there is no guarantee that they will answer those questions.

### Steps To Reproduce

To reproduce the error:

- Ensure TwinBuilder (AEDT) and PyAEDT are installed
- Copy the resistor example, e.g. C:\Program Files\AnsysEM\v231\Win64\Examples\Twin Builder\Components\Basic\Circuit\Passive_elements\Resistor\resistor.aedt, to a writeable directory
- Open a command prompt
- Change directory to where you copied the resistor example
- Run the script below by typing, 'python improper_argument.py resistor.aedt TR' or equivalent
- TwinBuilder will open, the model will load, the analysis will try to start, then the error will be thrown

improper_argument.py
```
# Script to reproduce error: "Simplorer Solver - Encountered an improper argument"
# Usage: python3 improper_argument.py path\to\model.aedt name_of_analysis_to_run
# e.g. python3 improper_argument.py E:\pyaedt-sandpit\resistor.aedt TR
from pyaedt import TwinBuilder
import sys

app = TwinBuilder()
app.load_project(sys.argv[1])
app.analyze_setup(sys.argv[2])
app.close_desktop()
```

To reproduce the working commands from an interactive session:

- Ensure TwinBuilder (AEDT) and PyAEDT are installed
- Copy the resistor example above to a writeable directory (or be sure to delete the lock file if continuing from the same directory as above)
- Open a command prompt
- Change directory to where you copied the resistor example
- Start an interactive session by typing, 'python'
- Enter the following commands into the Python interpreter:
    - import sys
    - sys.argv.append("resistor.aedt")
    - sys.argv.append("TR")
    - import improper_argument
- TwinBuilder will open, the model will load, the analysis runs successfully (plots will update), and AEDT will close, all without error

Here is the script provided by Ansys support to demonstrate that the ScriptEnv method works. The paths will need editing for your machine:
```
import sys

AEDT_PATH = r"C:\Program Files\AnsysEM\v232\Win64"
# Path to project - must be in location with read/write permissions
EXAMPLE_PATH = r"E:\pyaedt-sandpit\resistor.aedt"

sys.path.append(AEDT_PATH + r"\PythonFiles\DesktopPlugin")
import ScriptEnv

ScriptEnv.InitializeNew()
oDesktop.OpenProject(EXAMPLE_PATH)
oProject = oDesktop.SetActiveProject("resistor")
oDesign = oProject.SetActiveDesign("resistor")
oDesign.Analyze("TR")
oDesktop.CloseProject("resistor")

ScriptEnv.Shutdown()
```

And here is the combination script to demonstrate that the problem is within the PyAEDT analyze_setup function, rather than the opening of TwinBuilder or the model by PyAEDT. Again, the paths will need editing for your system:
```
# Script to test error: "Simplorer Solver - Encountered an improper argument"
# Usage: python3 combo_test.py
from pyaedt import TwinBuilder
import sys

AEDT_PATH = r"C:\Program Files\AnsysEM\v232\Win64"
sys.path.append(AEDT_PATH + r"\PythonFiles\DesktopPlugin")
import ScriptEnv

app = TwinBuilder()
app.load_project("resistor.aedt")

ScriptEnv.Initialize("TwinBuilder", port=app.odesktop.GetGrpcServerPort())
oProject = oDesktop.SetActiveProject("resistor")
oDesign = oProject.SetActiveDesign("resistor")
oDesign.Analyze("TR")
oDesktop.CloseProject("resistor")

ScriptEnv.Shutdown()
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

```
ansys-pythonnet==3.1.0rc3
cffi==1.16.0
clr-loader==0.2.6
plumbum==1.8.2
psutil==5.9.6
pyaedt==0.7.3
pycparser==2.21
pywin32==306
rpyc==5.3.1
```

Samuelopez-ansys, 2023-11-13:
@steve-biggs-fox I have tested a project from AEDT examples (AC_3Phase_Voltage_Source.aedt) and it works:

```
import pyaedt
app = pyaedt.TwinBuilder(specified_version="2023.2")
app.load_project(r"C:\Users\slopez\Downloads\AC_3Phase_Voltage_Source.aedt")
app.analyze_setup("TR")
app.close_desktop()
```
Which AEDT version are you using? Please if you can install the latest version, you will have fewer compatibility problems. In addition, I prefer this code ( but it is equivalent):
```
m1 = pyaedt.TwinBuilder(specified_version="2023.2",
                        new_desktop_session=False,
                        projectname=r"C:\Users\slopez\Downloads\AC_3Phase_Voltage_Source.aedt")

m1.analyze("TR")

m1.release_desktop()
```

Samuelopez-ansys, 2023-11-13:
I am using Pyaedt 7.0.4

steve-biggs-fox, 2023-11-13:
I already had the latest version of AEDT, 2023 R2.

I was already on PyAEDT 0.7.2, which was the latest a few weeks ago. I have upgraded to 0.7.4 (the latest) just now, but the problem persists.

I tested the model I was using (resistor.aedt) and the one you were using (AC_3Phase_Voltage_Source.aedt), and the error occurs with both models.

Did you run the commands as a script? i.e. from the command line:

`python improper_argument.py resistor.aedt TR`

where the commands are saved in improper_argument.py. **This is very important**, as the error does not occur if the commands are entered into an interactive Python session; the error only occurs when the commands are run as a script.

Assuming you have run it as a script, I'm trying to think of what else could be different between our environments:

- What AEDT license have you got? Local or server? Academic or commercial? Maybe if we have different license configurations that could be the source of the problem?
- What OS are you using? I'm on Windows. I cannot easily test Linux.

Samuelopez-ansys, 2023-11-13:
Hi @steve-biggs-fox ,

I am running the script as well, very likely you are using a different virtual environment with your command:
python improper_argument.py resistor.aedt TR

To be sure you are using the virtual environment with the correct PyAEDT version, first please activate it:
[image]

Then you will use the PyAEDT installed (pip freeze):
[image]

And finally, run the script:
[image]

My script has harcoded the project name, take care with the path because python needs the double \\ in the path or put an "r" at the beginning (please see my first example).

My license is the commercial one, I am running locally and I am using Windows.

steve-biggs-fox, 2023-11-14:
I already had the virtual environment activated, and I checked and it is definitely using the executable from the virtual environment.

But I've found the problem!

I tried running your script with the hardcoded filename and analysis name and it works. So, then I made the changes one at a time to get back to my script with sys.argv, and I found my script even works with the command as written, i.e.

`python improper_argument.py resistor.aedt TR`

Except I was not using that exact command! I was using the Windows PowerShell auto-completion, so it was this:

`python .\improper_argument.py .\resistor.aedt TR`

It seems there is a problem with `.\resistor.aedt` vs just `resistor.aedt`.

So, I tried a few combinations, and this is what I found:

- `.\resistor.aedt` as a command line argument gets read into Python as `.\\resistor.aedt`. Similarly, the full path gets read in with double slashes, e.g. in my case `E:\pyaedt-sandpit\resistor.aedt` becomes `E:\\pyaedt-sandpit\\resistor.aedt`. This suggests that the problem is not to do with `\r` becoming a carriage return or similar.
- When called correctly (either by giving no path when the model file is in the same directory, or the full path), then the model shows in the AEDT Message Manager as `resistor (E:/pyaedt-sandpit/)`.
- However, when called incorrectly (i.e. using the relative path `.\resistor.aedt`), then the model shows in the message manager as `resistor (E:/pyaedt-sandpit/ ./)`. Notice the extra space! This, combined with the fact that the model can be loaded, suggests that the model is loaded correctly because the relative path is valid, but that this incorrectly assigns an "argument" - let's call it the save directory - as "E:/pyaedt-sandpit/ ./" which is invalid because it has an extra space in it, hence why the argument later gets reported as "improper".
- The error happens even if I use the script provided by Ansys that only uses the ScriptEnv if the model path is hardcoded as `.\resistor.aedt`. Therefore, this is not a PyAEDT bug, but an Ansys EDT bug, where an extra space is being added to the save directory when the model file is given as a relative path.

So, thank you for your help in locating the root cause of this error. I will report back to Ansys with the findings. Since this is not a PyAEDT bug, this issue can be closed, though it may be worth noting this pecularity in the documentation somewhere.

Thanks again :)