ansys/pyaedt #3881: primitive property validation

ring630, 2023-11-15:
### Description of the feature

The layout database from EDA tools can contain illegal objects. One example is a copper shape(it is a type of primitive in EDB) without a associated net. This will potentially lead to error in simulation. There need to be a validation check on primitives.

scsipi, 2023-11-15:
It would be good if pyAEDTs validation checks follow the same validation/rejection logic of copper features as similar checks in the SIwave and AEDT (3D Layout) GUIs. Or, at least have an option (boolean parameter?) for the pyAEDT call to behave the same as the GUI (and IronPython API).

That way, if I am using pyAEDT to automate a workflow, I know I will get the same results as when I perform the same checks via the GUIs.

One way could be for the pyAEDT validation checks to have the option to ignore any objects outside of the board outline. This may be what the GUI checks currently do, but I don't know for sure.

ring630, 2023-11-15:
@scsipi absolutely. PyAEDT should behave the same as AEDT GUI. I think AEDT GUI does a certain fix on "$NONE$", do you see any related warning when importing ODB++?

scsipi, 2023-11-16:
@ring630 I see some warnigns: some are expected, but some have no description against them. There is nothing that mentions "$NONE$" or unnamed nets etc.

I've attached the Translator.log to Ansys Case 00032574 which _may_ be related to the handling of "$NONE$".