ansys/pyaedt #3888: Bug located in EDB padstack definitions when layer names are changed

pyAnsysUser, 2023-11-16:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When an EDB layer name is changed, a warning is given about unknown layer names (previous names) in padstack definitions. The resulting solution appears to correctly apply the layer names and implement the via padstacks; The bug presents as incorrectly reporting the warning message.

### Steps To Reproduce

Here I create a minimal example that reproduces the warning, however it errors on solve since ports are not defined
```
target_file = os.path.join(mytest_path1, r'edb/ANSYS-HSD_V1.aedb')
edb_version = "2023.1"
with pyaedt.Edb(target_file, edbversion = edb_version) as edb:
    signal_layer_list = []
    for _, layer in edb.stackup.stackup_layers.items():
        if layer.type == 'signal':
            signal_layer_list.append(layer)
    
    for n_layer, layer in enumerate(signal_layer_list):
        layer.name = f"new_signal_name_{n_layer}"

    for layer_name, layer in edb.stackup.stackup_layers.items():
        print(f'New layer name is {layer_name}')
    
    edb.save_edb()
    edb.solve_siwave()

    
    edb.close_edb()
```

```
Warning: Unknown layer <1_Top> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner2(PWR1)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner3(Sig1)> referenced in padstack definition <c100hn140>
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
cffi==1.16.0
clr-loader==0.2.6
plumbum==1.8.2
psutil==5.9.6
pyaedt==0.7.4
pycparser==2.21
pywin32==306
rpyc==5.3.1

ring630, 2023-11-20:
@pyAnsysUser can you reproduce the same problem on the example design?

[
_unittest/example_models/TEDB/ANSYS-HSD_V1.aedb/edb.def
](https://github.com/ansys/pyaedt/blob/main/_unittest/example_models/TEDB/ANSYS-HSD_V1.aedb/edb.def)

pyAnsysUser, 2023-11-20:
@ring630 the code I wrote above I believe to use this example design and the error is displayed for me when it is run

ring630, 2023-11-20:
@pyAnsysUser I cannot reproduce it on my machine.  can you try my code? 

```
import pyaedt

target_file = "ANSYS-HSD_V1.aedb"

edb_version = "2023.2"

with pyaedt.Edb(target_file, edbversion=edb_version) as edb:
    signal_layer_list = []
    for _, layer in edb.stackup.stackup_layers.items():
        if layer.type == "signal":
            signal_layer_list.append(layer)

    for n_layer, layer in enumerate(signal_layer_list):
        layer.name = f"new_signal_name_{n_layer}"

    for layer_name, layer in edb.stackup.stackup_layers.items():
        print(f"New layer name is {layer_name}")

    edb.close_edb()
```
[image]

pyAnsysUser, 2023-11-20:
@ring630 that's a puzzler. I installed 2023.2 on a second machine and am not able to replicate the issue, even if I use the 2023.1 installation:
```
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Logger is initialized in EDB.
PyAEDT INFO: pyaedt v0.7.4
PyAEDT INFO: Python version 3.11.1 (tags/v3.11.1:a7a450f, Dec  6 2022, 19:58:39) [MSC v.1934 64 
bit (AMD64)]
PyAEDT INFO: Database ANSYS-HSD_V1.aedb Opened in 2023.1
PyAEDT INFO: Cell main Opened
PyAEDT INFO: Builder was initialized.
PyAEDT INFO: EDB initialized.
New layer name is new_signal_name_0
New layer name is DE1
New layer name is new_signal_name_1
New layer name is DE2
New layer name is new_signal_name_2
New layer name is DE3
New layer name is new_signal_name_3
New layer name is Megtron4-1mm
New layer name is new_signal_name_4
New layer name is DE5
New layer name is new_signal_name_5
New layer name is DE6
New layer name is new_signal_name_6
New layer name is DE7
New layer name is new_signal_name_7
PyAEDT INFO: EDB file save time: 1.99ms
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:information 0050 siwave_ng 2023 R1 (Built: Nov 20 2022 23:34:23):
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:error 0506 command file "C:/Users/720057/Documents/VScode/ddr/data/pyaedt_matched/edb/ANSYS-HSD_V1.exec" could not be opened for reading:     
PyAEDT INFO: EDB file release time: 1.99ms
```

running the same file on my original machine, I still see the same message:
```
c:\Users\720057\Documents\Python_VCodes\pyaedt_workspace\.venv\Lib\site-packages\pyaedt\edb_core\edb_data\components_data.py:12: UserWarning: The NumPy module is required to run some functionalities of EDB.
Install with

pip install numpy

Requires CPython.
  warnings.warn(
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Logger is initialized in EDB.
PyAEDT INFO: pyaedt v0.7.4
PyAEDT INFO: Python version 3.11.1 (tags/v3.11.1:a7a450f, Dec  6 2022, 19:58:39) [MSC v.1934 64 bit (AMD64)]
PyAEDT INFO: Database ANSYS-HSD_V1.aedb Opened in 2023.1
PyAEDT INFO: Cell main Opened
PyAEDT INFO: Builder was initialized.
PyAEDT INFO: EDB initialized.
New layer name is new_signal_name_0
New layer name is DE1
New layer name is new_signal_name_1
New layer name is DE2
New layer name is new_signal_name_2
New layer name is DE3
New layer name is new_signal_name_3
New layer name is Megtron4-1mm
New layer name is new_signal_name_4
New layer name is DE5
New layer name is new_signal_name_5
New layer name is DE6
New layer name is new_signal_name_6
New layer name is DE7
New layer name is new_signal_name_7
PyAEDT INFO: EDB file save time: 1.99ms
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:information 0050 siwave_ng 2023 R1 
(Built: Nov 20 2022 23:34:23):
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:message Loading project...:
Warning: Unknown layer <1_Top> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <c100hn140>
[... 306 lines cut ...]
Warning: Unknown layer <1_Top> referenced in padstack definition <v60h25>
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <v60h25>     
Warning: Unknown layer <Inner2(PWR1)> referenced in padstack definition <v60h25>     
Warning: Unknown layer <Inner3(Sig1)> referenced in padstack definition <v60h25>     
Warning: Unknown layer <Inner4(Sig2)> referenced in padstack definition <v60h25>     
Warning: Unknown layer <Inner5(PWR2)> referenced in padstack definition <v60h25>     
Warning: Unknown layer <Inner6(GND2)> referenced in padstack definition <v60h25>     
Warning: Unknown layer <16_Bottom> referenced in padstack definition <v60h25>        
Warning: Unknown layer <1_Top> referenced in padstack definition <v70h30m0mx0>       
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <v70h30m0mx0>Warning: Unknown layer <Inner2(PWR1)> referenced in padstack definition <v70h30m0mx0>Warning: Unknown layer <Inner3(Sig1)> referenced in padstack definition <v70h30m0mx0>Warning: Unknown layer <Inner4(Sig2)> referenced in padstack definition <v70h30m0mx0>Warning: Unknown layer <Inner5(PWR2)> referenced in padstack definition <v70h30m0mx0>Warning: Unknown layer <Inner6(GND2)> referenced in padstack definition <v70h30m0mx0>Warning: Unknown layer <16_Bottom> referenced in padstack definition <v70h30m0mx0>   
380 capacitors read
10 inductors read
82 resistors read
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:message Initializing layout database...:
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:information 0050 Simulation "SYZ Sweep 1" started on host COLLWM006307 at Mon Nov 20 11:08:43 2023:
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:message Generating solver input files...:
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:progress  5.000000:
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:information 0050 [Simulation Directory]: "C:/Users/720057/Documents/Python_VCodes/pyaedt_workspace/layer_name/pyaedt_matched/edb/ANSYS-HSD_V1.siwaveresults/0000/":
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:error 0503 SYZ-parameter simulation canceled: no ports defined:   
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:information 0050 SYZ solver completed with return code 5:
C:\Program Files\AnsysEM\v231\Win64\siwave_ng.exe:error 0506 Simulation "SYZ Sweep 1" terminated on host COLLWM006307 at Mon Nov 20 11:08:43 2023 with exit code 5:
PyAEDT INFO: EDB file release time: 1.99ms
```

If you don't have any further ideas, I can chalk this up to something with either my installation or something that will be resolved when I uprev to 2023.2 on my personal machine.

pyAnsysUser, 2023-11-20:
@ring630 Reproduced with a few lines added to your code to create the .exec:

```
import pyaedt

target_file = "ANSYS-HSD_V1.aedb"

edb_version = "2023.2"

with pyaedt.Edb(target_file, edbversion=edb_version) as edb:
    signal_layer_list = []
    for _, layer in edb.stackup.stackup_layers.items():
        if layer.type == "signal":
            signal_layer_list.append(layer)

    for n_layer, layer in enumerate(signal_layer_list):
        layer.name = f"new_signal_name_{n_layer}"

    for layer_name, layer in edb.stackup.stackup_layers.items():
        print(f"New layer name is {layer_name}")

    edb.save_edb()
    edb.siwave.create_exec_file(add_dc = True)
    edb.solve_siwave()
```

```
c:\Users\720057\Documents\VScode\ddr\.venv\Lib\site-packages\pyaedt\edb_core\edb_data\components_data.py:12: UserWarning: The NumPy module is required to run some functionalities of EDB.      
Install with

pip install numpy

Requires CPython.
  warnings.warn(
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Logger is initialized in EDB.
PyAEDT INFO: pyaedt v0.7.4
PyAEDT INFO: Python version 3.11.1 (tags/v3.11.1:a7a450f, Dec  6 2022, 19:58:39) [MSC v.1934 64 
bit (AMD64)]
PyAEDT INFO: Database ANSYS-HSD_V1.aedb Opened in 2023.2
PyAEDT INFO: Cell main Opened
PyAEDT INFO: Builder was initialized.
PyAEDT INFO: EDB initialized.
New layer name is new_signal_name_0
New layer name is DE1
New layer name is new_signal_name_1
New layer name is DE2
New layer name is new_signal_name_2
New layer name is DE3
New layer name is new_signal_name_3
New layer name is Megtron4-1mm
New layer name is new_signal_name_4
New layer name is DE5
New layer name is new_signal_name_5
New layer name is DE6
New layer name is new_signal_name_6
New layer name is DE7
New layer name is new_signal_name_7
PyAEDT INFO: EDB file save time: 2.99ms
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:information 0050 siwave_ng 2023 R2.1 (Built: Sep 14 2023 18:57:48):
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:message Loading project...:
Warning: Unknown layer <1_Top> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner2(PWR1)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner3(Sig1)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner4(Sig2)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner5(PWR2)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <Inner6(GND2)> referenced in padstack definition <c100hn140>
Warning: Unknown layer <16_Bottom> referenced in padstack definition <c100hn140>
Warning: Unknown layer <1_Top> referenced in padstack definition <c120h70>
[... 301 lines cut ...]
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <v60h25>
Warning: Unknown layer <Inner2(PWR1)> referenced in padstack definition <v60h25>
Warning: Unknown layer <Inner3(Sig1)> referenced in padstack definition <v60h25>
Warning: Unknown layer <Inner4(Sig2)> referenced in padstack definition <v60h25>
Warning: Unknown layer <Inner5(PWR2)> referenced in padstack definition <v60h25>
Warning: Unknown layer <Inner6(GND2)> referenced in padstack definition <v60h25>
Warning: Unknown layer <16_Bottom> referenced in padstack definition <v60h25>
Warning: Unknown layer <1_Top> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <Inner1(GND1)> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <Inner2(PWR1)> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <Inner3(Sig1)> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <Inner4(Sig2)> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <Inner5(PWR2)> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <Inner6(GND2)> referenced in padstack definition <v70h30m0mx0>
Warning: Unknown layer <16_Bottom> referenced in padstack definition <v70h30m0mx0>
380 capacitors read
10 inductors read
82 resistors read
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:message Initializing layout database...:
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:information 0050 Simulation "DC IR Sim 1" started on host COL-L339731D001 at Mon Nov 20 11:41:25 2023:
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:message Generating solver input files...:
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:error 0504 DC IR drop simulation canceled: no 
sources, ports or terminals defined:
C:\Program Files\AnsysEM\v232\Win64\siwave_ng.exe:information 0050 DC solver completed with return code 6:
```

ring630, 2023-11-21:
@pyAnsysUser I see the problem. The padstackup definition is not updated with the new layer names. The parameters of the padstacks are correct, so it is not causing any problem in simulation. To eliminate all the warnings, all of the padstack definitions should be refreashed.

@svandenb-dev