ansys/pyaedt #3908: Bug located in creating the MeshOperation and MeshRegion object for Icepak.

sivasu37, 2023-11-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

There are no keys with the name MeshOperations and MeshRegion

In the MeshIcepak.py _app.design_properties["MeshRegion"]["MeshSetup"]["MeshOperation"] is used. This  gives KeyError as there is no key MeshOperation and passes through try except block and doesn't create the meshoperation object list.  It needs to be modified. 

Whether a particular setup is MeshOperation or not,  needs be inferred from the property and there is no key "MeshOperation". Similary there is no key "MeshRegion". 

Siva

### Steps To Reproduce

when try to get the existing mesh operations in a model get_design_mesh_operations function. It always return an empty list. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

ansys-pythonnet==3.1.0rc3
asttokens==2.2.1
backcall==0.2.0
cffi==1.16.0
clr-loader==0.2.6
colorama==0.4.6
comm==0.1.2
debugpy==1.6.6
decorator==5.1.1
executing==1.2.0
importlib-metadata==6.0.0
ipykernel==6.21.1
ipython==8.9.0
jedi==0.18.2
jupyter_client==8.0.2
jupyter_core==5.2.0
matplotlib-inline==0.1.6
nest-asyncio==1.5.6
packaging==23.0
parso==0.8.3
pickleshare==0.7.5
platformdirs==3.0.0
plumbum==1.8.2
prompt-toolkit==3.0.36
psutil==5.9.4
pure-eval==0.2.2
pyaedt==0.7.4
pycparser==2.21
Pygments==2.14.0
python-dateutil==2.8.2
pywin32==305
pyzmq==25.0.0
rpyc==5.3.1
six==1.16.0
stack-data==0.6.2
tornado==6.2
traitlets==5.9.0
wcwidth==0.2.6
zipp==3.12.1

sivasu37, 2023-11-23:
The issue is with 2024R1. The file structure got changed in 2024R1

In new structure the type whether MeshRegion or MeshOperation is inferred by DType
$begin 'MeshRegion1'
					DType='RegionT'
					ID=1
					IsComponent=false
					Enable=true
					MeshMethod='MesherHD'
					UserSpecifiedSettings=false
					MeshRegionResolution=3
					GeometryBasedMeshing=false
					MinGapX='1mm'
					MinGapY='1mm'
					MinGapZ='1mm'
					Objects(48)
					ProximitySizeFunction=true
					CurvatureSizeFunction=true
					EnableTransition=false
					OptimizePCBMesh=true
					Enable2DCutCell=false
					EnforceCutCellMeshing=false
					Enforce2dot5DCutCell=false
				$end 'MeshRegion1'
				$begin 'MeshOperation1'
					DType='OpT'
					ID=2
					IsComponent=false
					Enable=true
					MaxLevel='0'
					MinLevel='0'
					'Local Mesh Parameters Enabled'=false
					'Mesh Reuse Enabled'=false
					'Mesh Reuse File'=''
					Objects(59)
					'Mesh Object(s) Separately Enabled'=false
					'Local Mesh Parameters Type'='Sheet Local Mesh Parameters'
					'Along Normal (Positive) height'='0mm'
					'Along Normal (Positive) ratio'='0'
					'Along Normal (Negative) height'='0mm'
					'Along Normal (Negative) ratio'='0'
				$end 'MeshOperation1'

lorenzovecchietti, 2024-06-03:
I think this was solved in #4015

closed 2024-06-03 by lorenzovecchietti