ansys/pyaedt #3913: Bug located in pyaedt\application\Design.py

dcrawforAtAnsys, 2023-11-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

It doesn't seem to be possible to change the active design for a ``Maxwell2D`` instance that has multiple designs.

### Steps To Reproduce

A Maxwell2D object instance is associated with multiple designs. When trying to set 
```
m2d.design_name = m2d.design_names[0]
```
an error occurs.
```
Traceback (most recent call last):
  File "C:\Users\dcrawfor\repo\docs\esobu-techdocs\venv\lib\site-packages\pyaedt\generic\grpc_plugin.py", line 120, in __Invoke__
    return _retry_ntimes(
  File "C:\Users\dcrawfor\repo\docs\esobu-techdocs\venv\lib\site-packages\pyaedt\generic\general_methods.py", line 712, in _retry_ntimes
    raise AttributeError("Error in Executing Method {}.".format(function.__name__))
AttributeError: Error in Executing Method InvokeAedtObjMethod.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\dcrawfor\repo\docs\esobu-techdocs\venv\lib\site-packages\IPython\core\interactiveshell.py", line 3508, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-7-ee20d7945c19>", line 1, in <module>
    m2d.design_name = "01 reference"
  File "C:\Users\dcrawfor\repo\docs\esobu-techdocs\venv\lib\site-packages\pyaedt\application\Design.py", line 622, in design_name
    self.odesign.RenameDesignInstance(self.design_name, new_name)
  File "C:\Users\dcrawfor\repo\docs\esobu-techdocs\venv\lib\site-packages\pyaedt\generic\grpc_plugin.py", line 133, in DynamicFunc
    return self.__Invoke__(funcName, args)
  File "C:\Users\dcrawfor\repo\docs\esobu-techdocs\venv\lib\site-packages\pyaedt\generic\grpc_plugin.py", line 124, in __Invoke__
    raise GrpcApiError("Failed to execute grpc AEDT command: {}".format(funcName))
pyaedt.generic.general_methods.GrpcApiError: Failed to execute grpc AEDT command: RenameDesignInstance
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

all dev dependencies are installed.

dcrawforAtAnsys, 2023-11-25:
The correct method to use in this case is ``set_active_design()``. 

Therefore, if the property ``design_name`` is changed from one string that is in the ``design_names`` list to another design name that is also in the list, the setter for ``design_name`` should call ``set_active_design()`` and update the active design.

dcrawforAtAnsys, 2024-05-28:
- [ ] Review and update docstring for the Design class and corresponding methods.

Samuelopez-ansys, 2025-04-28:
Hi @Devin-Crawford , What is the status of this issue?

closed 2025-06-23 by Samuelopez-ansys