ansys/pyaedt #3931: Issue when running the example given in pyaedt HFSS

pavankumarkonchada, 2023-11-30:
I have pyaedt installed with python 11. When I am trying to run the example it is stopping when accessing the lumped_port. 
Throwing an attribute error saying Hfss doest have an attribute lumped_port

pavankumarkonchada, 2023-11-30:
The issue resolved on using 2023r1 version, but now I am getting bool object is not subscriptable on using Hfss Antenna example

pvinod13, 2023-12-01:
@pavankumarkonchada, 

Did you mean Python 3.11..?

May I know, which example you are refereeing to.

pavankumarkonchada, 2023-12-01:
Yes python 3.11 and the error I am facing is in HFSS Spiral Example
"""
HFSS: spiral inductor
---------------------
This example shows how you can use PyAEDT to create a spiral inductor, solve it, and plot results.
"""

#############################################################
# Perform required imports
# ~~~~~~~~~~~~~~~~~~~~~~~~
# Perform required imports.

import os
import pyaedt

project_name = pyaedt.generate_unique_project_name(project_name="spiral")

#############################################################
# Set non-graphical mode
# ~~~~~~~~~~~~~~~~~~~~~~
# Set non-graphical mode. 
# You can set ``non_graphical`` either to ``True`` or ``False``.

non_graphical = False

#############################################################
# Launch HFSS
# ~~~~~~~~~~~
# Launch HFSS 2023 R2 in non-graphical mode and change the
# units to microns.

hfss = pyaedt.Hfss(specified_version="2023.1", non_graphical=non_graphical, designname="A1", new_desktop_session=True)
hfss.modeler.model_units = "um"
p = hfss.modeler

#############################################################
# Define variables
# ~~~~~~~~~~~~~~~~
# Define input variables. You can use the values that follow or edit
# them.

rin = 10
width = 2
spacing = 1
thickness = 1
Np = 8
Nr = 10
gap = 3
hfss["Tsub"] = "6" + hfss.modeler.model_units

#############################################################
# Standardize polyline
# ~~~~~~~~~~~~~~~~~~~~
# Standardize the polyline using the ``create_line`` method to fix
# the width, thickness, and material.

def create_line(pts):
    p.create_polyline(pts, xsection_type="Rectangle", xsection_width=width, xsection_height=thickness, matname="copper")

################################################################
# Create spiral inductor
# ~~~~~~~~~~~~~~~~~~~~~~
# Create the spiral inductor. This spiral inductor is not
# parametric, but you could parametrize it later.

ind = hfss.modeler.create_spiral(
    internal_radius=rin,
    width=width,
    spacing=spacing,
    turns=Nr,
[... 168 lines cut ...]
PyAEDT ERROR: Check Online documentation on: https://aedt.docs.pyansys.com/version/stable/search.html?q=plot
PyAEDT INFO: Linear count sweep Sweep_GRCT9F has been correctly created.
PyAEDT INFO: Project Project3 Saved correctly
PyAEDT INFO: Key Desktop/ActiveDSOConfigurations/HFSS correctly changed.
PyAEDT INFO: Solving all design setups
PyAEDT INFO: Key Desktop/ActiveDSOConfigurations/HFSS correctly changed.
PyAEDT ERROR: Error in solving all setups (AnalyzeAll).
PyAEDT WARNING: Solution Data failed to load. Check solution, context or expression.
PyAEDT WARNING: No Data Available. Check inputs
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [1], in <cell line: 188>()
    182 ################################################################
    183 # Plot calculated values in Matplotlib
    184 # ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    185 # Plot the calculated values in Matplotlib.
    187 data = hfss.post.get_solution_data([L_formula, Q_formula])
--> 188 data.plot(curves=[L_formula, Q_formula], math_formula="re", xlabel="Freq", ylabel="L and Q")
    190 ################################################################
    191 # Export results to csv file
    192 # ~~~~~~~~~~~~~~~~~~~~~~~~~~
    193 # Export results to csv file
    194 data.export_data_to_csv(os.path.join(hfss.toolkit_directory,"output.csv"))

AttributeError: 'bool' object has no attribute 'plot'

SMoraisAnsys, 2023-12-01:
When a boolean result is obtained, it might be because the function did not work as expected.
Thus, instead of return an instance of some class `A` the result is `false`.
I don't know where this error is coming from but it might be related to your python environment or something else.
Could you please show us the python modules you have in your Python interpreter (e.g. result of `python -m pip freeze`) ?
Are you able to test pyaedt and do all test pass ? To run the tests:
1) install pyaedt and its test dependencies with `python -m pip install pyaedt[tests]`
2) from the project directory, run the tests with `pytest -n 6 --dist loadfile _unittest` (this will run on 6 computation units instead of sequential one, please adapt the value depending on your machine)

pavankumarkonchada, 2023-12-01:
aiohttp @ file:///C:/ci/aiohttp_1646806572557/workNote: you may need to restart the kernel to use updated packages.
aiosignal @ file:///tmp/build/80754af9/aiosignal_1637843061372/work
alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
altgraph==0.17.3
anaconda-client @ file:///C:/ci/anaconda-client_1635342725944/work
anaconda-navigator==2.1.4
anaconda-project @ file:///tmp/build/80754af9/anaconda-project_1637161053845/work
ansys-api-fluent==0.3.0
ansys-api-platform-instancemanagement==1.0.0b3
ansys-fluent-core==0.10.1
ansys-fluent-parametric==0.4.1
ansys-fluent-visualization==0.4.1
ansys-platform-instancemanagement==1.0.2
ansys-pythonnet==3.1.0rc3
anyio @ file:///C:/ci/anyio_1644481921011/work/dist
appdirs==1.4.4
argon2-cffi @ file:///opt/conda/conda-bld/argon2-cffi_1645000214183/work

argon2-cffi-bindings @ file:///C:/ci/argon2-cffi-bindings_1644551690056/work
arrow @ file:///opt/conda/conda-bld/arrow_1649166651673/work
astroid @ file:///C:/ci/astroid_1628063282661/work
astropy @ file:///C:/ci/astropy_1650634291321/work
asttokens @ file:///opt/conda/conda-bld/asttokens_1646925590279/work
async-timeout @ file:///tmp/build/80754af9/async-timeout_1637851218186/work
atomicwrites==1.4.0
attrs @ file:///opt/conda/conda-bld/attrs_1642510447205/work
Automat @ file:///tmp/build/80754af9/automat_1600298431173/work
autopep8 @ file:///opt/conda/conda-bld/autopep8_1639166893812/work
Babel @ file:///tmp/build/80754af9/babel_1620871417480/work
backcall @ file:///home/ktietz/src/ci/backcall_1611930011877/work
backports.functools-lru-cache @ file:///tmp/build/80754af9/backports.functools_lru_cache_1618170165463/work
backports.tempfile @ file:///home/linux1/recipes/ci/backports.tempfile_1610991236607/work
backports.weakref==1.0.post1
bcrypt @ file:///C:/ci/bcrypt_1607022693089/work
beautifulsoup4 @ file:///C:/ci/beautifulsoup4_1650293025093/work
binaryornot @ file:///tmp/build/80754af9/binaryornot_1617751525010/work
bitarray @ file:///C:/ci/bitarray_1648739663053/work
bkcharts==0.2
black==19.10b0
bleach @ file:///opt/conda/conda-bld/bleach_1641577558959/work
bokeh @ file:///C:/ci/bokeh_1638362966927/work
boto3 @ file:///opt/conda/conda-bld/boto3_1649078879353/work
botocore @ file:///opt/conda/conda-bld/botocore_1649076662316/work
Bottleneck @ file:///C:/ci/bottleneck_1648010904582/work
brotlipy==0.7.0
cachetools @ file:///tmp/build/80754af9/cachetools_1619597386817/work
certifi==2021.10.8
cffi @ file:///C:/ci_310/cffi_1642682485096/work
chardet @ file:///C:/ci/chardet_1607706937985/work
charset-normalizer @ file:///tmp/build/80754af9/charset-normalizer_1630003229654/work
click @ file:///C:/ci/click_1646038595831/work
cloudpickle @ file:///tmp/build/80754af9/cloudpickle_1632508026186/work
clr-loader==0.2.6
clyent==1.2.2
colorama @ file:///tmp/build/80754af9/colorama_1607707115595/work
colorcet @ file:///tmp/build/80754af9/colorcet_1611168489822/work
comtypes==1.1.10
conda==4.12.0
conda-build==3.21.8
conda-content-trust @ file:///tmp/build/80754af9/conda-content-trust_1617045594566/work
conda-pack @ file:///tmp/build/80754af9/conda-pack_1611163042455/work
conda-package-handling @ file:///C:/ci/conda-package-handling_1649106011304/work
conda-repo-cli @ file:///tmp/build/80754af9/conda-repo-cli_1620168426516/work
conda-token @ file:///tmp/build/80754af9/conda-token_1620076980546/work
conda-verify==3.4.2
constantly==15.1.0
cookiecutter @ file:///opt/conda/conda-bld/cookiecutter_1649151442564/work
cryptography @ file:///C:/ci/cryptography_1633520531101/work
cssselect==1.1.0
cx-Freeze==6.15.3
[... 282 lines cut ...]
vtk==9.1.0
w3lib @ file:///Users/ktietz/demo/mc3/conda-bld/w3lib_1629359764703/work
watchdog @ file:///C:/ci/watchdog_1638367441841/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci/websocket-client_1614804375980/work
Werkzeug @ file:///opt/conda/conda-bld/werkzeug_1645628268370/work
widgetsnbextension @ file:///C:/ci/widgetsnbextension_1644991377168/work
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win-unicode-console==0.5
wincertstore==0.2
wrapt @ file:///C:/ci/wrapt_1607574570428/work
wslink==1.8.2
xarray @ file:///opt/conda/conda-bld/xarray_1639166117697/work
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///opt/conda/conda-bld/xlsxwriter_1649073856329/work
xlwings==0.24.9
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl @ file:///C:/ci/yarl_1606940155993/work
zict==2.0.0
zipp @ file:///opt/conda/conda-bld/zipp_1641824620731/work
zope.interface @ file:///C:/ci/zope.interface_1625036252485/work

Above  is the result what i get when I run pip freeze 
python -m pip freeze

SMoraisAnsys, 2023-12-04:
@pavankumarkonchada Thanks for your feedback. What about the tests to see if the installation is working correctly ?

By the way, I still have a question regarding the input you provided us. You mentioned that you use python 3.11 but, from the stack trace you posted, I have the feeling that you work with python 3.9 (see `C:\Users\cadfem\AppData\Roaming\Python\Python39\site-packages\pyaedt\modules\AdvancedPostProcessing.py`). Make sure that you use Python3.11.X and tell me exactly which version it is :)

Since multiple packages have no literal version associated, it will be difficult to reproduce exactly your setup. However, we'll do our best after knowing your python version.

pavankumarkonchada, 2023-12-06:
I have updated still the issue persists as you said i am using python 3.9

pavankumarkonchada, 2023-12-06:
And about the previous query I think it may be due to license issue, but post processing is an issue still

pavankumarkonchada, 2023-12-07:
It is working fine when I m trying to run in pyaedt terminal in electronic desktop, but running using jupyter notebook it is throwing errors

pavankumarkonchada, 2023-12-07:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_12152\1302060210.py in <module>
     98                    name="port1"
     99                    )
--> 100 hfss.lumped_port(signal="port1", integration_line=pyaedt.constants.AXIS.Z)
    101 
    102 ################################################################

AttributeError: 'Hfss' object has no attribute 'lumped_port'

closed 2023-12-07 by pavankumarkonchada

SMoraisAnsys, 2023-12-07:
@pavankumarkonchada Could you tell us what version was causing the problem and which one solved it ? :)

pavankumarkonchada, 2023-12-19:
python 3.12