ansys/pyaedt #3934: Bug located in maxwell3d, Object3d.unite(object_list). More than hundreds of objects unite operation return error

Andrew993-chang, 2023-12-01:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I wanna convert the mnist pic into a 3D object. The pixel value equal the depth of the groove on the upper surface. So i create more than 100 vacuum cube that need unite. Same code It works well on Ansys EM 2021.2 and pyaedt v0.6.48. But recent i update to the  Ansys EM 2023.2 and pyaedt v0.7.6, it's always returen error like "PyAEDT ERROR: Error in uniting objects." I tried unite less than 50 object, it works well. But as i unite more entire 132 object, the error shows.  Please help me fix this problem. thanks.

### Steps To Reproduce

X, y = mnist_data()

for z in range(1,3):
    project_name = str(os.path.join(root, "array_"+str(z)+".aedt"))
    projects['project_' + str(z)] = Maxwell3d(projectname=project_name, designname="array_" + str(z),
                                                  close_on_exit=False, non_graphical=False)
    image = X[int(z)].reshape((28, 28)) / 255
    aluminum_dimensions = [44, 44, 2]
    aluminum_block = projects['project_' + str(z)].modeler.create_box([0, 0, 0], aluminum_dimensions,
                                                                      matname="steel_1008", name="test")
    for i in range(28):
        for j in range(28):
            if image[i, j] >= 0.5:
                air_block = projects['project_' + str(z)].modeler.create_box([i * 1.6, j * 1.6, 2],
                                                                             [1.6, 1.6, -(image[i, j])],
                                                                             matname="Vacuum", name="airblock"+str(i)+str(j))
    air_list = projects['project_' + str(z)].modeler.get_objects_w_string("airblock")
    _**projects['project_' + str(z)].modeler.unite(air_list, True, False)**_
    # projects['project_' + str(z)].modeler.subtract(["test"], ["airblock"], keepOriginals=False)
    # projects['project_' + str(z)].modeler.create_region(0, True)
    # projects['project_' + str(z)].analyze(blocking=False)
projects['project_1'].release_desktop(True, True)

D:\Anaconda\python.exe "E:\OneDrive - std.uestc.edu.cn\simu_project\for_temo\creat_environment.py" 
Initializing new desktop!
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Log on file is enabled
PyAEDT INFO: Log on Desktop Message Manager is enabled
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT outside AEDT with gRPC plugin.
PyAEDT INFO: Launching AEDT using the gRPC plugin.
PyAEDT INFO: AEDT file load (threaded) time: 0.33773088455200195
PyAEDT INFO: New AEDT session is starting on gRPC port 4939
PyAEDT INFO: AEDT installation Path D:\AnsysEM\v232\Win64
PyAEDT INFO: Ansoft.ElectronicsDesktop.2023.2 version started with process ID 2948.
PyAEDT INFO: pyaedt v0.7.6
PyAEDT INFO: Python version 3.9.13 (main, Aug 25 2022, 23:51:50) [MSC v.1916 64 bit (AMD64)]
PyAEDT INFO: AEDT 2023.2.0 Build Date 2023-05-16 22:25:19
PyAEDT INFO: Project array_1 has been opened.
PyAEDT INFO: Aedt Objects correctly read
PyAEDT INFO: Modeler3D class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Material library initialized and project materials loaded successfully! Elapsed time: 0m 0sec
PyAEDT ERROR: Error in uniting objects.
Returning found desktop with PID 2948!
PyAEDT INFO: Project array_2 has been opened.
PyAEDT INFO: AEDT file load (threaded) time: 1.5446498394012451
PyAEDT INFO: Aedt Objects correctly read
PyAEDT INFO: Modeler3D class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Material library initialized and project materials loaded successfully! Elapsed time: 0m 0sec
PyAEDT ERROR: Error in uniting objects.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

ansys-api-fluent                      0.3.12
ansys-api-mapdl                       0.5.1
ansys-api-mechanical                  0.1.0
ansys-api-meshing-prime               0.1.1
ansys-api-platform-instancemanagement 1.0.0
ansys-api-pyensight                   0.3.0
ansys-api-sherlock                    0.1.14
ansys-api-systemcoupling              0.1.0
ansys-dpf-composites                  0.2.0
ansys-dpf-core                        0.8.1
ansys-dpf-gate                        0.3.1
ansys-dpf-gatebin                     0.4.1
ansys-dpf-post                        0.4.0
ansys-dynamicreporting-core           0.4.0
ansys-fluent-core                     0.14.0
ansys-grantami-bomanalytics           1.1.3
ansys-grantami-bomanalytics-openapi   1.0.0
ansys-grantami-recordlists            1.0.0
ansys-grantami-serverapi-openapi      1.0.0
ansys-grpc-dpf                        0.8.1
ansys-mapdl-core                      0.64.1
ansys-mapdl-reader                    0.53.0
ansys-math-core                       0.1.1
ansys-mechanical-core                 0.8.0
ansys-meshing-prime                   0.4.1
ansys-motorcad-core                   0.1.3
ansys-openapi-common                  1.2.1
ansys-optislang-core                  0.2.1
ansys-platform-instancemanagement     1.1.1
ansys-pyensight-core                  0.5.3
ansys-pythonnet                       3.1.0rc3
ansys-seascape                        0.2.0
ansys-sherlock-core                   0.2.0
ansys-systemcoupling-core             0.1.3
pyaedt                                0.7.6

Samuelopez-ansys, 2023-12-05:
@Andrew993-chang This should work, please share a script that can reproduce the issue, end-to-end. We do not know the returned value: X, y = mnist_data()

Thanks

Andrew993-chang, 2023-12-05:
@Samuelopez-ansys Please refer to the attachment and run the create_environment.py file. Thanks
[for_debug.zip](https://github.com/ansys/pyaedt/files/13556597/for_debug.zip)

Samuelopez-ansys, 2023-12-05:
Hi @Andrew993-chang,

This is not a PyAEDT bug, I have run your code and it is an AEDT bug. In 2023, AEDT changed the CAD kernel, from ACIS to Parasolid, for that reason in 2022.1, you didn't have this issue. Please contact Ansys technical support with the model directly, because if you try to unite the boxes in the AEDT user interface, it will fail too, then it is not a scripting problem, it is an AEDT problem. I will close this issue.

Thanks,
Samuel

closed 2023-12-05 by Samuelopez-ansys