ansys/pyaedt #3958: Q3D terminals are not getting updated after deletion

mahesh-dhanekula, 2023-12-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

While deleting source(s)/sink(s) in Q3D design, PyAEDT method is returning values that doesn't reflect the deletion operation.

Code snippet is shown below and a simple microstrip line is used as test case.

[image]

Variables during debug.

[image]

AEDT nets tree before deletion.

[image]

AEDT nets tree after deletion.

[image]

### Steps To Reproduce

```
import pyaedt

q3d = pyaedt.Q3d(projectname="terminals", specified_version="2024.1", non_graphical=False)
# Nets having Source & Sink
nets = ["Box2_3", "Box3_1"]
src_bfr_deletion = q3d.net_sources(nets[0])
snk_bfr_deletion = q3d.net_sinks(nets[0])
# Delete the terminals of `Box2_3`
for src in q3d.boundaries_by_type["Source"]:
    if src.name == src_bfr_deletion[0]:
        src.delete()
src_after_deletion = q3d.net_sources(nets[0])
snk_after_deletion = q3d.net_sinks(nets[0])
# Below assertions should return True
print(src_bfr_deletion != src_after_deletion)  # -> Returned False (problem)
print(snk_bfr_deletion == snk_after_deletion)
q3d.release_desktop(False, False)
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.6
aiosignal==1.3.1
ansys-pythonnet==3.1.0rc3
appdirs==1.4.4
archspec @ file:///home/conda/feedstock_root/build_artifacts/archspec_1697661133851/work
asttokens==2.2.1
async-timeout==4.0.3
attrs==23.1.0
av==10.0.0
backcall==0.2.0
bcrypt @ file:///D:/bld/bcrypt_1695372400784/work
boltons @ file:///home/conda/feedstock_root/build_artifacts/boltons_1677499911949/work
Brotli @ file:///D:/bld/brotli-split_1695989908365/work
certifi==2023.7.22
cffi @ file:///D:/bld/cffi_1696001730392/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1698005359935/work
click==8.1.3
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.0
conda @ file:///D:/bld/conda_1698450945992/work
conda-package-handling @ file:///home/conda/feedstock_root/build_artifacts/conda-package-handling_1691048088238/work
conda_package_streaming @ file:///home/conda/feedstock_root/build_artifacts/conda-package-streaming_1691009212940/work
contourpy==1.0.7
cryptography @ file:///D:/bld/cryptography-split_1698192113114/work
cycler==0.11.0
debugpy==1.6.4
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
et-xmlfile==1.1.0
executing==1.2.0
Fiona==1.9.4.post1
fonttools==4.38.0
fpdf2==2.7.4
frozenlist==1.4.0
geopandas==0.13.2
idna @ file:///home/conda/feedstock_root/build_artifacts/idna_1663625384323/work
imageio==2.28.0
imageio-ffmpeg==0.4.8
ipykernel==6.19.4
ipython==8.7.0
ipywidgets==8.1.1
jedi==0.18.2
jsonpatch @ file:///home/conda/feedstock_root/build_artifacts/jsonpatch_1695536281965/work
jsonpointer @ file:///D:/bld/jsonpointer_1695397563781/work
jupyter_client==7.4.8
jupyter_core==5.1.1
jupyterlab-widgets==3.0.9
kiwisolver==1.4.4
matplotlib==3.7.1
matplotlib-inline==0.1.6
menuinst @ file:///D:/bld/menuinst_1698434032802/work
multidict==6.0.4
nest-asyncio==1.5.6
networkx==3.1
numpy==1.24.3
openpyxl==3.1.2
osmnx==1.3.1.post0
packaging==22.0
pandas==2.0.1
paramiko @ file:///home/conda/feedstock_root/build_artifacts/paramiko_1690718676607/work
parso==0.8.3
pickleshare==0.7.5
Pillow==9.4.0
platformdirs==2.6.0
plotly==5.16.1
pluggy @ file:///home/conda/feedstock_root/build_artifacts/pluggy_1693086607691/work
plumbum @ file:///home/conda/feedstock_root/build_artifacts/plumbum_1685457629260/work
pooch==1.6.0
prompt-toolkit==3.0.36
psutil==5.9.4
pure-eval==0.2.2
pyaedt==0.7.7
pycosat @ file:///D:/bld/pycosat_1696356014994/work
pycparser @ file:///home/conda/feedstock_root/build_artifacts/pycparser_1636257122734/work
Pygments==2.13.0
PyNaCl @ file:///D:/bld/pynacl_1695545061432/work
pyOpenSSL @ file:///home/conda/feedstock_root/build_artifacts/pyopenssl_1685514481738/work
pyparsing==3.0.9
pyproj==3.5.0
PySocks @ file:///D:/bld/pysocks_1661604991356/work
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2022.7.1
pyvista==0.38.6
pywin32==305
PyYAML==6.0.1
pyzmq==24.0.1
requests @ file:///home/conda/feedstock_root/build_artifacts/requests_1684774241324/work
rpyc==5.3.1
ruamel.yaml @ file:///D:/bld/ruamel.yaml_1698138763168/work
ruamel.yaml.clib @ file:///D:/bld/ruamel.yaml.clib_1695997037989/work
scikit-rf==0.27.1
scipy==1.10.1
scooby==0.7.1
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
stack-data==0.6.2
tenacity==8.2.3
tornado==6.2
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1691671248568/work
traitlets==5.8.0
trame==2.5.2
trame-client==2.12.6
trame-components==2.2.1
trame-deckgl==2.0.2
trame-markdown==2.0.2
trame-matplotlib==2.0.2
trame-plotly==2.1.1
trame-rca==0.4.1
trame-router==2.1.0
trame-server==2.12.1
trame-simput==2.3.3
trame-vega==2.0.3
trame-vtk==2.6.1
trame-vuetify==2.3.1
truststore @ file:///home/conda/feedstock_root/build_artifacts/truststore_1694154605758/work
tzdata==2023.3
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1697720414277/work
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.5
widgetsnbextension==4.0.9
win-inet-pton @ file:///D:/bld/win_inet_pton_1667051142467/work
wslink==1.12.4
yarl==1.9.2
zstandard==0.21.0

Samuelopez-ansys, 2023-12-12:
@mahesh-dhanekula Please share the project, or create a dummy project with PyAEDT to reproduce the issue.
Thanks

mahesh-dhanekula, 2024-02-01:
@gmalinve @Samuelopez-ansys 
Somehow I don't see your mentions in my inbox.
In zip file you can find the project.
[terminals.zip](https://github.com/ansys/pyaedt/files/14125239/terminals.zip)

gmalinve, 2024-02-01:
Hi @mahesh-dhanekula,
Your issue should be fixed in #4175.
Thanks for feedback!
Giulia

closed 2024-02-01 by gmalinve