ansys/pyaedt #3962: Problem using RMxprt User Defined Primitive in Maxwell 3D

DaveTwyman, 2023-12-11:
### Discussed in https://github.com/ansys/pyaedt/discussions/3941

<div type='discussions-op-text'>

<sup>Originally posted by **Porta-L** December  4, 2023</sup>
This describes a problem encountered when creating a coil in Maxwell 3D using the LapCoil User Defined Primitive from RMxprt.
When creating a LapCoil object, the InfoCoil parameter is used to define the type of object to be created according to:

- winding => InfoCoil = 0
- coil => InfoCoil = 1
- terminal => InfoCoil = 2
- etc.

To create a single coil, I set the _InfoCoil_ parameter to _1_ in the _udp_parameters_list_ of the following function:
_Modeler3D.create_udp(udp_dll_name, udp_parameters_list, upd_library='syslib', name=None)_

The object is then created and visible in the GUI, but appears as a complete winding and not a coil. When you open the parameter list from the GUI, you can see that the InfoCoil parameter has not been correctly transmitted from PyAEDT and does not have the correct value.

You can find a simple example in the attached script where the goal is to create a coil and a terminal.
[test_udp.zip](https://github.com/ansys/pyaedt/files/13547642/test_udp.zip)

Here are more information about the function used:
[Ansys Maxwell Doc](https://ansyshelp.ansys.com/account/secured?returnurl=/Views/Secured/Electronics/v232/en/home.htm%23../Subsystems/Maxwell/Content/UDPsforSlotCoresandCoils.html)
[PyAEDT Doc](https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.modeler.modeler3d.Modeler3D.create_udp.html)

Any support is welcome, thank you in advance.</div>

Samuelopez-ansys, 2023-12-12:
@DaveTwyman Did you debug the code to check which parameters are passed to the method?

Please record a script to compare the parameters you are passing.

DaveTwyman, 2023-12-12:
> @DaveTwyman Did you debug the code to check which parameters are passed to the method?
> 
> Please record a script to compare the parameters you are passing.

Thanks for your help and as discussed, looks likely issue is in AEDT UI side, bug 957132 raised.

DaveTwyman, 2024-01-31:
Whilst this github issue (of type bug) is now 'closed as completed' on the PyAEDT side, the bug against UDP AEDT behavior is still open and will be resolved in a future release of AEDT.

DaveTwyman, 2024-03-04:
@Samuelopez-ansys @maxcapodi78 Dev cam back with an update on this issue to tfs 957132. copied below

_"This bug is due to PyAEDT didn't set the UDP version on their side. Since the version is not set, we use the default version 1.0 which is very old and doesn't support InfoCoil parameter. And this is why this parameter is always zero. In order to fix this bug, we add a small feature on our side, PyAEDT team can send version -1.0 to AEDT to indicate the latest version."_

This looks like it needs a bit more verification of other UDPs created from PyAEDT.
It could be that we are getting lucky with the features passed using all UDPs requested from PyAEDT in other cases they exist in version 1 of the UDP so behavior was as expected. But this means if a UDP is called from PyAEDT with a feature that was not present in v1 of the UDP, we get errant behaviour.

Any ideas on how/when we could implement this UDP=-1 version flag to work with the AEDT version in the future it is implemented in?

temudschin, 2024-06-21:
Is there a workaround available?