ansys/pyaedt #3968: Error in is_solved property of setup for DCIR setup in HFSS 3D Layout

nnetake, 2023-12-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

If I try to check if the solution is available for the DCIR setup object, then it gives an error that none object is not iterable.

    def check_if_solution_available(setup_name):
        """
        Check the solution is available for the given setup name
        :return:
        True if the solution is available
        """
        setup_object = self.hfssapp.get_setup(setup_name)
        return setup_object.is_solved

### Steps To Reproduce

Use above described method to check for solution.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
cffi==1.16.0     
clr-loader==0.2.6
plumbum==1.8.2   
psutil==5.9.6    
pyaedt==0.7.7    
pycparser==2.21
pywin32==306
rpyc==5.3.1

gmalinve, 2023-12-12:
Hi @nnetake,

Could you provide more info? because I was successfully able to get the desired setup in an HFSS3DL design.

Thanks,

Giulia

gmalinve, 2023-12-12:
We would need a new elif case such as: 
`elif self.props.get("SolveSetupType", "HFSS") == "SIwave":`
in order to cover the Siwave DC IR setup case.

closed 2023-12-13 by gmalinve