ansys/pyaedt #3969: Help page search is not working

FabianSchwartau, 2023-12-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The search on the help page
https://aedt.docs.pyansys.com/version/stable/search.html
never returns any results.
I tried multiple browsers/OS.

### Steps To Reproduce

Go to https://aedt.docs.pyansys.com/version/stable/search.html and search for something, hit enter, you get an empty page.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

What do python packages that I have installed to do with the web page?

Samuelopez-ansys, 2023-12-12:
@FabianSchwartau Sometimes it is a bit slow, could you type a word (like box) and wait? 
[image]

FabianSchwartau, 2023-12-12:
Ok, that works.
I never waited for anything, I just hit enter. This just brings me to a blank page with the search box and nothing in it.
It is also quite annoying that Ctrl+Backspace does not work in the search edit box.

Samuelopez-ansys, 2023-12-12:
@FabianSchwartau Sorry for the confusion, but it follows PyAnsys search bar, we can not change it only for PyAEDT.

What do you mean with Ctrl+Backspace? Not sure what this combination should do.

FabianSchwartau, 2023-12-12:
I don't understand what you mean by "it follows PyAnsys search bar" - so it is out of "your" control?
I can live with the workaround now, but its a bit confusing and not clear. I have been working around this now for days before posting here and I am sure there are other people who are as confused as I was. Together with the fact that the help is not indexed by google, made it quite hard as a beginner to find anything.

How can you live without Ctrl+backspace? :P
It deletes the complete word left of the cursor.

Samuelopez-ansys, 2023-12-12:
@FabianSchwartau PyAEDT is part of the PyAnsys initiative, so the documentation  style is part of the PyAnsys organization.

I will add two members to check if this performance could be improved :)

Samuelopez-ansys, 2023-12-12:
@MaxJPRey @SMoraisAnsys

SMoraisAnsys, 2023-12-19:
@FabianSchwartau Hi, thanks for the feed back ! I understand your remarks about the documentation. I'll do my best to free up some time to have a look at it but I can't give a date. Btw, I wasn't adapted to CTR + Backspace but I love it :D

closed 2024-02-22 by Samuelopez-ansys