ansys/pyaedt #3993: Bug located in pyaedt.modeler.modelerpcb.py class

dcrawforAtAnsys, 2023-12-17:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

``oEditor.SetActiveUnits`` in the native API fails.

### Steps To Reproduce

The ``Modeler3DLayout.model_units`` setter fails with the message:
``pyaedt.generic.general_methods.GrpcApiError: Failed to execute grpc AEDT command: GetActiveUnits``

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

All required packages are installed.

dcrawforAtAnsys, 2023-12-17:
@ring630 : ``The oEditor.GetActiveUnits()`` does not appear to be used anywhere else right now. Perhaps the ``model_units`` should be saved as a local property with the class since this does not appear to be supported by the 3D Layout native API.

dcrawforAtAnsys, 2023-12-17:
There was a syntax error in ``SetActiveUnits()``

ring630, 2024-01-08:
@dcrawforAtAnsys oEditor.SetActiveUnits is working in my test. The error you attached is grpc related, maybe it is fixed by another PR. model_units is effective when creating objects in layout editor. When the model_units is changed from "mm" to "um", the default unit is changed. Below example code is as below. However, the change of model_units is not visible in GUI. Please let me know if the example code works on your side.

I saw you fixed the syntax error. Thanks! 

`
import pyaedt

non_graphical = False
NewThread = True

desktopVersion = "2024.1"

d = pyaedt.launch_desktop(desktopVersion, non_graphical, NewThread)
h3d = pyaedt.Hfss3dLayout()
assert h3d.modeler.model_units == "mm"
h3d.modeler.create_rectangle("Outline", [0,0], [1,1])

h3d.modeler.model_units = "um"
assert h3d.modeler.model_units == "um"
h3d.modeler.create_rectangle("Outline", [0,0], [1,1])

h3d.close_project()
`

closed 2024-02-09 by ring630