ansys/pyaedt #4001: Bug located in ...get_object_material_properties

tungnguyen1003, 2023-12-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Test command (did not work)
hfss.get_object_material_properties()

Found issue (worked)
hfss.materials
hfss.get_object_material_properties()

### Steps To Reproduce

As in description

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

pyaedt latest version

SMoraisAnsys, 2023-12-21:
Hi @tungnguyen1003, thanks for the feedback. To make it easier for us to identify and resolve the bug, please answer the following question.

What do you mean by:

Test command (did not work)
hfss.get_object_material_properties()

Found issue (worked)
hfss.materials
hfss.get_object_material_properties()

Does the second part is a workaround to make `get_object_material_properties` work ?

Could you provide us with a simple reproducer to show the bug ?

SMoraisAnsys, 2024-01-05:
@tungnguyen1003 Hi and happy new year, could you please answer the previous comment ? This would help us to provide you with the right answer.

tungnguyen1003, 2024-01-07:
Hi

Sorry for my late reply.
Also, my sincere apologies for the lack of explanation on the issue.

We thought that the issue was not with the command itself, but with the explanation on how to make it work, described in the below URL.
https://aedt.docs.pyansys.com/version/0.7/API/_autosummary/pyaedt.hfss.Hfss.get_object_material_properties.html

As you can see in the screenshot below, at first, we opened a simple test design and tested the command, but errors popped up. Then we had to look deeply in the code for this command and got it up and running by adding “app.materials” before calling this command.

We assumed that the project materials had to be loaded before this generic object-materials command can be used. It took me more than 30 minutes to figure out this simple process, so I thought a little more explanation would be very helpful. That s all.

***@***.***

Thank you for reaching out.

Happy New Year
**------------------------------------------------------------------------**
Tung Nguyen (トゥン　グェン), Ph.D. / Lead Tech. Support Engineer - RF Modeling, Electronics BU, ACE Japan  / Technical Division
***@***.***<https://www.ansys.com/>
アンシス・ジャパン株式会社 / 〒160-0023 東京都新宿区西新宿6-10-1 / 日土地西新宿ビル18F
tel: 03-5324-7357 / fax : 03-5324-7308
***@***.******@***.***>  /  www.ansys.com<https://nam10.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.ansys.com%2F&data=04%7C01%7Ctung.nguyen%40ansys.com%7Ca1ad9d773fe242f6e77508d896952fa0%7C34c6ce6715b84eff80e952da8be89706%7C0%7C0%7C637424914820974195%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=iwg6Z5A4BHi10m%2FuYLRwJM1qV7WHY%2BbZ3WgX3RVNQZE%3D&reserved=0>

[External Sender]

SMoraisAnsys, 2024-01-16:
@tungnguyen1003 Thanks for you answer and sorry for the delay. I'm not really sure to understand what you are trying to do and how. Could you please provide me with extra details ?

On my side I'm able to use the method `get_object_material_properties` with the following code. Can you tell me what type of extra information you needed ? 

```python
from pyaedt import Hfss
app = Hfss(projectname="my_project", designname="my_design", specified_version="23.2")
app.insert_design("ObjMat")
ms = app.modeler.create_box([4, 5, 0], [1, 100, 0.2], name="MS1", matname="copper")
props = app.get_object_material_properties("MS1", "conductivity")
print(props) # This prints {'MS1': {'conductivity': '58000000'}}
```

Extra note:
> As you can see in the screenshot below

I'm not able to see any screenshot in your comment

tungnguyen1003, 2024-01-17:
Hi

Thanks for your answer.

Could you try the following and see if you are able to see the error.

1.Open AEDT 2023.2 and load your project “my_project”
2.Acess your existing design from PyAEDT and check the command,
from pyaedt import Hfss
app = Hfss(projectname=’my_project’, designname=’my_design’,specified_version=’2023.2’,new_desktop_session=False,aedt_process_id=12832)
# aedt_process_id should be modified accordingly, check the Task Manager
props = app.get_object_material_properties("MS1", "conductivity")

Now I think you can see some pyaedt ERROR

The code you gave me worked, but the thing is when you added the MS1 object, there is a line saying [pyaedt INFO: Successfully loaded project materials !]
Without this INFO, app.get_object_material_properties will produce Error.
I add a line
App.materials
before calling app.get_object_material_properties, and it worked too.

So the issue is not with the command but you have to somehow LOADED PROJECT MATERIALS before using the command.

Best regards,
**------------------------------------------------------------------------**
Tung Nguyen (トゥン　グェン), Ph.D. / Lead Tech. Support Engineer - RF Modeling, Electronics BU, ACE Japan  / Technical Division
***@***.***<https://www.ansys.com/>
アンシス・ジャパン株式会社 / 〒160-0023 東京都新宿区西新宿6-10-1 / 日土地西新宿ビル18F
tel: 03-5324-7357 / fax : 03-5324-7308
***@***.******@***.***>  /  www.ansys.com<https://nam10.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.ansys.com%2F&data=04%7C01%7Ctung.nguyen%40ansys.com%7Ca1ad9d773fe242f6e77508d896952fa0%7C34c6ce6715b84eff80e952da8be89706%7C0%7C0%7C637424914820974195%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=iwg6Z5A4BHi10m%2FuYLRwJM1qV7WHY%2BbZ3WgX3RVNQZE%3D&reserved=0>

[External Sender]

SMoraisAnsys, 2024-01-19:
@tungnguyen1003 Thanks for the extra information, I'll take some time to look into that today 👍

SMoraisAnsys, 2024-01-20:
@tungnguyen1003 I'm able to reproduce the bug but I think the problem comes from the way you try to interact through pyaedt.
@Samuelopez-ansys Correct me if I'm wrong but in the previous code provided (see below) it is normal that an exception is raised as there is no object named "MS1" in the project. Since the resulting error might be confusing, what about adding some logger error saying that the object was not found ? If my understanding of the problem is wrong, please correct me. If not, I'll add the logger error to clarify this issue.

```python
from pyaedt import Hfss

process_id = 12832 # See your own process id through task managed
app = Hfss(projectname=’my_project’, designname=’my_design’, specified_version=’2023.2’, new_desktop_session=False, aedt_process_id=12832)

props = app.get_object_material_properties("MS1", "conductivity")
```

@tungnguyen1003 Again, thanks for the feedback !

closed 2024-01-30 by SMoraisAnsys