ansys/pyaedt #4010: Allow primitive properties to be passed as keyword arguments.

dcrawforAtAnsys, 2023-12-27:
### Description of the feature

Properties such as ``color`` and ``transparency`` should be passed as optional keywords when creating primitives.

### Steps for implementing the feature

```python
def create_rectangle(self, csPlane, position, dimension_list, name=None, matname=None, is_covered=True, **kwargs):

# code

if "color" in kwargs.keys():
   self.color = kwargs["color"]
```

### Useful links and references

I think this is a reasonable way to simplify syntax when a new primitive should have a specific color, transparency, etc. Rather than creating the primitive and then assigning a value to the property.

Samuelopez-ansys, 2023-12-27:
@dcrawforAtAnsys I like it

dcrawforAtAnsys, 2023-12-30:
To be addressed in PR #4019.

gmalinve, 2024-01-03:
Closed by #4019.

closed 2024-01-03 by gmalinve