ansys/pyaedt #4012: Redirect messages from the desktop to the logger when in non-graphical mode.

dcrawforAtAnsys, 2023-12-28:
### Description of the feature

Failures are caught by the ``_function_handler_wrapper()`` in the file ``.\generic\general_methods.py``. Debugging would be much easier if the messages from the native API that are posted to the desktop when the AEDT UI is visible could also be available from this wrapper method.

### Steps for implementing the feature

This may require an enhancement to the native API.

dcrawforAtAnsys, 2023-12-28:
This should be handled in ``_log_method()`` in the file ``general_methods.py``.

maxcapodi78, 2024-01-03:
I'm currently refactoring the logger to support the history of messages from pyaedt and from Desktop.

closed 2024-10-01 by Samuelopez-ansys