ansys/pyaedt #4017: ZStart must be 0 in Maxwell 2D

dcrawforAtAnsys, 2023-12-29:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The property ``transparency`` for 2D primitives does not exist if the ``ZStart`` value passed to the native API is non-zero. This can be seen in the unit test ``test_04_create_rectangle`` when trying to modify transparency after creating a rectangle. 

### Steps To Reproduce

Run the following script in the native API. The rectangle transparency cannot be modified.
```python
import ScriptEnv
ScriptEnv.Initialize("Ansoft.ElectronicsDesktop")
oProject = oDesktop.GetActiveProject()
oDesign = oProject.GetActiveDesign()
oEditor = oDesign.SetActiveEditor("3D Modeler")
oEditor.CreateRectangle(
	['NAME:RectangleParameters', 
    'IsCovered:=', True, 
    'XStart:=', '10mm', 
    'YStart:=', '-2mm', 
    'ZStart:=', '1mm', 
    'Width:=', '3mm', 
    'Height:=', '10mm', 
    'WhichAxis:=', 'Z'], 
	['NAME:Attributes', 
    'Name:=', 'MyRectangle', 
    'Flags:=', '', 
    'Color:=', '(242  140  102)', 
    'Transparency:=', 0.2, 
    'PartCoordinateSystem:=', 'Global', 
    'SolveInside:=', True, 
    'MaterialValue:=', '"copper"', 
    'UDMId:=', '', 
    'SurfaceMaterialValue:=', '"Steel-oxidised-surface"', 
    'ShellElement:=', False, 
    'ShellElementThickness:=', '0mm', 
    'IsMaterialEditable:=', True, 
    'UseMaterialAppearance:=', False, 
    'IsLightweight:=', False
    ])
```
change ``ZStart`` to ``0mm`` and all is well again.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

all dependences are installed.

dcrawforAtAnsys, 2023-12-30:
There should be a message written to the log when an attempt to update transparency fails.

gmalinve, 2024-01-02:
@dcrawforAtAnsys see my comment in #4019 -> I'd suggest to add an additional check in _default_object_attributes for transparency.

Samuelopez-ansys, 2024-02-14:
@dcrawforAtAnsys Is this feature already impleted right? Please close the issue if it is.

dcrawforAtAnsys, 2024-03-27:
This is an issue in the native API but should not come up when using PyAEDT.

closed 2024-03-27 by dcrawforAtAnsys