ansys/pyaedt #4034: Bug located in ...pyaedt\edb_core\components

Acaptian, 2024-01-08:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Cannot use this “components”module to create a coaxial port

### Steps To Reproduce

edb = pyaedt.Edb(edbpath="path")
nets = ["E_MDC","E_MDIO"]
edb.components.create_port_on_component(component="U301"  ,net_list=ntets,  port_type=SourceType.CoaxPort, do_pingroup=True, reference_net="GND", port_name=None
    )
edb.save_edb()
edb.close_edb()
#the bug is "SourceType” is not define

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

pyaedt
os
time
pandas
openpyxl

svandenb-dev, 2024-01-10:
Hi @Acaptian,

I think you are missing the SourceType import.

here is a code snippet which will also work on your machine

import pyaedt
from pyaedt.generic.constants import SourceType

project_path = pyaedt.generate_unique_folder_name()
target_aedb = pyaedt.downloads.download_file('edb/ANSYS-HSD_V1.aedb', destination=project_path)
print("Project folder will be", target_aedb)

aedt_version = "2023.2"
edb = pyaedt.Edb(edbpath=target_aedb, edbversion=aedt_version)

edb.components.create_port_on_component(component="U1", net_list=["PCIe_Gen4_RX0_P",
                                                                  "PCIe_Gen4_RX0_N",
                                                                  "PCIe_Gen4_RX1_P",
                                                                  "PCIe_Gen4_RX1_N"],
                                        reference_net=["GND"],
                                        port_type=SourceType.CoaxPort)
edb.save_edb()
edb.close_edb()
hfss = pyaedt.Hfss3dLayout(projectname=target_aedb, specified_version=aedt_version)
hfss.release_desktop(False, False)

Also I can advise you to have a look at this example
https://aedt.docs.pyansys.com/version/stable/examples/00-EDB/09_Configuration.html#sphx-glr-examples-00-edb-09-configuration-py

It is using the SImulatiomConfiguration which allows you to create an end to end Signal and Power integrity automated workflow. It is very easy to use and you can also import/export config files as json.

Thanks

Acaptian, 2024-01-11:
Hi svandenb-dev
Thanks for your reply, I will check my code based on your suggestions and examples.

Acaptian, 2024-01-11:
Hi svandenb-dev,
Thanks for your reply, I will check my code based on your suggestions and examples.

ring630, 2024-01-15:
@Acaptian Please submit a new issue on xml import.

ring630, 2024-01-16:
@Acaptian for xml import discussion, refer to issue 4082
https://github.com/ansys/pyaedt/issues/4082#issue-2083256136

closed 2024-01-16 by ring630