ansys/pyaedt #4040: Bug located in pyaedt\edb_core\components

Acaptian, 2024-01-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

What parameters should I fill in for the "_is_top_component(self, cmp)" function, 
when I get it "Ansys.Ansoft.Edb.Cell.Hierarchy.Component" object , It returns some errors
Can you give me an example to do this? 
And how to call the " self._pedb.component"?
Thank you.

### Steps To Reproduce

import os 
import time
import pyaedt
import openpyxl
import pandas as pd
edb = pyaedt.Edb(edbpath='D:\PCB\825501014_RDA_Layout003.aedb')
comp_by_name = edb.components.get_component_by_name("U1301") 
comp_is_top = edb.components._is_top_component(cmp=comp_by_name) 
print (comp_is_top)
edb.save_edb()
edb.close_edb()
#PyAEDT ERROR: TYPE ERROR on _is_top_component
#PyAEDT ERROR: '<Layers>d__18' object is not subscriptable
#PyAEDT ERROR:File "D:\python310\Lib\site-packages\pyaedt\modeler\cad\component_test.py", line 10, in <module>
#PyAEDT ERROR:comp_is_top = edb.components._is_top_component(cmp=comp_by_name)
#PyAEDT ERROR:File "D:\python310\lib\site-packages\pyaedt\edb_core\components.py", line 1352, in _is_top_component
#PyAEDT ERROR: Method arguments:PyAEDT ERROR:cmp = Ansys.Ansoft.Edb.Cell.Hierarchy.Component
#False

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

import os 
import time
import pyaedt
import openpyxl
import pandas as pd

svandenb-dev, 2024-01-09:
Hi @Acaptian,
thank you for your feedbacks. actually you are trying to use an internal command as it is starting with "_" so this command won't be documented as we don't expect users to use it. However this is working. 

What it happening in your case is the fact that the command get_component_by_name returns and internal EDB component instance not PyAET one. 

what you can do id using dictionaries like this :

aedt_version = "2023.2"
edb = pyaedt.Edb(edbpath=target_aedb, edbversion=aedt_version)
component_instance = edb.components.instances["U1301"]
test = edb.components._is_top_component(component_instance)

edb.component.instances is a dictionary containing components refdes as key and EDBComponent object that the internal command _is_top_component is expecting. 

I will add a new property in EDBComponent class to return the results you are testing so that you won't need to use that internal command anymore. 

It should be included in the next version by the end of the week so please upgrade your PyAEDT package if you can.

svandenb-dev, 2024-01-09:
Just create a PR 
https://github.com/ansys/pyaedt/pull/4052

for next release you will be able to access the is_top property

edbapp = Edb(target_path, desktop_version)
edbapp.components.instances["U1"].is_top

Acaptian, 2024-01-10:
Hi svandenb-dev,
Thank you for your reply, but here's a new error.
PyAEDT ERROR: ATTRIBUTE ERROR on _is_top_component
PyAEDT ERROR: 'EDBComponent' object has no attribute 'GetLayout'
Maybe there's something wrong with the properties of this 'EDBComponent' object or CLASS
Expect it to be fixed in the next release，thanks.

svandenb-dev, 2024-01-10:
@Acaptian 
with the new release on Friday please use the new property
edbapp = Edb(target_path, desktop_version)
edbapp.components.instances["U1"].is_top_mounted
that will work as you expect. Thanks

closed 2024-01-10 by svandenb-dev