ansys/pyaedt #4042: In convert_to_3d_microvias adding ability for user to specify upper diameter and lower diameter instead of Angle of laser penetration

amichel0205, 2024-01-09:
### Description of the feature

User will be able to use either:

- Angle of laser penetration
or
 Upper and lower microvia diameter

### Steps for implementing the feature

- add parameters upper_diameter and lower_diameter
- Check upper_diameter is greater or equal to lower_diameter
- Check upper_diameter is smaller or equal to upper padstack
- 
- If user specify both angle and upper and lower diameter, upper and lower parameter take precedence

Samuelopez-ansys, 2024-01-10:
@amichel0205 Is this Issue related to the closed Pull request?

svandenb-dev, 2024-02-05:
reopening upon users request as it is not yet finished