ansys/pyaedt #4045: stackup xml issue

louseph, 2024-01-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When conductivity information is missing in stackup.xml, pyaedt fails to import dielectric materials.

### Steps To Reproduce

stackup_file = 'stackup.xml'
edb = Edb()
edb.stackup.load(stackup_file)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

absl-py==2.0.0
aiohttp==3.8.5
aiosignal==1.3.1
alpha-vantage==2.3.1
ansi2html==1.8.0
appdirs==1.4.4
asttokens==2.2.1
astunparse==1.6.3
async-timeout==4.0.3
attrs==23.1.0
backcall==0.2.0
bcrypt==4.0.1
beautifulsoup4==4.12.2
blinker==1.7.0
cachetools==5.3.1
certifi==2023.7.22
cffi==1.15.1
chardet==3.0.4
charset-normalizer==3.2.0
click==8.1.7
clr-loader==0.2.6
colorama==0.4.6
colour==0.1.5
comm==0.1.3
contourpy==1.1.0
cryptography==41.0.3
cycler==0.11.0
dash==2.14.1
dash-core-components==2.0.0
dash-html-components==2.0.0
dash-table==5.0.0
dateutils==0.6.12
debugpy==1.6.7
decorator==5.1.1
descartes==1.1.0
EbookLib==0.18
executing==1.2.0
Flask==3.0.0
flatbuffers==23.5.26
fonttools==4.41.1
fpdf==1.7.2
frozendict==2.3.8
frozenlist==1.4.0
gast==0.5.4
google-auth==2.23.3
google-auth-oauthlib==1.0.0
google-pasta==0.2.0
googletrans==3.0.0
grpcio==1.59.0
h11==0.9.0
h2==3.2.0
h5py==3.10.0
hpack==3.0.0
hstspreload==2023.1.1
html5lib==1.1
httpcore==0.9.1
httpx==0.13.3
hyperframe==5.2.0
idna==2.10
imageio==2.31.5
importlib-metadata==6.8.0
install==1.3.5
ipykernel==6.25.0
ipython==8.14.0
itsdangerous==2.1.2
jedi==0.19.0
Jinja2==3.1.2
jupyter_client==8.3.0
jupyter_core==5.3.1
keras==2.14.0
kiwisolver==1.4.4
lazy_loader==0.3
Levenshtein==0.22.0
libclang==16.0.6
lxml==4.9.3
Markdown==3.5
MarkupSafe==2.1.3
matplotlib==3.7.2
matplotlib-inline==0.1.6
ml-dtypes==0.2.0
MouseInfo==0.1.3
multidict==6.0.4
multitasking==0.0.11
nest-asyncio==1.5.7
networkx==3.2
nsepy==0.8
nsetools==1.0.11
numpy==1.25.1
oauthlib==3.2.2
opencv-python==4.8.0.76
opt-einsum==3.3.0
packaging==23.1
pandas==2.1.1
paramiko==3.3.1
parso==0.8.3
peewee==3.17.0
pickleshare==0.7.5
Pillow==10.0.0
platformdirs==3.10.0
plotly==5.18.0
plumbum==1.8.2
pooch==1.7.0
prompt-toolkit==3.0.39
protobuf==4.24.4
psutil==5.9.5
pure-eval==0.2.2
pyaedt==0.6.93
pyasn1==0.5.0
pyasn1-modules==0.3.0
PyAutoGUI==0.9.54
pycparser==2.21
pyDOE==0.3.8
PyGetWindow==0.0.9
Pygments==2.15.1
PyMsgBox==1.0.9
PyNaCl==1.5.0
pyparsing==3.0.9
PyPDF2==3.0.1
pyperclip==1.8.2
PyQt5==5.15.9
PyQt5-Qt5==5.15.2
PyQt5-sip==12.12.2
PyRect==0.2.0
PyScreeze==0.1.30
PySide2==5.15.2.1
pytesseract==0.3.10
python-dateutil==2.8.2
python-docx==1.1.0
pythonnet==3.0.2
pytweening==1.0.7
pytz==2023.3.post1
pyvista==0.42.0
pywin32==306
pyzmq==25.1.0
rapidfuzz==3.3.1
reportlab==4.0.7
requests==2.31.0
requests-oauthlib==1.3.1
retrying==1.3.4
rfc3986==1.5.0
rpyc==5.3.1
rsa==4.9
Rtree==1.0.1
scikit-image==0.22.0
scikit-rf==0.29.1
scipy==1.11.1
scooby==0.7.2
shapely==2.0.1
shiboken2==5.15.2.1
six==1.16.0
sniffio==1.3.0
soupsieve==2.5
stack-data==0.6.2
style==1.1.0
tenacity==8.2.3
tensorboard==2.14.1
tensorboard-data-server==0.7.1
tensorflow==2.14.0
tensorflow-estimator==2.14.0
tensorflow-intel==2.14.0
tensorflow-io-gcs-filesystem==0.31.0
termcolor==2.3.0
tifffile==2023.9.26
tornado==6.3.2
traitlets==5.9.0
trame==3.2.4
trame-client==2.11.3
trame-server==2.11.7
trame-vtk==2.5.8
typing_extensions==4.8.0
tzdata==2023.3
update==0.0.1
urllib3==2.0.4
vtk==9.2.6
wcwidth==0.2.6
webencodings==0.5.1
Werkzeug==3.0.0
wrapt==1.14.1
wslink==1.11.4
yarl==1.9.2
yfinance==0.2.31
zipp==3.17.0

Samuelopez-ansys, 2024-01-10:
@louseph Please share the xml file to reproduce the issue.

louseph, 2024-01-10:
Please find the xml here.

Thanks & Regards,
Lindo Ouseph / Senior Application Engineer

[cid:61c0d40a-d0af-4432-bedf-69dfd57905c0]

Bangalore / India

***@***.***  /  www.ansys.com<http://www.ansys.com/>

[External Sender]

Samuelopez-ansys, 2024-01-10:
Hi @louseph, I think you attached the file from your email. To attach a file, you need to open GitHub website.

@ring630 @svandenb-dev

louseph, 2024-01-10:
[cc.zip](https://github.com/ansys/pyaedt/files/13886339/cc.zip)
Earlier i tried to attach xml, which failed. Thats why I shared through email.
Please unzip the file here for xml file.

Acaptian, 2024-01-10:
Hi Samuelopez-ansys,s, I have a similar question.
h3d = pyaedt.Hfss3dLayout()
h3d_import_cad(cad_path="xxx.brd",cad_format="brd", aedb_path=None, xml_path="D:\XML\output1.xml")
When I used the above code to import the stackup xml of the project, but it didn't work.

ring630, 2024-01-15:
> [cc.zip](https://github.com/ansys/pyaedt/files/13886339/cc.zip) Earlier i tried to attach xml, which failed. Thats why I shared through email. Please unzip the file here for xml file.

@louseph I got below error when importing the file you attached. Loss tangent is required for a dielectric material.
`PyAEDT ERROR: DielectricLossTangent`

ring630, 2024-01-15:
> Hi Samuelopez-ansys, I have a similar question. h3d = pyaedt.Hfss3dLayout() h3d._import_cad(cad_path="xxx.brd",cad_format="brd", aedb_path=None, xml_path="D:\XML\output1.xml") When I used the above code to import the stackup xml of the project, but it didn't work. The zip file is the stackup xml file. [stackup.zip](https://github.com/ansys/pyaedt/files/13911070/stackup.zip)

@Acaptian I can import your xml into an empty layout. Please attach the error message you get. Can you import this xml file from UI?

Acaptian, 2024-01-16:
[ring630](https://github.com/ring630)
Yes ，I can import this xml manually on the UI.
But when I try to use “_import_cad(xml_path=...)” to import a new stackup to replace the old stack, It can‘t work .
And there are no error messages, It runs normally, but has no effect.

ring630, 2024-02-09:
@Acaptian is it the same qusetion you asked in another issue? 
https://github.com/ansys/pyaedt/issues/4082#issue-2083256136