ansys/pyaedt #4082: hfss3dlayout import stackup in xml format failure

ring630, 2024-01-16:
Hi Samuelopez-ansys, I have a similar question.
h3d = pyaedt.Hfss3dLayout()
h3d._import_cad(cad_path="xxx.brd",cad_format="brd", aedb_path=None, xml_path="D:\XML\output1.xml")
When I used the above code to import the stackup xml of the project, but it didn't work.
The zip file is the stackup xml file.
[stackup.zip](https://github.com/ansys/pyaedt/files/13911070/stackup.zip)

_Originally posted by @Acaptian in https://github.com/ansys/pyaedt/issues/4045#issuecomment-1884695592_

ring630, 2024-01-16:
@Acaptian I created a new issue for your problem.

ring630, 2024-01-16:
@Acaptian, to import a brd file here is the correct procedure.

```
from pyaedt import Edb
from pyaedt import Hfss3dLayout

edbapp = Edb("layout.brd") # Translate brd into EDB
edbapp.stackup.load(r"stackup.xml") # Import stackup into Edb

# Save and Close Edb
edbapp.save_edb_as("layout.aedb") 
edbapp.close_edb()

# import Edb into 3D Layout
h3d = Hfss3dLayout("layout.aedb", specified_version="2023.2", new_desktop_session=True)

```

ring630, 2024-01-16:
@Acaptian if the problem persists, please attach your complete code which I can run on my machine.

Acaptian, 2024-01-17:
@[ring630](https://github.com/ring630)
Yes, it can correctly run,when I use the above code.
Thank you!

Acaptian, 2024-01-17:
Hi @[ring630](https://github.com/ring630)
I hope you can check this:

from pyaedt import Edb
from pyaedt import Hfss3dLayout

edbapp = Edb("XXX.aedb") # 
edbapp.stackup.load(r"stackup.xml") # Import stackup into Edb

# Save and Close Edb
edbapp.save_edb_as("XXX.aedb") 
edbapp.close_edb()

# import Edb into 3D Layout
h3d = Hfss3dLayout("layout.aedb", specified_version="2023.2", non_graphical,new_desktop_session=True)
Whether edb can be imported only when UI is started,because this is a error about above code.(using a exist aedb file)
error:Measures dosen't exist in xml

ring630, 2024-01-17:
@Acaptian this load measured is tested internally on our example board. To investigate the problem you have, I need below two files
1, the default stackup from the layout. You can export it from UI.
2, stackup.xml you want to import.

Acaptian, 2024-01-17:
Hi @[ring630](https://github.com/ring630)
Here is my stackup about the test project.
[Uploading XML.zip…]()

ring630, 2024-01-17:
@Acaptian COPPER_1 has no definition in UI_export.
<Material Name="COPPER_1">
with above line removed from UI_export.xml, I can execute below code without any problem. Please let me know how to reproduce your problem.

```
edbapp = Edb()

edbapp.stackup.load("UI_export.xml")
edbapp.stackup.load("stackup.xml")
```

Acaptian, 2024-01-18:
Hi [ring630](https://github.com/ring630)
Maybe you can reproduce  this problem in this condition:
1.Just have a exist xxx.aedt project with its xxx.aedb file 
2.Use a new xml stackup file to replace the old stackup 
3.Try this code to excute the above operation:

from pyaedt import Edb
from pyaedt import Hfss3dLayout

edbapp = Edb("XXX.aedb")
edbapp.stackup.load(r"stackup.xml")

edbapp.save_edb_as("XXX.aedb")
edbapp.close_edb()

h3d = Hfss3dLayout("XXX.aedb", specified_version="2023.2", non_graphical=True,new_desktop_session=True)

Eventually ,there will be a error : something dosen't exist in xml.

ring630, 2024-01-18:
@Acaptian can you attach the full error message?

Acaptian, 2024-02-05:
Hi [ring630](https://github.com/ring630)
here is the full error message:
PyAEDT INFO: Edb version 2023.2
PyAEDT INFO: Stdout is enabled
PyAEDT INFO: Logger is initialized in EDB.PyAEDT INFO: pyaedt ve.7.11
PyAEDT INFO: Python version 3.10.7 
PyAEDT INFO: Translation correctly completed
PyAEDT INFO: Database 825502207_HVAC_Layout.aedb Opened in None
PyAEDT INFO: Cell 825502207_HVAC_Layout Opened
PyAEDT INFO: Builder was initialized.
PyAEDT INFO: EDB D:/PCB\825502207_HVAC_Layout.aedb was created correctly from rd file.
PyAEDT INFO: EDB initialized.
PyAEDT ERROR: GND doesn't exist in xml
PyAEDT INFO: EDB file save time: 1.05ms
Please check the " edbapp.stackup.load("stackup.xml")"code .Because I can import my stackup.xml in HFSS3Dlayout UI,but when I use the script code to import xml,it is always wrong .

ring630, 2024-02-05:
@Acaptian Do you have a layer named GND in stackup? Do you want to remove it afer xml import?

Acaptian, 2024-02-06:
Hi [ring630](https://github.com/ring630)
Yes, I use different names to distinguish different layers of GND, but it does not affect the real “GND” of network.It's just a name of layer.

closed 2024-02-09 by ring630