ansys/pyaedt #4093: Maxwell3D Setup ElectricTransient

WynAAAAA, 2024-01-17:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I tried creating a Setup in Maxwell3D for the Solution Type "Electric Transient". It never worked, saying that the setup is not compatible in current solution type.

If you exchange both "ElectricTransient" with "ElectroDCConduction" it works normal, so i assume it to be a bug. 

### Steps To Reproduce

```
import pyaedt

m3d = pyaedt.Maxwell3d(
    projectname= pyaedt.generate_unique_project_name(),
    specified_version="2023.2",
    new_desktop_session=False,
    non_graphical=False,
    student_version=True,
    close_on_exit=True,
    solution_type="ElectricTransient",       #"ElectricTransient",ElectroDCConduction
    designname="Ansys_Test"
)

#Model
elec1 = m3d.modeler.create_box([-2, -2.1, -2], [4, 0.6, 4], name="Elektrode1", matname="steel_stainless")
elec2 = m3d.modeler.create_box([-2, 1.5, -2],[4, 0.6, 4], name="Elektrode2", matname="steel_stainless")

region = m3d.modeler.create_region(300, True)

#Voltage
Spannung= 2000000

m3d.assign_voltage(elec1.faces[0], Spannung, "Voltage1.1")
m3d.assign_voltage(elec1.faces[1], Spannung, "Voltage1.2")
m3d.assign_voltage(elec1.faces[2], Spannung, "Voltage1.3")
m3d.assign_voltage(elec1.faces[3], Spannung, "Voltage1.4")
m3d.assign_voltage(elec1.faces[4], Spannung, "Voltage1.5")
m3d.assign_voltage(elec1.faces[5], Spannung, "Voltage1.6")

m3d.assign_voltage(elec2.faces[0], -Spannung, "Voltage2.1")
m3d.assign_voltage(elec2.faces[1], -Spannung, "Voltage2.2")
m3d.assign_voltage(elec2.faces[2], -Spannung, "Voltage2.3")
m3d.assign_voltage(elec2.faces[3], -Spannung, "Voltage2.4")
m3d.assign_voltage(elec2.faces[4], -Spannung, "Voltage2.5")
m3d.assign_voltage(elec2.faces[5], -Spannung, "Voltage2.6")

#Setup
setup2 = m3d.create_setup(setupname="Setup2",setuptype="ElectricTransient")

#Analysis
m3d.analyze()
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

pyansys

gmalinve, 2024-01-24:
Hi @WynAAAAA, 
Thanks for reporting the issue, if you are interested we are fixing this issue in PR #4123.
We'll let you know as soon as it gets available!
@IreneWoyna

WynAAAAA, 2024-01-24:
Hello @IreneWoyna, 
Thank you for taking care of the issue.☺️
I have no prior experience with GitHub/ pyaedt, so im kinda wondering how long wir will take for resolving this? Days, Weeks or more like months till the next softwareupdate?
Kind regards, @WynAAAAA

gmalinve, 2024-01-24:
Hi @WynAAAAA,

The PR has just been merged and will be available in the next release most probably next week.

Thanks,

Giulia

closed 2024-01-24 by gmalinve