ansys/pyaedt #4105: Update get_scalar_field_value after this bug is resolved: 961653 (after 2024R2)

nnetake, 2024-01-19:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The current implementation of this method uses exporting a temp file and extracting the value by reading the file. However, there is a native API to extract the calculator's evaluated value -  GetTopEntryValue. This API is very slow or failing for large models. Once this is resolved on native API side we can revert implementation to use this function.

### Steps To Reproduce

NA

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

ansys-pythonnet==3.1.0rc3
attrs==23.2.0        
cffi==1.15.1
cfgv==3.3.1
clr-loader==0.2.6    
colorama==0.4.6      
contourpy==1.1.0     
cycler==0.11.0       
defusedxml==0.7.1    
distlib==0.3.6       
exceptiongroup==1.1.1
filelock==3.12.2     
fonttools==4.40.0
fpdf2==2.7.7
identify==2.5.24
iniconfig==2.0.0
jsonschema==4.21.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.4
matplotlib==3.7.1
nodeenv==1.8.0
numpy==1.25.0
packaging==23.1
Pillow==9.5.0
platformdirs==3.8.0
pluggy==1.2.0
plumbum==1.8.2
pre-commit==3.3.3
psutil==5.9.5
pyaedt @ file:///D:/git_src/pyaedt/pyaedt
pycparser==2.21
pyparsing==3.1.0
pytest==7.4.0
python-dateutil==2.8.2
pythonnet==3.0.1
pywin32==306
PyYAML==6.0
referencing==0.32.1
rpds-py==0.17.1
rpyc==5.3.1
six==1.16.0
tomli==2.0.1
virtualenv==20.23.1

Samuelopez-ansys, 2025-06-23:
Hi @nnetake , which is the status of this bug? Please if now it works, close the issue.

nnetake, 2025-06-23:
Hi @Samuelopez-ansys, TFS bug has been resolved. The pyaedt bug was raised for the customer consulting project for API improvement however, things are working well as of now, and I don't see the need to make any changes on the pyaedt side. We can close this bug.

closed 2025-06-23 by nnetake