ansys/pyaedt #4122: PyGranta Material .amat is failed to load

pvinod13, 2024-01-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Unable to load .amat file sucessfully

https://ansys-my.sharepoint.com/:p:/p/vinodkumar_pallapati/ESlk90hm7ghIgT_xF8ayhuoBA5Gok-RRg_JzUxPO3Mivtw?e=gxAlnS

### Steps To Reproduce

1. Install PyGranta Material Library
2. Export material into .amat file
3. Instantiate HFss Instance and add material  
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

ansys-pythonnet==3.1.0rc3
cffi==1.16.0
clr-loader==0.2.6
granta-miscriptingtoolkit @ file:///D:/2023_projects/Goals/granta_pyaedt/Granta_lib/Installers/Windows/granta_miscriptingtoolkit-3.2.164-py3-none-a
ny.whl
importlib-metadata==6.8.0
jaraco.classes==3.3.0
keyring==23.13.1
more-itertools==10.1.0
plumbum==1.8.2
psutil==5.9.6
pyaedt==0.7.3
pycparser==2.21
pywin32==306
pywin32-ctypes==0.2.2
rpyc==5.3.1
zipp==3.17.0

Samuelopez-ansys, 2024-01-31:
Hi @pvinod13,

Are you working on this bug? There are other priorities right now.

pvinod13, 2024-01-31:
Hi @Samuelopez-ansys ,

There are some priorities...!, I can start working from mid-feb

pvinod13, 2024-02-28:
The work around is:

import pyaedt.generic.LoadAEDTFile as load_file
from pyaedt import Hfss

def create_datasets(hfss, mat_dict:dict):
    for item in mat_dict.keys():
        if type(mat_dict[item])==dict and "X" in list(mat_dict[item].keys()) and "Y" in list(mat_dict[item].keys()):
            new_item = item.strip("$")
            hfss.create_dataset(new_item, xlist=mat_dict[item]['X'],ylist=mat_dict[item]['Y'])

def assign_material_properties(hfss,name, mat_dict:dict):
    hfss.materials.add_material(name, props=mat_dict)

def add_reference_data(hfss, mat_dict):
    if 'RefDatasets' in list(mat_dict.keys()):
        create_datasets(hfss, mat_dict["RefDatasets"])
        return mat_dict

    else:
        for key in list(mat_dict.keys()):
            print(key)
            if type(mat_dict[key])==dict:
                add_reference_data(hfss, mat_dict[key])

def add_material_defination(hfss,name, mat_dict):
    if 'AttachedData' in list(mat_dict.keys()):
        assign_material_properties(hfss, name, mat_dict)
    else:
        for key in list(mat_dict.keys()):
            if type(mat_dict[key])==dict:
                add_material_defination(hfss,name, mat_dict[key])

hfss = Hfss()
# extracting the material data from .amat file
path = "file path" # [FR_material_card.zip](https://github.com/ansys/pyaedt/files/14430938/FR_material_card.zip)
mat = load_file.load_entire_aedt_file(path)

name = "FR4_amat" 
add_reference_data(hfss,mat)
add_material_defination(hfss, name, mat)

closed 2024-02-29 by Samuelopez-ansys