ansys/pyaedt #4124: twinbuilder.py add_q3d_dynamic_component() local variable 'list_A' referenced before assignment at line 470, Q3D dynamic link

cmesibov, 2024-01-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When attempting to create a dynamic link to a Q3D model in Simplorer, a name error occurs.  "local variable 'list_A' referenced before assignment" occurs and line 470 is identified.

### Steps To Reproduce

q3d = Q3d(Q3dProjectPath,new_desktop_session=True)
tb = TwinBuilder(new_desktop_session=False)
tb.add_q3d_dynamic_component(q3d.project_name, q3d.design_name, SetupName, SweepName, coupling_matrix_name="Original", state_space_dynamic_link_type="RLGC", component_name="Q3D_RLGC")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

 python -m pip freeze
ansys-pythonnet==3.1.0rc3
attrs==23.1.0
certifi==2023.11.17
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.0
cycler==0.12.1
defusedxml==0.7.1
et-xmlfile==1.1.0
exceptiongroup==1.2.0
fiona==1.9.5
fonttools==4.46.0
fpdf2==2.7.7
geopandas==0.14.1
idna==3.6
imageio==2.33.1
iniconfig==2.0.0
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.8.0
networkx==3.2.1
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.8.0
packaging==23.2
pandas==2.1.1
Pillow==10.1.0
platformdirs==4.1.0
pluggy==1.3.0
plumbum==1.8.2
pooch==1.8.0
psutil==5.9.6
pyaedt==0.7.10
pycparser==2.21
pyparsing==3.1.1
pyproj==3.6.1
pytest==7.4.3
python-dateutil==2.8.2
pytz==2023.3.post1
pyvista==0.42.2
pywin32==306
referencing==0.32.1
requests==2.31.0
rpds-py==0.17.1
rpyc==5.3.1
scikit-rf==0.30.0
scipy==1.11.4
scooby==0.9.2
shapely==2.0.2
six==1.16.0
SRTM.py==0.3.7
tomli==2.0.1
tzdata==2023.3
urllib3==2.1.0
utm==0.7.0
vtk==9.2.6

cmesibov, 2024-01-23:
A Q3D net does not necessarily have a sink or source assigned.  If it doesn't the Q3D net loop should move on to the next net.

gmalinve, 2024-01-24:
Hi @cmesibov, thanks for reporting this.
Would you be able to provide me an example to reproduce this issue? It would help me to better understand.
Thanks very much!
Giulia

cmesibov, 2024-01-24:
Hi Giulia,
In the Q3D case, the design may not have sinks or sources on each net.  So when you loop thru the nets (starting at line 461) the app.net_sources(net) and app.net_sinks(net) may return nothing.

[image]
If either one of these returns an empty list the loop should **_continue_** to the next net. 
[image]
If you like you can contact me at Ansys.  

Chris Mesibov

gmalinve, 2024-01-25:
Hi @cmesibov!

Thanks for reporting this!
I just created a PR that might fixes the issue: #4135.
Could you please test it and let me know?
I encountered a strange behavior while creating the component when the source design is a Q3D but this is due to the native IronPython method CreateComponent().
Could you please give me a feedback?

Thanks! :)

Giulia

closed 2024-01-26 by gmalinve