ansys/pyaedt #4125: Memory Error: creation many object on 3D Layout

ngkazu, 2024-01-24:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I would like to create so many objects(> 90,000) on HFSS 3D Layout.
But when number of objects is reached over around 30,000, memory error is occurred like:
---
Fatal Python error: none_dealloc: deallocating None
Python runtime state: initialized
---
 

### Steps To Reproduce

Run code like:
----
from pyaedt import Hfss3dLayout

aedt_file= 'D:/Projects/PyAEDT/Layout/sample.aedt'
aedt_vers= '2023.2'

nX= 300
nY= 150

h3d= Hfss3dLayout(projectname= aedt_file, specified_version= aedt_vers, new_desktop_session= True,  non_graphical= True)

for iX in range(nX):
    print('*S1: '+str(iX))
    x1= float(iX)
    for iY in range(nY):
        y1= float(iY)
        h3d.modeler.create_rectangle('S1', [x1,y1], [1.,1.], name= 'S1'+str(iX)+str(iY), net_name= "dummy" )

for iX in range(nX):
    print('*S2 : '+str(iX))
    x1= float(iX)
    for iY in range(nY):
        y1= float(iY)
        h3d.modeler.create_rectangle('S2', [x1,y1], [1.,1.], name= 'S2'+str(iX)+str(iY), net_name= "dummy" )

h3d.save_project()
h3d.release_desktop(True, True)
----

I attach the files for your reference.
[sample.zip](https://github.com/ansys/pyaedt/files/14031919/sample.zip)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph==0.17.4
ansicon==1.89.0
ansys-pythonnet==3.1.0rc3
appdirs==1.4.4
blessed==1.20.0
cffi==1.15.1
clr-loader==0.2.6
contourpy==1.1.1
cycler==0.12.1
et-xmlfile==1.1.0
flyingcircus==0.1.4.1
fonttools==4.43.1
jinxed==1.2.0
kiwisolver==1.4.5
lxml==4.9.3
matplotlib==3.8.1
numpy==1.26.0
openpyxl==3.1.2
packaging==23.2
pandas==2.1.1
pefile==2023.2.7
Pillow==10.1.0
plumbum==1.8.2
psutil==5.9.5
pyaedt==0.7.7
pycparser==2.21
pyinstaller==6.2.0
pyinstaller-hooks-contrib==2023.10
pyparsing==3.1.1
python-dateutil==2.8.2
python-pptx==0.6.23
pythonnet==3.0.2
pytk==0.0.2.1
pytz==2023.3.post1
pywin32==306
pywin32-ctypes==0.2.2
rpyc==5.3.1
scipy==1.11.3
setuptools-scm==8.0.4
six==1.16.0
tomli==2.0.1
typing_extensions==4.8.0
tzdata==2023.3
wcwidth==0.2.12
XlsxWriter==3.1.9

ring630, 2024-01-25:
@ngkazu
To create a large number of objects in 3D Layout, please use Edb which is a lot efficient. Below is an example code. 

```
from pyaedt import Hfss3dLayout
from pyaedt import Edb

aedt_file = 'D:/Projects/PyAEDT/Layout/sample.aedt'
aedt_vers = '2023.2'

nX = 300
nY = 150

for iX in range(nX):
    print('*S1: ' + str(iX))
    x1 = float(iX)

for iY in range(nY):
    y1 = float(iY)

edbapp = Edb("sample.aedb", edbversion=aedt_vers)

edbapp.modeler.create_rectangle("S1", "dummy", [x1, y1], [x1+1, y1+1])
#h3d.modeler.create_rectangle('S1', [x1, y1], [1., 1.], name='S1' + str(iX) + str(iY), net_name="dummy")

for iX in range(nX):
    print('*S2 : ' + str(iX))
x1 = float(iX)
for iY in range(nY):
    y1 = float(iY)
edbapp.modeler.create_rectangle("S1", "dummy", [x1, y1], [x1+1, y1+1])
#h3d.modeler.create_rectangle('S2', [x1, y1], [1., 1.], name='S2' + str(iX) + str(iY), net_name="dummy")

edbapp.save_edb_as("new.aedb")

h3d = Hfss3dLayout(projectname="new.aedb", specified_version=aedt_vers, new_desktop_session=True, non_graphical=True)
h3d.release_desktop(True, True)
```

ngkazu, 2024-01-26:
thank you for advice @ring630 
I'll try your suggestion later...

ngkazu, 2024-01-26:
thanks again @ring630 
I had confirmed this issue was avoideed with EDB API.
// Close this issue

closed 2024-01-26 by ngkazu