ansys/pyaedt #4126: edb.export_maxwell() produces export script for SiWave which contains string termination errors

Jan-Hammer, 2024-01-24:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Exporting from EDB via SiWave to Maxwell with python 3.12 results in a script generated named "export_cad.py" with the following content: 

```
import os
edbpath = r'C:\temp\Test.aedb'
exportOptions = os.path.join(r'C:\temp\', 'options.config')
oDoc.ScrImportEDB(edbpath)
oDoc.ScrSaveProjectAs(os.path.join(r'C:\temp\','test.siw'))
oDoc.ScrSetOptionsFor3DModelExport(exportOptions)
q3d_filename = os.path.join(r'C:\temp\', 'Export')
oDoc.ScrExport3DModel('Maxwell', q3d_filename)
oDoc.ScrCloseProject()
oApp.Quit()
```
However, this code is not working as the  os.path.join() function is called with a non-terminated string:
r'C:\temp\'  DOES NOT WORK 
r'C:\temp\ ' OR r'C:\temp\\' WORKS

Please fix the string termination.

### Steps To Reproduce

import pyaedt
aedb_path ="C:\\temp\\Test.aedb" 
Version="2023.2"
path="C:\\temp\\"
ExportName="Export"
HideSiWave=False
edb = pyaedt.Edb(edbversion=Version, edbpath=aedb_path)
edb.save_edb() # Save the edb file
options_config = {
'LAUNCH_Q3D' : 0,
'UNITE_NETS' : 1,
'ASSIGN_SOLDER_BALLS_AS_SOURCES' : 0,
'Q3D_MERGE_SOURCES' : 0,
'Q3D_MERGE_SINKS' : 0,
'CREATE_PORTS_FOR_PWR_GND_NETS' : 0,
'PORTS_FOR_PWR_GND_NETS' : 0,
'GENERATE_TERMINALS' : 0,
'SOLVE_CAPACITANCE' : 0,
'SOLVE_DC_RESISTANCE' : 0,
'SOLVE_DC_INDUCTANCE_RESISTANCE' : 0,
'SOLVE_AC_INDUCTANCE_RESISTANCE' : 0,
'CreateSources' : 0,
'CreateSinks' : 0,
'LAUNCH_Q3D' : 0,
'LAUNCH_HFSS' : 0
}
edb.write_export3d_option_config_file("C:\\temp\\", options_config)
edb.export_maxwell(path_to_output=path,aedt_file_name=ExportName,hidden=HideSiWave)
edb.close_edb()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
cffi==1.16.0
clr-loader==0.2.6
defusedxml==0.7.1
fonttools==4.47.2
fpdf2==2.7.7
numpy==1.26.3
pandas==2.2.0
pillow==10.2.0
plumbum==1.8.2
psutil==5.9.8
pyaedt==0.7.10
pycparser==2.21
python-dateutil==2.8.2
pytz==2023.3.post1
pywin32==306
rpyc==5.3.1
six==1.16.0
tzdata==2023.4

Jan-Hammer, 2024-01-25:
Following solution works:

os.path.join(r'C:\temp\\', ....

Samuelopez-ansys, 2024-01-29:
Hi @Jan-Hammer ,

In python, this is wrong:
aedb_path ="C:\temp\Test.aedb"

You need to escape special characters:
aedb_path =r"C:\temp\Test.aedb"

or

aedb_path ="C:\\temp\\Test.aedb"

https://stackoverflow.com/questions/36397342/python-adds-special-characters-to-path-string

Then for the method: write_export3d_option_config_file , you should also pass a correct path:
r"C:\temp" or "C:\\temp", without the last backslash.

If you go inside the PyAEDT code, you will see how the method is wrapped:

https://github.com/ansys/pyaedt/blob/main/pyaedt/edb.py#L2655

Let us know if it is clear, or if you need anything else.
Samuel

Jan-Hammer, 2024-01-29:
Hi @Samuelopez-ansys , thanks for the support. It seemed that the path including the backslash was the issue.

closed 2024-01-29 by Jan-Hammer