ansys/pyaedt #4132: Problem with "edit_y_axis_scaling" function on PyAEDT

giuseppe1692, 2024-01-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi all, 
I have a issue with my code that is used to Open AEDT, create resistors, inductors, ports and then create a new setup and plot all the Z impedance. 
The code that I wrote is:

`
    non_graphical = False
    new_thread = True
    desktop_version = "2023.2" 
    nome_progetto = "nuovo_progetto"  #variabile da modificare con nome progetto
    nome_design = "PH_MLO"              #variabile da modificare 

    desktop = pyaedt.launch_desktop(desktop_version, non_graphical, new_thread)
    aedt_app = pyaedt.Circuit(projectname=nome_progetto, designname=nome_design)
    
    
    aedt_app.modeler.schematic.schematic_units = "mil"
    

    setup_ac = aedt_app.create_setup("AnalisiAC")
    setup_ac.props["SweepDefinition"]["Data"] = "DEC 1kHz 1GHz 301"
    setup_tab = aedt_app.create_setup("Tabella")
    setup_tab.props["SweepDefinition"]["Data"] = "100MHz"

    offset_y = 0
    offset_x = 0
    for i in range (0,j):
        port = aedt_app.modeler.components.create_interface_port(name=lista_net_needle[i], location=[offset_x, offset_y] )
        inductor = aedt_app.modeler.schematic.create_inductor(value=str(lista_ind_needle[i]) + "n/" + str(lista_pad_needle[i]), location=[offset_x + 500 , offset_y])
        resistor = aedt_app.modeler.schematic.create_resistor(value=str(lista_res_needle[i])+ "m/"+ str(lista_pad_needle[i]), location=[offset_x + 1000, offset_y])
        pins_resistor = resistor.pins
        port.pins[0].connect_to_component(component_pin=inductor.pins[0], use_wire=True)
        inductor.pins[1].connect_to_component(component_pin=resistor.pins[1], use_wire=True)
        offset_y = offset_y - 100
    aedt_app.analyze_setup("AnalisiAC")
    traces = aedt_app.get_traces_for_plot(get_self_terms= True, get_mutual_terms = False, category = "mag(Z")
    
    for i in range (0, len(traces)): 
        aedt_app.post.create_report(expressions= traces[i], plotname= lista_net_needle[i])
        Standard(app = aedt_app, report_category="mag(Z", setup_name="AnalisiAC", expressions=traces[i]).edit_y_axis_scaling(linear_scaling=False, min_scale=None, max_scale=None, minor_tick_divs=5, min_spacing=None, units='MHz')
    aedt_app.release_desktop(close_desktop=False, close_projects=False)`

The code above works but the line below give me an ERROR :

` Standard(app = aedt_app, report_category="mag(Z", setup_name="AnalisiAC", expressions=traces[i]).edit_y_axis_scaling(linear_scaling=False, min_scale=None, max_scale=None, minor_tick_divs=5, min_spacing=None, units='MHz')`

The ERROR that appear is: 
`Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2032.0_x64__qbz5n2kfra8p0\Lib\tkinter\__init__.py", line 1948, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "c:\Users\simulatore06\Desktop\Code\# TOOL PER CREARE SCRIPT TCL PER SIGRITY.py", line 648, in SECONDARY_funtion
    Standard(app = aedt_app, report_category="mag(Z", setup_name="AnalisiAC", expressions=traces[i]).edit_y_axis_scaling(linear_scaling=False, min_scale=None, max_scale=None, minor_tick_divs=5, min_spacing=None, units='MHz')
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\simulatore06\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyaedt\modules\report_templates.py", line 1892, in __init__
    CommonReport.__init__(self, app, report_category, setup_name, expressions)
  File "C:\Users\simulatore06\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\pyaedt\modules\report_templates.py", line 345, in __init__
    for el, k in self._post._app.available_variations.nominal_w_values_dict.items():
                 ^^^^^^^^^^^^^^^
AttributeError: 'Circuit' object has no attribute '_app'
`

Someone could help me to solve described issue? Thank you in advance.

### Steps To Reproduce

- Create new project
- Create new circuit
- Create a list of port inductance and resistance
- Create a new analysis 
- Run analysis
- Create Z impedance Rectangular Plot report for each Port created 
- edit_y_axis_scaling (the part of code that give me the error)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph==0.17.4
ansys-pythonnet==3.1.0rc3
anyio==4.2.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.14.0
beautifulsoup4==4.12.2
bleach==6.1.0
certifi==2023.11.17
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.1
contourpy==1.2.0
cycler==0.12.1
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
executing==2.0.1
fastjsonschema==2.19.1
fiona==1.9.5
fonttools==4.47.2
fpdf2==2.7.7
fqdn==1.5.1
geopandas==0.14.2
idna==3.6
imageio==2.33.1
ipykernel==6.28.0
ipython==8.20.0
ipywidgets==8.1.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.3
json5==0.9.14
jsonpointer==2.4
jsonschema==4.20.0
jsonschema-specifications==2023.12.1
jupyter==1.0.0
jupyter-console==6.6.3
jupyter-events==0.9.0
jupyter-lsp==2.2.1
jupyter_client==8.6.0
jupyter_core==5.7.1
jupyter_server==2.12.4
jupyter_server_terminals==0.5.1
jupyterlab==4.0.10
jupyterlab-widgets==3.0.9
jupyterlab_pygments==0.3.0
jupyterlab_server==2.25.2
kiwisolver==1.4.5
MarkupSafe==2.1.3
matplotlib==3.8.0
matplotlib-inline==0.1.6
mistune==3.0.2
nbclient==0.9.0
nbconvert==7.14.1
nbformat==5.9.2
nest-asyncio==1.5.8
networkx==3.2.1
notebook==7.0.6
notebook_shim==0.2.3
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.8.1
overrides==7.4.0
packaging==23.2
pandas==2.1.1
pandocfilters==1.5.0
parso==0.8.3
pefile==2023.2.7
pillow==10.2.0
platformdirs==4.1.0
plumbum==1.8.2
pooch==1.8.0
prometheus-client==0.19.0
prompt-toolkit==3.0.43
psutil==5.9.7
pure-eval==0.2.2
pyaedt==0.7.4
pycparser==2.21
Pygments==2.17.2
pyinstaller==6.2.0
pyinstaller-hooks-contrib==2023.10
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
python-json-logger==2.0.7
pytz==2023.3.post1
pyvista==0.42.2
pywin32==306
pywin32-ctypes==0.2.2
pywinpty==2.0.12
PyYAML==6.0.1
pyzmq==25.1.2
qtconsole==5.5.1
QtPy==2.4.1
referencing==0.32.1
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.17.1
rpyc==5.3.1
scikit-rf==0.30.0
scipy==1.11.4
scooby==0.9.2
seaborn==0.13.1
Send2Trash==1.8.2
setuptools==69.0.2
shapely==2.0.2
six==1.16.0
sniffio==1.3.0
soupsieve==2.5
SRTM.py==0.3.7
stack-data==0.6.3
terminado==0.18.0
tinycss2==1.2.1
tk==0.1.0
tornado==6.4
traitlets==5.14.1
types-python-dateutil==2.8.19.20240106
tzdata==2023.4
uri-template==1.3.0
urllib3==2.1.0
utm==0.7.0
vtk==9.3.0
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.7.0
widgetsnbextension==4.0.9

gmalinve, 2024-01-25:
Hi @giuseppe1692,

what are you trying to do in this for loop?
```
for i in range (0, len(traces)): 
    aedt_app.post.create_report(expressions= traces[i], plotname= lista_net_needle[i])
    Standard(app = aedt_app, report_category="mag(Z", setup_name="AnalisiAC", expressions=traces[i]).edit_y_axis_scaling(linear_scaling=False, min_scale=None, max_scale=None, minor_tick_divs=5, min_spacing=None, units='MHz')
```
If your intention is to edit the y axis scaling then you can assign create_report to a variable: 
`myreport = aedt_app.post.create_report(expressions= traces[i], plotname= lista_net_needle[i])`
so you get the report object and then edit the y axis scaling from the report object, i.e:
`myreport.edit_y_axis_scaling(linear_scaling=False, min_scale="-50", max_scale="10", minor_tick_divs=10, min_spacing="5")`
You don't need to instantiate the Standard report object because you already have it in line above.

Hope this helps,

Giulia

giuseppe1692, 2024-01-25:
Hi @gmalinve,
thank you so much for your reply. Now my code works. 
I take this opportunity to ask you another question.
Is possible to change the axis Number Format from "Auto" to "Decimal"?

Thank you in advance.
Regards,
Giuseppe

gmalinve, 2024-01-25:
Hi @giuseppe1692,

Unfortunately is not available yet. Please submit an issue and someone from the community will take care of this!
I'll close this one since it seems you sorted your original problem.

Thanks,

Giulia

closed 2024-01-25 by gmalinve

giuseppe1692, 2024-01-25:
Thank you so much!