ansys/pyaedt #4134: In pyeadt new version (0.7.10), Error at "oProject = oDesktop.GetActiveProject()", returning a 'NoneType' object

Phedro-Shousa, 2024-01-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Error occurs right at the begging when opening or creating a project. The script opens Ansys Eletronic Desktop, and when going to select/activate the design it ouputs error, since the method  `oDesktop.GetActiveProject() `is returning wrongly or nothing.
This problem happens in pyaedt 0.7.10 but works fine with pyaedt 0.6.76

### Steps To Reproduce

Given the following code:
```
import sys
sys.path.append('C:/Program Files/AnsysEM/v231/')
sys.path.append('C:/Program Files/AnsysEM/v231/Win64/PythonFiles/DesktopPlugin/')
import ScriptEnv

ScriptEnv.Initialize("Ansoft.ElectronicsDesktop")
oDesktop.RestoreWindow()
oProject = oDesktop.SetActiveProject()
oDesign = oProject.SetActiveDesign("HFSSDesign1")
```
a error is generated:

'> oDesign = oProject.SetActiveDesign("Maxwell3DDesign1") 
AttributeError: 'NoneType' object has no attribute 'SetActiveDesign'

Meaning that the method 'oDesktop.SetActiveProject()' is returning a NoneType object, halting the script execution

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

alabaster==0.7.13
ansys-pythonnet==3.1.0rc3
appdirs==1.4.4
Babel==2.13.1
bcrypt==4.0.1
beautifulsoup4==4.12.2   
black==23.11.0
bump2version==1.0.1      
certifi==2023.7.22       
cffi==1.16.0
cfgv==3.4.0
charset-normalizer==3.3.2
click==8.1.7
clr-loader==0.2.6        
codespell==2.2.6
colorama==0.4.6
contourpy==1.0.7
coverage==6.5.0
coveralls==3.3.1
cryptography==41.0.5
cycler==0.11.0
distlib==0.3.7
docopt==0.6.2
docutils==0.18.1
et-xmlfile==1.1.0
exceptiongroup==1.1.3
filelock==3.13.1
flake8==6.1.0
fonttools==4.38.0
furo==2023.9.10
hfsslib==0.0.0
identify==2.5.31
idna==3.4
imagesize==1.4.1
iniconfig==2.0.0
isort==5.12.0
Jinja2==3.1.2
kiwisolver==1.4.4
markdown-it-py==3.0.0
MarkupSafe==2.1.3
matplotlib==3.7.0
mccabe==0.7.0
mdit-py-plugins==0.4.0
mdurl==0.1.2
mypy-extensions==1.0.0
myst-parser==2.0.0
nodeenv==1.8.0
numpy==1.24.2
openpyxl==3.1.2
packaging==23.0
pandas==1.5.3
paramiko==3.3.1
pathspec==0.11.2
patsy==0.5.3
Pillow==9.4.0
platformdirs==3.11.0
plotly==5.18.0
plotly-express==0.4.1
pluggy==1.3.0
plumbum==1.8.2
pre-commit==3.5.0
psutil==5.9.7
pyaedt==0.7.10
pycodestyle==2.11.1
pycparser==2.21
pyeit==1.2.4
pyflakes==3.1.0
Pygments==2.16.1
pylibiio==0.23.1
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.10
PyQt5-Qt5==5.15.2
PyQt5-sip==12.13.0
pyserial==3.5
pytest==7.4.3
pytest-cov==4.1.0
pytest-html==4.1.1
pytest-libiio==0.0.13
pytest-metadata==3.0.0
python-dateutil==2.8.2
pythonnet==3.0.1
pytoml==0.1.21
pytz==2022.7.1
PyVISA==1.14.0
pywin32==306
PyYAML==6.0.1
requests==2.31.0
rpyc==5.3.1
scapy==2.5.0
scipy==1.10.1
scriptenv==0.3.0
shapely==2.0.1
six==1.16.0
snowballstemmer==2.2.0
soupsieve==2.5
Sphinx==7.2.6
sphinx-basic-ng==1.0.0b2
sphinx-favicon==1.0.1
sphinx-rtd-theme==1.3.0
sphinxcontrib-applehelp==1.0.7
sphinxcontrib-devhelp==1.0.5
sphinxcontrib-htmlhelp==2.0.4
sphinxcontrib-jquery==4.1
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-mermaid==0.9.2
sphinxcontrib-qthelp==1.0.6
sphinxcontrib-serializinghtml==1.1.9
statsmodels==0.14.0
tenacity==8.2.3
tk==0.1.0
tomli==2.0.1
trimesh==3.20.0
typing_extensions==4.8.0
urllib3==2.1.0
virtualenv==20.24.6

gmalinve, 2024-01-25:
Hi @Phedro-Shousa,

You are not using PyAEDT to achieve your intent but you are still using IronPython native API.

Giulia

Phedro-Shousa, 2024-01-25:
Hi @gmalinve 
Forgive me, I pasted the wrong code example block of when i was trying to debug test. Originally i was using oDesktop.GetActiveProject() 
However, I get the same result in both cases.
Here the original code;

```
from pyaedt import Maxwell3d
from pyaedt import generate_unique_project_name

non_graphical = False
Design_Name = "Maxwell3DDesign1"
Solver = "ACConduction"
DesktopVersion = "2023.1"

M3D = Maxwell3d(
    projectname=generate_unique_project_name(),
    designname=Design_Name,
    solution_type=Solver,
    specified_version=DesktopVersion,
    non_graphical=non_graphical,
    new_desktop_session=True
)

M3D.modeler.model_units = "mm"
modeler = M3D.modeler
Plot = M3D.odesign.GetModule("ReportSetup")

import sys
sys.path.append('C:/Program Files/AnsysEM/v231/')
sys.path.append('C:/Program Files/AnsysEM/v231/Win64/PythonFiles/DesktopPlugin/')
import ScriptEnv

ScriptEnv.Initialize("Ansoft.ElectronicsDesktop")
oDesktop.RestoreWindow()
oProject = oDesktop.GetActiveProject()
oDesign = oProject.SetActiveDesign("Maxwell3DDesign1")
```

gmalinve, 2024-01-25:
@Phedro-Shousa  what are you trying to achieve? your code snippet is a mix of IronPython API and PyAEDT.
You can retreive your project name or project list through your Maxwell instance for example.
Above all you don't retrieve a report in this way: Plot = M3D.odesign.GetModule("ReportSetup") if this is your intention. 
If you want to retrieve a report you can do so with m3d.post.reports_by_category for example.
But again, I don't know your intention so this is just a suggestion.

Hope this helps.

Giulia

Phedro-Shousa, 2024-01-26:
Thank you @gmalinve 

The intention was to through a python script to create a new project on Ansys Eletronic Desktop and use the Maxwell3D design functinalities (taking advantage of record/run script service from Ansys Eletronic Desktop).
For some reason with the newest version of pyaedt package (0.7.10) the `oProject = oDesktop.GetActiveProject()` would return nothing or an empty object, which would cause a `<'NoneType' object has no attribute 'SetActiveDesign'>` error since `oProject` was empty
While when i downgraded it to the pyaedt 0.6.76, `oDesktop.GetActiveProject()` returns a `PyDesktopPlugin.AedtPropServer object`, and works quite fine. It is how I'm using it at the moment.
I thought that it could be relevant to post the issue to pyaedt. 
But if it's not related with pyaedt, I'm sorry and maybe we could close/delete this issue.

Thank you for your time,

Pedro

gmalinve, 2024-01-26:
Hi @Phedro-Shousa,

If your intention is to access the desktop object you can do so by doing:
`from pyaedt import Maxwell3d
m3d = Maxwell3d()
desktop = m3d.desktop_class`

I noticed from your previous code snippet that you want to set an active design. You could do so with PyAEDT without the need of using IronPython API:
`m3d.set_active_design(design_name)`

Hope this helps!

Giulia

Phedro-Shousa, 2024-01-26:
Understood,
Thank you @gmalinve 

Pedro

closed 2024-01-26 by Phedro-Shousa

iaskgithub, 2024-02-20:
> Hi @gmalinve Forgive me, I pasted the wrong code example block of when i was trying to debug test. Originally i was using oDesktop.GetActiveProject() However, I get the same result in both cases. [...]

> Hi @gmalinve Forgive me, I pasted the wrong code example block of when i was trying to debug test. Originally i was using oDesktop.GetActiveProject() However, I get the same result in both cases. [...]

Hey@Phedro-Shousa , I am doing the same works this days as you did, can u hidden the maxwell gui windows when use the  IronPython native API  ?

Phedro-Shousa, 2024-02-21:
@iaskgithub Not sure if I understood. I'm not running anymore the pyaedt 0.7.10 version that would throw the error. But i remember Ansoft ElectronicsDesktop desktop would open, but halt the execution before even opening the Maxwell GUI.

iaskgithub, 2024-02-22:
> @iaskgithub Not sure if I understood. I'm not running anymore the pyaedt 0.7.10 version that would throw the error. But i remember Ansoft ElectronicsDesktop desktop would open, but halt the execution before even opening the Maxwell GUI.

yes, the Ansoft ElectronicsDesktop desktop  will open when use the IronPython native API,  
but controll it with non_graphical=True or False with pyaedt