ansys/pyaedt #4137: Unable to install \pyaedt\.[tests] with python 3.8

anspraksaph, 2024-01-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Unable to install \pyaedt\.[tests] with python 3.8 with the error shown below

[image]

ERROR: Ignored the following versions that require a different python version: 1.11.0 Requires-Python <3.13,>=3.9; 1.11.0rc1 Requires-Python <3.13,>=3.9; 1.11.0rc2 Requires-Python <3.13,>=3.9; 1.11.1 Requires-Python <3.13,>=3.9; 1.11.2 Requires-Python <3.13,>=3.9; 1.11.3 Requires-Python <3.13,>=3.9; 1.11.4 Requires-Python >=3.9; 1.12.0 Requires-Python >=3.9; 1.12.0rc1 Requires-Python >=3.9; 1.12.0rc2 Requires-Python >=3.9; 1.2.3 Requires-Python >=3.9; 1.25.0 Requires-Python >=3.9; 1.25.0rc1 Requires-Python >=3.9; 1.25.1 Requires-Python >=3.9; 1.25.2 Requires-Python >=3.9; 1.26.0 Requires-Python <3.13,>=3.9; 1.26.0b1 Requires-Python <3.13,>=3.9; 1.26.0rc1 Requires-Python <3.13,>=3.9; 1.26.1 Requires-Python <3.13,>=3.9; 1.26.2 Requires-Python >=3.9; 1.26.3 Requires-Python >=3.9; 1.3.0 Requires-Python >=3.9; 1.3.1 Requires-Python >=3.9; 1.4.0 Requires-Python >=3.9; 1.4.0rc1 Requires-Python >=3.9; 2.1.0 Requires-Python >=3.9; 2.1.0rc0 Requires-Python >=3.9; 2.1.1 Requires-Python >=3.9; 2.1.2 Requires-Python >=3.9; 2.1.3 Requires-Python >=3.9; 2.1.4 Requires-Python >=3.9; 2.2.0 Requires-Python >=3.9; 2.2.0rc0 Requires-Python >=3.9; 3.8.0 Requires-Python >=3.9; 3.8.0rc1 Requires-Python >=3.9; 3.8.1 Requires-Python >=3.9; 3.8.2 Requires-Python >=3.9; 8.13.1 Requires-Python >=3.9; 8.13.2 Requires-Python >=3.9; 8.14.0 Requires-Python >=3.9; 8.15.0 Requires-Python >=3.9; 8.16.0 Requires-Python >=3.9; 8.16.1 Requires-Python >=3.9; 8.17.0 Requires-Python >=3.9; 8.17.1 Requires-Python >=3.9; 8.17.2 Requires-Python >=3.9; 8.18.0 Requires-Python >=3.9; 8.18.1 Requires-Python >=3.9; 8.19.0 Requires-Python >=3.10; 8.20.0 Requires-Python >=3.10
ERROR: Could not find a version that satisfies the requirement scipy==1.11.4; extra == "tests" (from pyaedt[tests]) (from versions: 0.8.0, 0.9.0, 0.10.0, 0.10.1, 0.11.0, 0.12.0, 0.12.1, 0.13.0, 0.13.1, 0.13.2, 0.13.3, 0.14.0, 0.14.1, 0.15.0, 0.15.1, 0.16.0, 0.16.1, 0.17.0, 0.17.1, 0.18.0, 0.18.1, 0.19.0, 0.19.1, 1.0.0, 1.0.1, 1.1.0, 1.2.0, 1.2.1, 1.2.2, 1.2.3, 1.3.0, 1.3.1, 1.3.2, 1.3.3, 1.4.0, 1.4.1, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 1.5.4, 1.6.0, 1.6.1, 1.6.2, 1.6.3, 1.7.0, 1.7.1, 1.7.2, 1.7.3, 1.8.0rc1, 1.8.0rc2, 1.8.0rc3, 1.8.0rc4, 1.8.0, 1.8.1, 1.9.0rc1, 1.9.0rc2, 1.9.0rc3, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0rc1, 1.10.0rc2, 1.10.0, 1.10.1)
ERROR: No matching distribution found for scipy==1.11.4; extra == "tests"

### Steps To Reproduce

With python 3.8 installed, clone pyaedt (says at C:\pyaedt ) and in command line windows type:
- py -3.8 -m venv C:\venv_3.8
- call C:\venv_3.8\Scripts\activate
- python -m pip install pip -U
- python -m pip install wheel setuptools -U
- python -m pip install C:\pyaedt\.[tests]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

attrs==23.2.0
colorama==0.4.6
exceptiongroup==1.2.0
importlib-resources==6.1.1
iniconfig==2.0.0
packaging==23.2
pluggy==1.4.0
pytest==7.4.4
pytest-azurepipelines==1.0.5
pytest-nunit==1.0.4
tomli==2.0.1
zipp==3.17.0

gmalinve, 2024-01-26:
Hi @anspraksaph, could you please type:
`pip install .[tests]`
from pyaedt repo directory?

SMoraisAnsys, 2024-01-26:
Hi @anspraksaph, thanks for notifying us of such an issue.
Seems like a side effect of #4118, I'll try to fix that today.

SMoraisAnsys, 2024-01-26:
I did a little digging and it seems that it's not the first time that such an issue happened (cf #3532).
I'll try to update the CI to perform smoke testing on multiple python versions.

MaxJPRey, 2024-01-26:
@SMoraisAnsys If not already the case, please use matrix for Python version at least for the smoke tests.

PR #4139 (2024-01-26): MAINT: use scikit-rf 0.31.0
This change happens following the fix on scikit-rf side.

Reminder: scikit-rf was trying to import a specific function from scipy while not bounding the version of scipy. The newest version of scipy had no longer the required function leading to CI breaks on our side as we no longer could run the tests.

Btw, I was able to test this change on a container with python 3.7 and it was working fine.

Closes #4137
closed 2024-01-26 by SMoraisAnsys