ansys/pyaedt #4153: Bug located in "variation_dict" parameter in "ipk.post.export_field_file"

kinielsen, 2024-01-30:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Dear Team, 

We have discovered that the "ipk.post.export_field_file" gives an error when trying to use the variation_dict parameter. 
We get the following error. 

[image]

Please see the attached script and example project file (with results) 
[ExFieldField_Git.zip](https://github.com/ansys/pyaedt/files/14095982/ExFieldField_Git.zip)

From that you should be able to reproduce. 

If there is an alternative way of extracting field file from big parametric sweep, that would also be very helpful. :-)

Best regards, 
Kim 

### Steps To Reproduce

Replace "project_path" in ExportFieldFile.py and run script. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

ansi2html==1.8.0
ansys-pythonnet==3.1.0rc3
astunparse==1.6.3
attrs==23.1.0
Brotli==1.1.0
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.0
cov-core==1.15.0
coverage==7.3.0
cycler==0.11.0
dash==2.4.1
dash-core-components==2.0.0
dash-design-kit==1.6.7
dash-html-components==2.0.0
dash-snapshots==0.0.1
dash-table==5.0.0
defusedxml==0.7.1
dill==0.3.7
diskcache==5.6.3
docutils==0.20.1
et-xmlfile==1.1.0
Fiona==1.9.4.post1
Flask==2.2.5
Flask-Compress==1.14
fonttools==4.42.1
fpdf2==2.7.7
geopandas==0.13.2
gunicorn==20.0.4
idna==3.4
imageio==2.31.3
importlib-metadata==6.8.0
itsdangerous==2.1.2
Jinja2==3.1.2
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
MarkupSafe==2.1.3
matplotlib==3.8.0
multiprocess==0.70.15
nest-asyncio==1.5.7
networkx==3.1
nose2==0.13.0
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.6.0
packaging==23.1
pages==0.3
pandas==2.1.1
piglet-templates==1.3.0
Pillow==10.0.0
platformdirs==3.10.0
plotly==5.16.1
plumbum==1.8.2
pooch==1.7.0
psutil==5.9.5
pyaedt==0.7.10
pycparser==2.21
PyMySQL==1.1.0
pyparsing==3.0.9
pyproj==3.6.0
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.42.2
pywin32==306
PyYAML==6.0.1
referencing==0.33.0
requests==2.31.0
retrying==1.3.4
rpds-py==0.17.1
rpyc==5.3.1
scikit-rf==0.29.0
scipy==1.11.2
scooby==0.7.2
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
tenacity==8.2.3
tk==0.1.0
typing_extensions==4.7.1
tzdata==2023.3
urllib3==2.0.4
utm==0.7.0
vtk==9.2.6
Werkzeug==2.2.3
zipp==3.17.0

gmalinve, 2024-01-31:
Hi @kinielsen, we were able to reproduce the bug and we are fixing it.
The fix should be available in the next release.
Thanks for feedback!
Giulia

closed 2024-02-02 by Samuelopez-ansys

kinielsen, 2024-02-07:
Hi @gmalinve and @Samuelopez-ansys ,

Thank you very much for your ultra quick help! I will test the latest version soon! 

Best regards, 
Kim