ansys/pyaedt #4159: How is it possible to export magnetic field Values Bx, By for every (x,y) meshelement in Maxwell2DAdd ...

briewo, 2024-01-31:
### Description of the feature

i tried to use the functions export_mesh_obj() and export_field_file_on_grid() with some options but i did not get a result. Result is always None

Example code:

import pyaedt as ed
app = ed.Maxwell2d(specified_version=specified_version, projectname=name_of_project, designname=name_of_design)
mesh = app.post.export_mesh_obj(setup_name="Setup1", intrinsic_dict=None)

gmalinve, 2024-01-31:
Hi @briewo,

Have you tried using export_field_plot?
`m2d.post.export_field_plot(plot_name, file_path)`
This return the exported file path.
This file (which can be an .aedtplot or .case) contains all the information about mesh data and field data.

Hope this helps!

Giulia

gmalinve, 2024-01-31:
@briewo 
Another way to do it is using the plot_field() which creates a plot using Pyvista (if you are interested more in the visualization of your results). This creates an object through which you can access PyVista functionalities + creates an .aedtplot file and exports the field plot as an image.
Please look at this example to see how it is used: https://aedt.docs.pyansys.com/version/stable/examples/03-Maxwell/Maxwell2D_DCConduction.html

gmalinve, 2024-02-07:
Hi @briewo,

Does this solve your issue? 
if it does I'll close this issue.
Please do let us know!

Giulia

closed 2024-02-14 by Samuelopez-ansys