ansys/pyaedt #4186: pyaedt-icepak- creating and assigning values to a region becomes extremely time-consuming

YLSHA, 2024-02-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I encountered an issue in the following scenario: creating 400 square regions, each with an independent heat flux density setting. In actual execution using PyAEDT, the creation of the first few regions is very fast. However, after about 60 regions, creating and assigning values to a region becomes extremely time-consuming. While using `ipk.assign_source` or `ipk.assign_stationary_wall_with_heat_flux` for 400 regions takes 20 minutes, the first 20 regions may only take a few seconds.
The python version is 3.11.5.
Thank you.

### Steps To Reproduce

```python

REC = []
for y in range(20):
    for x in range(20):
        REC.append(f"rec_{x}_{y}")
        REC[-1] = ipk.modeler.create_rectangle(csPlane=2, position=[50.*x, 50.*y, 500], dimension_list=[50.,50.])
        ipk.assign_stationary_wall_with_heat_flux(
            REC[-1].bottom_face_z.id,
            heat_flux=f"{surface_flux[x,y]}W_per_m2",
            thickness="0mm",
            material="Mat1",
        )
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

archspec @ file:///croot/archspec_1697725767277/work
boltons @ file:///C:/ci_311/boltons_1677729932371/work
Brotli @ file:///C:/ci_311/brotli-split_1676435766766/work
certifi @ file:///C:/b/abs_91u83siphd/croot/certifi_1700501720658/work/certifi
cffi @ file:///C:/b/abs_924gv1kxzj/croot/cffi_1700254355075/work
charset-normalizer @ file:///tmp/build/80754af9/charset-normalizer_1630003229654/work
colorama @ file:///C:/ci_311/colorama_1676422310965/work
conda @ file:///C:/b/abs_88drkd35d4/croot/conda_1701719561256/work
conda-content-trust @ file:///C:/b/abs_e3bcpyv7sw/croot/conda-content-trust_1693490654398/work
conda-libmamba-solver @ file:///croot/conda-libmamba-solver_1702997573971/work/src
conda-package-handling @ file:///C:/b/abs_b9wp3lr1gn/croot/conda-package-handling_1691008700066/work
conda_package_streaming @ file:///C:/b/abs_6c28n38aaj/croot/conda-package-streaming_1690988019210/work
cryptography @ file:///C:/b/abs_e8cnom_zw_/croot/cryptography_1702071486468/work
distro @ file:///C:/b/abs_a3uni_yez3/croot/distro_1701455052240/work
idna @ file:///C:/ci_311/idna_1676424932545/work
jsonpatch @ file:///tmp/build/80754af9/jsonpatch_1615747632069/work
jsonpointer==2.1
libmambapy @ file:///C:/b/abs_2euls_1a38/croot/mamba-split_1704219444888/work/libmambapy
menuinst @ file:///C:/b/abs_e8p75b4m9q/croot/menuinst_1702390332729/work
packaging @ file:///C:/b/abs_28t5mcoltc/croot/packaging_1693575224052/work
platformdirs @ file:///C:/b/abs_b6z_yqw_ii/croot/platformdirs_1692205479426/work
pluggy @ file:///C:/ci_311/pluggy_1676422178143/work
pycosat @ file:///C:/b/abs_31zywn1be3/croot/pycosat_1696537126223/work
pycparser @ file:///tmp/build/80754af9/pycparser_1636541352034/work
pyOpenSSL @ file:///C:/b/abs_08f38zyck4/croot/pyopenssl_1690225407403/work
PySocks @ file:///C:/ci_311/pysocks_1676425991111/work
requests @ file:///C:/b/abs_316c2inijk/croot/requests_1690400295842/work
ruamel.yaml @ file:///C:/ci_311/ruamel.yaml_1676439214109/work
tqdm @ file:///C:/b/abs_f76j9hg7pv/croot/tqdm_1679561871187/work
truststore @ file:///C:/b/abs_55z7b3r045/croot/truststore_1695245455435/work
urllib3 @ file:///C:/b/abs_9cmlsrm3ys/croot/urllib3_1698257595508/work
win-inet-pton @ file:///C:/ci_311/win_inet_pton_1676425458225/work
zstandard==0.19.0

Samuelopez-ansys, 2024-02-05:
@YLSHA I have tested your code, and indeed, when there are some boundaries created, the UI starts to be slow.

I have created 300 regions and it took 3 minutes and 23 secs.

I have tested using the AEDT API directly as an alternative:

```
import pyaedt
import math

ipk = pyaedt.Icepak(specified_version="2023.2")

REC = []
ipk.logger.reset_timer()

cont = 1
for y in range(20):
    for x in range(10):
        REC.append(f"rec_{x}_{y}")
        new = ipk.modeler.create_rectangle(csPlane=2, position=[50. * x, 50. * y, 500], dimension_list=[50., 50.])
        # ipk.assign_stationary_wall_with_heat_flux(
        #     new.bottom_face_z.id,
        #     heat_flux="2irrad_W_per_m2",
        #     thickness="0mm",
        #     material="Al-Extruded",
        # )

        ipk.oboundary.AssignStationaryWallBoundary(["NAME:StationaryWall_" + str(cont),
                                                    "Faces:=", [new.bottom_face_z.id],
                                                    "Thickness:=", "0mm",
                                                    "Solid Material:=", "Al-Extruded",
                                                    "External Condition:=", "Heat Flux",
                                                    "Heat Flux:=", "2irrad_W_per_m2",
                                                    "Temperature:=", "AmbientTemp",
                                                    "Heat Transfer Coefficient:=", "0w_per_m2kel",
                                                    "Reference Temperature:=", "AmbientTemp",
                                                    [
                                                        "NAME:Heat Transfer Data",
                                                        "Heat Transfer Correlation:=", False,
                                                        "Heat Transfer Convection Type:=", "Forced Convection"
                                                    ],
                                                    [
                                                        "NAME:Radiation",
                                                        "Radiate:=", False
                                                    ],
                                                    "Shell Conduction:=", False,
                                                    "Slip Flow Boundary:=", False,
                                                    "External Surface Radiation:=", False,
                                                    "External Material:=", "Steel-oxidised-surface",
                                                    "External Radiation Reference Temperature:=", "AmbientTemp",
                                                    "External Radiation View Factor:=", "1"
                                                    ])
        cont += 1

ipk.logger.info_timer("Assignment finished")
ipk.release_desktop(False, False)
```

In this case, the code took: 1 min 49 secs

@lorenzovecchietti @siva-krishnaswamy Could you take a look at this issue?

Final comment, running the code in non graphical mode would also help.

YLSHA, 2024-02-06:
Thank you very much！ This method is much faster than the previous one.

lorenzovecchietti, 2024-02-15:
@Samuelopez-ansys 
Although it is true that with many BCs, the code is slower, I cannot see any significant difference with respect to using native APIs. So this does not seem related to pyaedt, but rather Icepak backend, especially in graphical mode.

@YLSHA your code was raising warnings for the unit used (`W_per_m2` instead of `irrad_W_per_m2` that Icepak was expecting).

Comparing 400 rectangle creation with native and pyaedt assignment:
- `PyAEDT INFO: Native assignment finished Elapsed time: 2m 28sec`
- `PyAEDT INFO: pyAEDT assignment finished Elapsed time: 2m 29sec`

Code used:
```
import pyaedt

ipk = pyaedt.Icepak(specified_version="241")

REC = []
ipk.logger.reset_timer()
cont = 1
for y in range(14):
    for x in range(14):
        REC.append(f"rec_{x}_{y}")
        new = ipk.modeler.create_rectangle(csPlane=2, position=[50. * x, 50. * y, 500], dimension_list=[50., 50.])
        ipk.oboundary.AssignStationaryWallBoundary(["NAME:StationaryWall_" + str(cont),
                                                    "Faces:=", [new.bottom_face_z.id],
                                                    "Thickness:=", "0mm",
                                                    "Solid Material:=", "Al-Extruded",
                                                    "External Condition:=", "Heat Flux",
                                                    "Heat Flux:=", "2irrad_W_per_m2",
                                                    "Temperature:=", "AmbientTemp",
                                                    "Heat Transfer Coefficient:=", "0w_per_m2kel",
                                                    "Reference Temperature:=", "AmbientTemp",
                                                    [
                                                        "NAME:Heat Transfer Data",
                                                        "Heat Transfer Correlation:=", False,
                                                        "Heat Transfer Convection Type:=", "Forced Convection"
                                                    ],
                                                    [
                                                        "NAME:Radiation",
                                                        "Radiate:=", False
                                                    ],
                                                    "Shell Conduction:=", False,
                                                    "Slip Flow Boundary:=", False,
                                                    "External Surface Radiation:=", False,
                                                    "External Material:=", "Steel-oxidised-surface",
                                                    "External Radiation Reference Temperature:=", "AmbientTemp",
                                                    "External Radiation View Factor:=", "1"
                                                    ])
        cont += 1
ipk.logger.info_timer("Native assignment finished")

old=ipk.design_list[0]
ipk.insert_design("pyaedtFuncs")
ipk.delete_design(old)

REC = []
ipk.logger.reset_timer()
for y in range(14):
    for x in range(14):
        REC.append(f"rec_{x}_{y}")
        REC[-1] = ipk.modeler.create_rectangle(csPlane=2, position=[50. * x, 50. * y, 500], dimension_list=[50., 50.])
        ipk.assign_stationary_wall_with_heat_flux(
            REC[-1].bottom_face_z.id,
            heat_flux="5irrad_W_per_m2",
            thickness="0mm",
            material="Al-Extruded",
        )
ipk.logger.info_timer("pyAEDT assignment finished")

ipk.release_desktop(False, False)
```

closed 2024-02-15 by lorenzovecchietti