ansys/pyaedt #4187: Error when launching EDB

ring630, 2024-02-05:
Hi @ring630, 

I'm looking deeper into PyAEDT, and I am getting an error where if I just start an EDB class, it says I am trying to access memory that could be corrupted or is protected and will crash the program. I have attached the image for your reference. I correctly installed pyaedt and have a functional ANSYS tool for SIwave and Electronics Desktop. Is there a reason why I am getting this error, and/or did I forget to potentially install something or set a path?

[image]

_Originally posted by @loganchoi in https://github.com/ansys/pyaedt/discussions/4156#discussioncomment-8350908_

ring630, 2024-02-05:
@Samuelopez-ansys @svandenb-dev

svandenb-dev, 2024-02-05:
@ring630,
I just made a test in PyEDB and all is working fine for me. I would suggest waiting we release PyEDB and hopefully it should be resolved automaticallly.
edb = Edb()
PyEDB INFO: Edb version 2024.1
PyEDB INFO: StdOut is enabled
PyEDB INFO: Logger is initialized in EDB.
PyEDB INFO: legacy v0.1.dev0
PyEDB INFO: Python version 3.9.4 (tags/v3.9.4:1f2e308, Apr  6 2021, 13:40:21) [MSC v.1928 64 bit (AMD64)]
PyEDB INFO: No EDB is provided. Creating a new EDB C:\Users\svandenb\Documents\layout_BJ98OP.aedb.
PyEDB INFO: EDB C:\Users\svandenb\Documents\layout_BJ98OP.aedb created correctly.
PyEDB INFO: EDB initialized.

ring630, 2024-02-05:
@loganchoi, I cannot reproduce your problem. Does your machine have access to licenser server?

loganchoi, 2024-02-05:
@ring630 The ANSYS SIWave versions I utilize are 2022 R1 and 2023 R1 and I have already added the licenses to all the ANSYS products to my environment variables.

ring630, 2024-02-06:
@loganchoi by the error log, it looks like a .NET framwork related issue. One possible reason is your Python environment is corrupted. Can you create a new virtual environment and install the latest PyAEDT? and give it a try.

Samuelopez-ansys, 2024-02-06:
Maybe the default directory does not have write permissions?

SMoraisAnsys, 2024-02-06:
It could be related to the compatibility between the DLL and the python environment.
Btw, it seems that the test is performed in python 3.12 and we only support:

-    "Programming Language :: Python :: 3.7",
-    "Programming Language :: Python :: 3.8",
-    "Programming Language :: Python :: 3.9",
-    "Programming Language :: Python :: 3.10"

We might update this list if we ensure that the tests are working fine.

loganchoi, 2024-02-06:
@Samuelopez-ansys I specified a different directory where I have permissions and it doesn't throw the corrupted memory error anymore. However, it says I failed to initialize DLLs and that I have a list index out of range error. 

@SMoraisAnsys I will try that next and let you know what happened.

@ring630 I have the correct pyaedt version of 7.4 and reinstalled it as well.

loganchoi, 2024-02-06:
@SMoraisAnsys I changed my python version to be 3.10 and I still get the same errors.

loganchoi, 2024-02-07:
@svandenb-dev I have the ANSYS tool licenses in my path in the environment variables. Do I need to add something else for me to access it via user terminal?

svandenb-dev, 2024-02-08:
@loganchoi No I think it should be ok. Can you confirm you can start Desktop like HFSS from PyAEDT ? If yes you may have some conflicts in your libraries which prevent loading the EDB dll. It's not easy to figure out without more information.

Samuelopez-ansys, 2024-02-08:
Hi @loganchoi ,

Please try:

1) Delete the virtual environment

2) Create a new virtual environment

3) Install pyaedt (pip install pyaedt)

4) Check if AEDT 2023R1 or 2023R2 is opening correctly on your computer, and check that you do not have any license issue.

5) Try the script

If after all these steps, it is still not working, please send an email: pyansys.support@ansys.com

loganchoi, 2024-02-12:
@svandenb-dev Yeah, I can start a desktop application and start an HFSS project by using PyAEDT. 

@Samuelopez-ansys I created a new virtual environment and AEDT 2023R1 can open correctly, but can't run an Edb() command.

loganchoi, 2024-02-12:
@Samuelopez-ansys I sent an email to pyansys.support@ansys.com and I got an automatic email saying that the email will not be monitored anymore and that the best support is to utilize this github or on discussion forums on ansys.com.

SMoraisAnsys, 2024-02-13:
@loganchoi Thanks for your feeback. From your previous input and answers, I have the feeling that the LOC responsible for your issue is `self._active_cell = list(self.top_circuit_cells)[0]`. Can you try the following code ? It it extracted from an example and will download an aedb file that we use for the documentation. 

```python
import pyaedt

targetfile = pyaedt.downloads.download_file('edb/ANSYS-HSD_V1.aedb')
edb = pyaedt.Edb(edbpath=targetfile)
```

If this lines of code work on your side then the reason for the failure is probably related to the AEDB file that you provide. We expect it to have some cells and are unable to get one during `open_edb()`. If the code does not work, please provide the resulting error message.

loganchoi, 2024-02-13:
@SMoraisAnsys I've gotten a similar error of PyAEDT failing to initialize DLLs.

[image]

SMoraisAnsys, 2024-02-13:
@loganchoi With what kind of python interpreter did you created your virtual environment ? Did you created your virtual environment (the python base was installed from scratch ? microsoft installation ? something else ?

Could you try to create a new python environment with the CPython embedded with AEDT ? The path should be something like `"C:\Program Files\AnsysEM\v231\Win64\commonfiles\CPython\3_7\winx64\Release\python\python.exe"` or `"C:\Program Files\AnsysEM\v232\Win64\commonfiles\CPython\3_10\winx64\Release\python\python.exe"`.

From PowerShell, the full command to create the virtual environment would be ` &"C:\Program Files\AnsysEM\v231\Win64\commonfiles\CPython\3_7\winx64\Release\python\python.exe" -m venv VirtualEnv`. Then proceed as before.

loganchoi, 2024-02-13:
@Samuelopez-ansys I utilized the python based to create the virtual environment. with the command 

python3.10 -m venv environment

I followed your instructions and still get the DLLs error.

Samuelopez-ansys, 2024-02-14:
@loganchoi Could you check your virtual environments? If you have anyone related to Ansys, or pointing to any other AEDT version, please remove it.

In addition, can you install any other AEDT version (2023R2 for instance)? Maybe your AEDT version is corrupted because it is my first time seeing this issue.

SMoraisAnsys, 2024-02-14:
> @Samuelopez-ansys I utilized the python based to create the virtual environment. with the command
> 
> python3.10 -m venv environment
> 
> I followed your instructions and still get the DLLs error.

Am I right to assume that this message was addressed to me ? If you still have the same error while using the python interpreter included in AEDT common files, I would suggestion to follow Samuel's comment. There might be an issue with your AEDT installation.