ansys/pyaedt #4190: issue#1478 seems to persist in pyaedt 0.7: Changing plating thickness of Vias does not have effect on exported files to Maxwell (Vias are exported filled)

Jan-Hammer, 2024-02-06:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello, 

it appears that issue #1478 is still persistent in 0.7. When trying to export from EDB to Maxwell the vias are filled regardless of the assigned plating thickness 

### Steps To Reproduce

import pyaedt

Version="2022.2"
ExportName="Export"
filename="Test.aedb"
PlatingThickness=0.1E-3  

path=os.path.abspath(os.getcwd())
aedb_path =os.path.join(path,filename).replace('\\','\\\\')
HideSiWave=True
edb = pyaedt.Edb(edbversion=Version, edbpath=aedb_path)
edb.save_edb() # Save the edb file

options_config = {
'LAUNCH_MAXWELL' : 0,
'NUM_PADS_FACET_COUNT': 8,
'NUM_ANTIPAD_FACET_COUNT': 8,
'DEFAULT_SOLDERBALL_FACET_COUNT' : 8,
'VIA_SEGMENTS' : 8,
'DEFAULT_BONDWIRE_FACET_COUNT' : 6,
'UNITE_NETS' : 0,
'EXCLUDE_TERMINALS_FROM_UNITE' : 1,
'SIMPLY_VIA_MODEL' : 0,
'IGNORE_DIELECTRICS' : 0,
'SEPARATE_DIELECTRICS' : 1,
'UNITE_LAYERS_WITH_SAME_MATERIALS' : 1,
'IGNORE_UNCONNECTED_PADS' : 0,
'CLIP_TRACES' : 0,
'CUT_DIELECTRICS' : 0,
'CREATE_SHEET_BODIES' : 0,
'GENERATE_TERMINALS' : 0,
'IGNORE_PLANES_WITH_AREA_LESS_THAN_THRESOLD' : 1,
'IGNORE_FLOAT_BODIES' : 0,
'MIN_PLANE_AREA' : 0.358979,
'MIN_EDGE_LENGTH_PADS' : '1um',
'MIN_EDGE_LENGTH_PLANES' : '1um',
'MIN_EDGE_LENGTH_TRACES' : '1um',
'MIN_DIELECTRIC_EDGE_LENGTH' : '10um',
'DIELECTRIC_EXPANSION_FACTOR' : 0.100000,
'IGNORE_HOLES' : 0,
'MIN_HOLE_AREA' : 0.358979,
'REMOVE-PLATING_TAILS' : 0,
'SUBTRACT_METAL_FROM_SUBSTRATE' : 0,
'DISCRETIZE_ARCS' : 0,
'CHOP_TRACE_ENDS' : 0,
'AIRBOX_THICKNESS_FACTOR' : 1.100000,
'AIRBOX_PAD_AMOUNT_PLUS_Z' : 0.500000,
'AIRBOX_PAD_AMOUNT_MINUS_Z' : 0.500000,
'PORT_PAD_AMOUNT' : 0.500000,
'CREATE_PORTS_FOR_PWR_GND_NETS' : 0,
'PORTS_FOR_PWR_GND_NETS' : 0,
'LAUNCH_HFSS' : 0,
'USE_CAUSAL_MATERIALS' : 1,
'AUTO_DC_THICKNESS' : 0,
'SOLVE_CAPACITANCE' : 0,
'SOLVE_DC_RESISTANCE' : 0,
'SOLVE_DC_INDUCTANCE_RESISTANCE' : 0,
'SOLVE_AC_INDUCTANCE_RESISTANCE' : 0,
'SOLVE_PROJECT' : 0,
'LAUNCH_Q3D' : 0,
'ASSIGN_SOLDER_BALLS_AS_SOURCES' : 0,
'Q3D_MERGE_SOURCES' : 0,
'Q3D_MERGE_SINKS' : 0,
'ACIS_VERSION' : 0
}
edb.write_export3d_option_config_file(path, options_config)

PadStackNames=[x for x in edb.core_padstack.padstacks] #API CALL: read all padstacks
for x in PadStackNames:
    try:
        edb.core_padstack.padstacks[x].hole_plating_ratio=0
        edb.core_padstack.padstacks[x].hole_plating_thickness=PlatingThickness # API CALL: set plating thickness         
        
    except Exception as e:
        continue        
edb.save_edb()

edb.export_maxwell(path_to_output=path,aedt_file_name=ExportName,hidden=HideSiWave)
edb.save_edb() #API CALL: Save file
edb.close_edb()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

C:\temp>python -m pip freeze
ansys-pythonnet==3.1.0rc3
cffi==1.16.0
clr-loader==0.2.6
defusedxml==0.7.1
fonttools==4.47.2
fpdf2==2.7.7
numpy==1.26.3
pandas==2.2.0
pillow==10.2.0
plumbum==1.8.2
psutil==5.9.8
pyaedt==0.7.10
pycparser==2.21
python-dateutil==2.8.2
pytz==2023.3.post1
pywin32==306
rpyc==5.3.1
six==1.16.0
tzdata==2023.4

Samuelopez-ansys, 2024-02-06:
@Jan-Hammer In the Closed Issue, @maxcapodi78 mentioned that it was a limitation in SIWave API, so PyAEDT can not solve it until SIwave does not expose the method.

I am not sure if it is fixed, but if it is, you will need to update to a new Ansys version.

Please @ring630 @svandenb-dev , this is an EDB related issue, could you help?

Thanks

ring630, 2024-02-06:
@Samuelopez-ansys may be we can implement this in PyAEDT.

ring630, 2024-02-15:
@Jan-Hammer it looks like there is no keyword in configure file to toggle 100% via fill checkbox. I am waiting answer from development.

ring630, 2024-02-16:
@Jan-Hammer the keyword is TOTAL_VIA_FILL. The SIwave documetation will be updated. By setting it to 0, the via will have a hole  in maxwell. 

options_config = {'TOTAL_VIA_FILL' : 0}

[image]