ansys/pyaedt #4193: Bug located in Tools->Run script ->PyAEDTInstallerFromDesktop

alvaro-rod-esp, 2024-02-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello,
When i try to use PyAEDT i follow the steps of Tools->Run script ->PyAEDTInstallerFromDesktop 
However when i do that y get the message:

*Global - Messages
  [info] Installing PyAEDT.
  [info] PyAEDT setup complete.
  [info] "C:\Users\alvar\AppData\Roaming\pyaedt_env_ide\v232\Scripts\python.exe" "c:\users\alvar\appdata\local\temp\configure_pyaedt.py"
  [error] Error configuring the configuration Tab.

And in automation window nothing change, and in Tools->Toolkit still says (BETA)  install PyAEDT

[image]

### Steps To Reproduce

Tools->Run script ->PyAEDTInstallerFromDesktop 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

i am tryng to install PyAEDT i am new to the program

alvaro-rod-esp, 2024-02-07:
I am using ansys maxwell for student

Samuelopez-ansys, 2024-02-07:
@alvaro-rod-esp I can not reproduce the issue, it is working on my side. I would retry again, but first, remove all the info you have in your PersonalLib folder:
[image]

And also the virtual environment:
AppData\Roaming\pyaedt_env_ide\v232

If it does not work, you will need to check the script, and run it step by step to see where it is failing. 

Regards,
Samuel

alvaro-rod-esp, 2024-02-07:
Thank you @Samuelopez-ansys  !!!!! ,
it work when i remove the info in the PersonalLib Folder, don`t know why but it work and now i have the options i am supposed to when i install PyAEDT
Thank you very much

closed 2024-02-07 by Samuelopez-ansys