ansys/pyaedt #4194: Icepak "assign_monitor_point()" not updated properly without opening GUI

kinielsen, 2024-02-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Dear Team, 

We have a tool that takes a ODB++ file, creates Icepak PCB and places monitor points in each location where a component from HFSS 3DL exists, and runs a simulation.
This can potentially be several hundreds of monitor points. 

The code below is used in the for loop for each component. 
[image]

We have discovered that when we have many monitor points and try to run an analysis without opening the GUI - we get the following errors for (I think) each monitor point:

[image]

Validation is successful prior to the analysis. 

If we open the preprocessed model before analyzing, we see no issues. All monitor points are created and the analysis will run with no errors. 

It seems like some sort of check or update is done to the model with all the monitor points when opening in the GUI. 
And I was wondering if there is a way to do the same kind of update with pyAEDT? 

I know this is a bit vague, but I am unfortunately not able to share the entire code. 

I hope you can help? 

Best regards, 
Kim 

### Steps To Reproduce

...

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.6.0
ansi2html==1.9.1
ansys-api-sherlock==0.1.22
ansys-pythonnet==3.1.0rc3
ansys-sherlock-core==0.4.0
attrs==23.1.0
Brotli==1.1.0
certifi==2023.7.22
cffi==1.16.0
charset-normalizer==3.3.1
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.1.1
cycler==0.12.1
dash==2.14.2
dash-core-components==2.0.0
dash-design-kit==1.6.7
dash-html-components==2.0.0
dash-snapshots==0.0.1
dash-table==5.0.0
defusedxml==0.7.1
dill==0.3.7
diskcache==5.6.3
et-xmlfile==1.1.0
fiona==1.9.5
Flask==2.2.5
Flask-Compress==1.14
fonttools==4.43.1
fpdf==1.7.2
fpdf2==2.7.7
geopandas==0.14.0
grpcio==1.59.0
gunicorn==20.0.4
idna==3.4
imageio==2.31.6
importlib-metadata==7.0.1
itsdangerous==2.1.2
Jinja2==3.1.2
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
MarkupSafe==2.1.3
matplotlib==3.8.0
multiprocess==0.70.15
nest-asyncio==1.6.0
networkx==3.2
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.7.0
packaging==23.2
pandas==2.1.1
Pillow==10.0.1
platformdirs==3.11.0
plotly==5.16.1
plumbum==1.8.2
pooch==1.8.0
protobuf==3.20.3
psutil==5.9.6
pyaedt==0.7.11
pycparser==2.21
pydantic==2.5.3
pydantic_core==2.14.6
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
pythonnet==3.0.2
pytz==2023.3.post1
pyvista==0.42.2
pywin32==306
PyYAML==6.0.1
referencing==0.33.0
requests==2.31.0
retrying==1.3.4
rpds-py==0.17.1
rpyc==5.3.1
scikit-rf==0.29.1
scipy==1.11.3
scooby==0.9.2
shapely==2.0.2
six==1.16.0
SRTM.py==0.3.7
tenacity==8.2.3
typing_extensions==4.9.0
tzdata==2023.3
urllib3==2.0.7
utm==0.7.0
vtk==9.2.6
Werkzeug==3.0.1
zipp==3.17.0

Samuelopez-ansys, 2024-02-14:
@lorenzovecchietti Could you check this issue?

lorenzovecchietti, 2024-02-14:
@kinielsen What AEDT version are you using?
Also, at what point in your code does the issue occur? It seems from your message that it is happening after `ipk.analyze`, but the code should never go through there with that command. Could you specify at what point in the code the error occurs?

kinielsen, 2024-04-11:
Hi @lorenzovecchietti ,

I am sorry for the late response. 
Yes, it is happening after ipk.analyze(). 
We assign monitor point on the location of each component found in the "hfss3dlayout.modeler.components", saving the project and analyzing. 
Unfortunately I have not have time to test it any further, but I have seen that the problem disappears when I exclude some components - for example component with Reference Designator starting with "R" or "C". 
I am planning on doing more tests to see if I can find out if it related to certain components (placement etc.) or the number of components to which I assign mount points. 

I will let you know how it goes. 

Best regards, 
Kim

lorenzovecchietti, 2024-10-01:
> I will let you know how it goes.
How did it went? :)

Let me know if I can close the issue.

kinielsen, 2024-10-02:
@lorenzovecchietti  

Hi Lorenzo, 
I apologize for not getting back to you. It is working just fine, now. 👍 
You can close the issue! 
Best regards, 
Kim

closed 2024-10-02 by lorenzovecchietti