ansys/pyaedt #4196: Can't create AEDT project: ModuleNotFoundError: No module named 'jsonschema'

onyx22574, 2024-02-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Can't start and open a new project from AEDT Desktop with using conda env

```
import sys
import atexit
from pyaedt import *
import pyaedt
pyaedt.settings.use_grpc_api=False

desktop = Desktop(
        specified_version= version,
        new_desktop_session=False,
        non_graphical=False,
        close_on_exit=False,
        student_version=False,
    )
myQ = pyaedt.Q3d(projectname=pyaedt.generate_unique_project_name(),
               specified_version="2023.2",
               non_graphical=False,
               new_desktop_session=True)

```
Initializing new desktop!
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Log on file is enabled
PyAEDT INFO: Log on Desktop Message Manager is enabled
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT outside AEDT with gRPC plugin.
PyAEDT INFO: Launching AEDT using the gRPC plugin.
PyAEDT INFO: New AEDT session is starting on gRPC port 51670
PyAEDT INFO: AEDT installation Path C:\Program Files\AnsysEM\v232\Win64
PyAEDT INFO: Ansoft.ElectronicsDesktop.2023.2 version started with process ID 2696.
PyAEDT INFO: pyaedt v0.7.11
PyAEDT INFO: Python version 3.10.13 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:24:38) [MSC v.1916 64 bit (AMD64)]
PyAEDT INFO: AEDT 2023.2.1 Build Date 2023-10-16 09:58:56
Traceback (most recent call last):
  File "", line 13, in <module>
    myQ = pyaedt.Q3d(projectname=pyaedt.generate_unique_project_name(),
  File "", line 1192, in Q3d
    from pyaedt.q3d import Q3d as app
  File "", line 11, in <module>
    from pyaedt.application.Analysis3D import FieldAnalysis3D
  File "", line 7, in <module>
    from pyaedt.generic.configurations import Configurations
  File "", line 32, in <module>
    from jsonschema import exceptions
ModuleNotFoundError: No module named 'jsonschema'

Type "help", "copyright", "credits" or "license" for more information.
>>> import pyaedt
>>> pyaedt.version
'0.7.11'
>>>

### Steps To Reproduce

```
import sys
import atexit
from pyaedt import *
import pyaedt
pyaedt.settings.use_grpc_api=False

#desktop = pyaedt.Desktop(specified_version="2023.2", non_graphical=False)

desktop = Desktop(
        specified_version= version,
        new_desktop_session=False,
        non_graphical=False,
        close_on_exit=False,
        student_version=False,
    )
myQ = pyaedt.Q3d(projectname=pyaedt.generate_unique_project_name(),
               specified_version="2023.2",
               non_graphical=False,
               new_desktop_session=True)

```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

(pyaedt_py310) C:\>python -m pip freeze
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1698341106958/work
bcrypt @ file:///C:/Windows/Temp/abs_36kl66t_aw/croots/recipe/bcrypt_1659554334050/work
Bottleneck @ file:///C:/Windows/Temp/abs_3198ca53-903d-42fd-87b4-03e6d03a8381yfwsuve8/croots/recipe/bottleneck_1657175565403/work
Brotli @ file:///C:/Windows/Temp/abs_63l7912z0e/croots/recipe/brotli-split_1659616056886/work
certifi @ file:///home/conda/feedstock_root/build_artifacts/certifi_1707022139797/work/certifi
cffi @ file:///C:/b/abs_924gv1kxzj/croot/cffi_1700254355075/work
charset-normalizer @ file:///tmp/build/80754af9/charset-normalizer_1630003229654/work
clr-loader @ file:///C:/b/abs_10l9j6yukj/croot/clr_loader_1675268053891/work
colorama @ file:///C:/b/abs_a9ozq0l032/croot/colorama_1672387194846/work
comm @ file:///home/conda/feedstock_root/build_artifacts/comm_1704278392174/work
contourpy @ file:///C:/b/abs_853rfy8zse/croot/contourpy_1700583617587/work
cryptography @ file:///C:/b/abs_e8cnom_zw_/croot/cryptography_1702071486468/work
cycler @ file:///tmp/build/80754af9/cycler_1637851556182/work
debugpy @ file:///C:/b/abs_c0y1fjipt2/croot/debugpy_1690906864587/work
decorator @ file:///home/conda/feedstock_root/build_artifacts/decorator_1641555617451/work
exceptiongroup @ file:///home/conda/feedstock_root/build_artifacts/exceptiongroup_1704921103267/work
executing @ file:///home/conda/feedstock_root/build_artifacts/executing_1698579936712/work
fonttools==4.25.0
idna @ file:///C:/b/abs_bdhbebrioa/croot/idna_1666125572046/work
importlib-metadata @ file:///home/conda/feedstock_root/build_artifacts/importlib-metadata_1703269254275/work
ipykernel @ file:///D:/bld/ipykernel_1707182954185/work
ipython @ file:///D:/bld/ipython_1706795789991/work
jedi @ file:///home/conda/feedstock_root/build_artifacts/jedi_1696326070614/work
jupyter_client @ file:///home/conda/feedstock_root/build_artifacts/jupyter_client_1699283905679/work
jupyter_core @ file:///D:/bld/jupyter_core_1704727156030/work
kiwisolver @ file:///C:/b/abs_88mdhvtahm/croot/kiwisolver_1672387921783/work
loguru @ file:///C:/b/abs_1bt9_bp4tc/croot/loguru_1675318496956/work
matplotlib @ file:///C:/b/abs_e26vnvd5s1/croot/matplotlib-suite_1698692153288/work
matplotlib-inline @ file:///home/conda/feedstock_root/build_artifacts/matplotlib-inline_1660814786464/work
mkl-fft @ file:///C:/b/abs_19i1y8ykas/croot/mkl_fft_1695058226480/work
mkl-random @ file:///C:/b/abs_edwkj1_o69/croot/mkl_random_1695059866750/work
mkl-service==2.4.0
munkres==1.1.4
nest_asyncio @ file:///home/conda/feedstock_root/build_artifacts/nest-asyncio_1705850609492/work
numexpr @ file:///C:/b/abs_5fucrty5dc/croot/numexpr_1696515448831/work
numpy @ file:///C:/b/abs_16b2j7ad8n/croot/numpy_and_numpy_base_1704311752418/work/dist/numpy-1.26.3-cp310-cp310-win_amd64.whl#sha256=e84057072c37569bd0e11652dc2a75980d4d360f2391adf6a29a2fb1622d20ff
packaging @ file:///C:/b/abs_28t5mcoltc/croot/packaging_1693575224052/work
pandas @ file:///C:/b/abs_fej9bi0gew/croot/pandas_1702318041921/work/dist/pandas-2.1.4-cp310-cp310-win_amd64.whl#sha256=186ecb939535861d7a5c9527cbb8529ba1031ca6e0c2ddfb054baa09e2fcc2e0
paramiko @ file:///opt/conda/conda-bld/paramiko_1640109032755/work
parso @ file:///home/conda/feedstock_root/build_artifacts/parso_1638334955874/work
pickleshare @ file:///home/conda/feedstock_root/build_artifacts/pickleshare_1602536217715/work
pillow @ file:///C:/b/abs_e22m71t0cb/croot/pillow_1707233126420/work
platformdirs @ file:///C:/b/abs_b6z_yqw_ii/croot/platformdirs_1692205479426/work
plumbum @ file:///home/conda/feedstock_root/build_artifacts/plumbum_1685457629260/work
pooch @ file:///C:/b/abs_a8nmng7d_x/croot/pooch_1695850149827/work
prompt-toolkit @ file:///home/conda/feedstock_root/build_artifacts/prompt-toolkit_1702399386289/work
psutil @ file:///C:/Windows/Temp/abs_b2c2fd7f-9fd5-4756-95ea-8aed74d0039flsd9qufz/croots/recipe/psutil_1656431277748/work
pure-eval @ file:///home/conda/feedstock_root/build_artifacts/pure_eval_1642875951954/work
pyaedt @ file:///D:/bld/pyaedt_1706946327976/work
pycparser @ file:///tmp/build/80754af9/pycparser_1636541352034/work
Pygments @ file:///home/conda/feedstock_root/build_artifacts/pygments_1700607939962/work
PyNaCl @ file:///C:/Windows/Temp/abs_d5c3ajcm87/croots/recipe/pynacl_1659620667490/work
pyOpenSSL @ file:///C:/b/abs_08f38zyck4/croot/pyopenssl_1690225407403/work
pyparsing @ file:///C:/Users/BUILDE~1/AppData/Local/Temp/abs_7f_7lba6rl/croots/recipe/pyparsing_1661452540662/work
PySocks @ file:///C:/ci_310/pysocks_1642089375450/work
python-dateutil @ file:///tmp/build/80754af9/python-dateutil_1626374649649/work
pythonnet @ file:///home/conda/feedstock_root/build_artifacts/pythonnet_1693338669030/work
pytz @ file:///C:/b/abs_19q3ljkez4/croot/pytz_1695131651401/work
pyvista @ file:///home/conda/feedstock_root/build_artifacts/pyvista_1707239096884/work
pywin32==305.1
pyzmq @ file:///C:/b/abs_89aq69t0up/croot/pyzmq_1705605705281/work
requests @ file:///C:/b/abs_316c2inijk/croot/requests_1690400295842/work
rpyc @ file:///home/conda/feedstock_root/build_artifacts/rpyc_1610420736758/work
scooby @ file:///home/conda/feedstock_root/build_artifacts/scooby_1697858861125/work
six @ file:///tmp/build/80754af9/six_1644875935023/work
stack-data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1669632077133/work
tornado @ file:///D:/bld/tornado_1656937966227/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1704212992681/work
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1702176139754/work
tzdata @ file:///croot/python-tzdata_1690578112552/work
urllib3 @ file:///C:/b/abs_9cmlsrm3ys/croot/urllib3_1698257595508/work
vtk===-PKG-VERSION
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1704731205417/work
win-inet-pton @ file:///C:/ci_310/win_inet_pton_1642658466512/work
win32-setctime @ file:///home/tkoch/Workspace/win32_setctime/win32_setctime_1643630045199/work
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1695255097490/work

(pyaedt_py310)

onyx22574, 2024-02-07:
I have v0.7.11

```
import pyaedt
pyaedt.version
'0.7.11'
```

Samuelopez-ansys, 2024-02-07:
@onyx22574 Not sure how you did update the pyaedt version, but in 0.7.11 version, the toml file was updated:
https://github.com/ansys/pyaedt/blob/main/pyproject.toml

It means jsonschema package is now part of the main dependencies, doing, pip install pyaedt -U should update it and also add the new library, if it does not work, please install it manually:

pip install jsonschema

onyx22574, 2024-02-07:
not working...

onyx22574, 2024-02-07:
I did install as you mentioned in [https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html](url)

Install PyAEDT in Conda virtual environment
Create virtual environment

`conda create --name pyaedt_py310 python=3.10`
Activate virtual environment

`conda activate pyaedt_py310`
You can also install PyAEDT from Conda-Forge with this command:

`conda install -c conda-forge pyaedt`

Samuelopez-ansys, 2024-02-07:
@onyx22574 If you install the jsonschema library in the same virtual environment you are running, it should work, because the error you shared is clear. Your venv does not have this library installed.

I guess you installed pyaedt in one python environment and jsonschema in another. It is very important to understand the virtual environment concept. 

Please check jsonschema is installed in the same environment where you are using pyaedt.

Samuelopez-ansys, 2024-02-07:
In summary, when you do python -m pip freeze, you should see as well jsonschema module

onyx22574, 2024-02-07:
I followed the steps from [https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html](url)

there is no "jnsonschema" mentioned.

Please share what needs to be installed in that "pyaedt_py310"

Samuelopez-ansys, 2024-02-07:
It is not mentioned because it should work automatically when you update PyAEDT. But in your case, it is not working, for that reason I am giving the solution...

The workaround I think should work for you is:

1) Activate the virtual environment where pyaedt is installed in a Terminal, it should be called ¨pyaedt_py310· (if you do not know how to activate a virtual environment, please go to Internet and see how to activate a conda existing environment)

2) Once you have it enabled: pip install jsonschema

3) Check if the module is installed: pip freeze

Then it should work. If you do not like the workaround, you could remove the virtual environment (the folder pyaedt_py310), and try again .

If both solutions does not work, we will deeper investigation. Most of PyAEDT user does  not use conda, they directly use CPython from AEDT (first option in the installation documentation)

Samuelopez-ansys, 2024-02-14:
@onyx22574 Any update?

closed 2024-02-22 by Samuelopez-ansys