ansys/pyaedt #4201: Fatal Error when starting EDB when using version higher than 2022R2.

Ouam74, 2024-02-08:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi,

i am trying the following code :

**import pyaedt
import shutil
import os

version = "2024.1"

working_dir = r"D:\Working_dir"

if os.path.isdir(working_dir):
    shutil.rmtree(working_dir)
    os.mkdir(working_dir)

aedb_path = os.path.join(working_dir, pyaedt.generate_unique_name("pcb") + ".aedb")
print(aedb_path)
edb = pyaedt.Edb(edbpath=aedb_path, edbversion=version)

edb.stackup.add_layer("GND")
edb.stackup.add_layer("Diel", "GND", layer_type="dielectric", thickness="0.1mm", material="FR4_epoxy")
edb.stackup.add_layer("TOP", "Diel", thickness="0.05mm")

edb.save_edb()
edb.close_edb()

h3d = pyaedt.Hfss3dLayout(projectname=aedb_path, specified_version=version, new_desktop_session=True, non_graphical=False)
h3d.release_desktop()**

It runs on my colleagues computer but not on mine when using : 
**version = "2023.2" / version = "2024.1" / version = "2024.2"**

but it works fine for version = "2022.2".

I tried to run the program in Pycharm and in the command line.

I get following errors : 

[image]

[image]

[image]

It seems than whatever version i am using, Pyaedt seems to search only in the **2022** installation folder : **C:\Program Files\AnsysEM\v222\Win64**

### Steps To Reproduce

Described previously

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
attrs==23.2.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.0
cycler==0.12.1
defusedxml==0.7.1
et-xmlfile==1.1.0
fiona==1.9.5
fonttools==4.48.1
fpdf2==2.7.7
geopandas==0.14.3
idna==3.6
imageio==2.33.1
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.8.0
networkx==3.2.1
numpy==1.26.0
openpyxl==3.1.2
requests==2.31.0
rpds-py==0.17.1
rpyc==5.3.1
scikit-rf==0.31.0
scipy==1.12.0
scooby==0.9.2
shapely==2.0.2
six==1.16.0
SRTM.py==0.3.7
tzdata==2023.4
urllib3==2.2.0
utm==0.7.0
vtk==9.2.6

Ouam74, 2024-02-10:
I found the problem. 
I had a path to 2022R2 in my environment variables list...
Sorry ;-)

closed 2024-02-10 by Ouam74

ring630, 2024-02-12:
@Samuelopez-ansys  @svandenb-dev FYI