ansys/pyaedt #4209: 3d layout component is not imported properly

ring630, 2024-02-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

hfss.modeler.insert_layout_component(
    comp_file=str(fdir_model / model_3d["pcb"]),
    coordinate_system="Guiding_Pin",
    name="pcb",
    layout_coordinate_systems=["Global", "C5", "CS_1", "H0"],
)

### Steps To Reproduce

3dlayout comp is imported into HFSS, but not properly.
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

all

ring630, 2024-02-09:
missing info of "Info:="

https://github.com/ansys/pyaedt/blob/55a89f7dd04a594dc47b4f7d0abf0325ecd3c1ef/pyaedt/modeler/cad/Primitives3D.py#L1688

```
oComponentManager.Add(
    [
        "NAME:DCDC-Converter-App_main",
        "Info:=",
        [],
        "CircuitEnv:=", 0,
        "Refbase:=", "U",
        "NumParts:=", 1,
        "ModSinceLib:=", True,
        "CompExtID:=", 9,
        "ModelEDBFilePath:=",
        "D:\\_pycharm_project\\my_docs\\workflows\\1_WF_cad_assembly\\models\\DCDC-Converter-App_main.aedbcomp",
        "EDBCompPassword:=", ""
    ])
```

Below is an example of correct code recording from UI.
```
oComponentManager.Add(
    [
        "NAME:DCDC-Converter-App_main",
        "Info:=",
        [
            "Type:=", 29,
            "NumTerminals:=", 0,
            "DataSource:=", "",
            "ModifiedOn:=", 1707488232,
            "Manufacturer:=", "",
            "Symbol:=", "DCDC-Converter-App_main",
            "ModelNames:=", "",
            "Footprint:=", "",
            "Description:=", "",
            "InfoTopic:=", "",
            "InfoHelpFile:=", "",
            "IconFile:=", "BlueDot.bmp",
            "Library:=", "",
            "OriginalLocation:=", "Project",
            "IEEE:=", "",
            "Author:=", "",
            "OriginalAuthor:=", "",
            "CreationDate:=", 1707488231,
            "ExampleFile:=", "",
            "HiddenComponent:=", 0,
            "CircuitEnv:=", 0,
            "GroupID:=", 0
        ],
        "CircuitEnv:=", 0,
        "Refbase:=", "U",
        "NumParts:=", 1,
        "ModSinceLib:=", True,
        "CompExtID:=", 9,
        "ModelEDBFilePath:=",
        "D:\\_pycharm_project\\my_docs\\workflows\\1_WF_cad_assembly\\models\\DCDC-Converter-App_main.aedbcomp",
        "EDBCompPassword:=", ""
    ])
```

Samuelopez-ansys, 2025-04-28:
Hi @hui-zhou-a , is this issue fixed?

closed 2025-10-15 by hui-zhou-a