ansys/pyaedt #4210: Bug when using function `create_frequency_sweep` in solver Q3D

Nanaaaaaaa19, 2024-02-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When adding a sweep frequency to our setup in the Q3D solver, using the function `create_frequency_sweep` , the sweep is created but the values and type of sweep is wrong 

### Steps To Reproduce

import pyaedt 
import pyvista
pyvista.set_jupyter_backend("trame")
import matplotlib.pyplot as plt
"""Launch HFSS and save project"""
non_graphical = False
project_name = pyaedt.generate_unique_project_name(project_name="Spiral_inductor")
app_q3d = pyaedt.Q3d(projectname = project_name,
                specified_version="2024.1",
                non_graphical=non_graphical,
                new_desktop_session=True
                )
app_q3d.design_settings() #checking we have the right settings 
app_q3d.change_material_override()
app_q3d.design_settings() #checking we have the right settings 

"""Create spiral inductor"""
my_modeler = app_q3d.modeler
my_modeler.model_units = "mm" #set to milimeter 
my_udmPairs = []
mypair = ["ILD Thickness (ILD)", "0.006mm"]
my_udmPairs.append(mypair)
mypair = ["Line Spacing (LS)", "0.004mm"]
my_udmPairs.append(mypair)
mypair = ["Line Thickness (LT)", "0.005mm"]
my_udmPairs.append(mypair)
mypair = ["Line Width (LW)", "0.004mm"]
my_udmPairs.append(mypair)
mypair = ["No. of Turns (N)", 2]
my_udmPairs.append(mypair)
mypair = ["Outer Diameter (OD)", "0.15mm"]
my_udmPairs.append(mypair)
mypair = ["Substrate Thickness", "0.2mm"]
my_udmPairs.append(mypair)
mypair = ["Inductor Type","Octagonal,"]
my_udmPairs.append(mypair)
mypair = ["Underpass Thickness (UT)", "0.001mm"]
my_udmPairs.append(mypair)
mypair = ["Via Thickness (VT)", "0.001mm"]
my_udmPairs.append(mypair)

obj_udm = my_modeler.create_udm(udmfullname="Q3DExtractor/OnDieSpiralInductor.py", udm_params_list=my_udmPairs) 
""" Automatically creates model geometry 
    Geometry is parametrized 
"""
"""Assign nets/ sources/ sink """
nets = app_q3d.auto_identify_nets()
face_source = my_modeler.get_object_faces("Inductor")[45]
face_sink = my_modeler.get_object_faces("Inductor")[41]
source = app_q3d.assign_source_to_objectface(face_source, source_name="Source1", net_name = "Inductor")
sink = app_q3d.assign_sink_to_objectface(face_sink, sink_name="Sink1", net_name = "Inductor")

# Defining setup 
setup1 = app_q3d.create_setup(setupname="Setup1")
setup1.props["SaveFields"] = True
assert setup1.props["AdaptiveFreq"] == "1GHz"

#create the sweep as following in the picture attached 
[image]
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

python -m pip freeze
aiohttp==3.9.3
aiosignal==1.3.1
ansys-pythonnet==3.1.0rc3
asttokens==2.4.1
attrs==23.2.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.1
contourpy==1.2.0
cycler==0.12.1
decorator==5.1.1
defusedxml==0.7.1
executing==2.0.1
fonttools==4.47.2
fpdf2==2.7.7
frozenlist==1.4.1
idna==3.6
ipython==8.21.0
ipywidgets==8.1.1
jedi==0.19.1
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
jupyterlab-widgets==3.0.9
kiwisolver==1.4.5
matplotlib==3.8.2
matplotlib-inline==0.1.6
more-itertools==10.2.0
multidict==6.0.5
numpy==1.26.3
packaging==23.2
parso==0.8.3
pillow==10.2.0
platformdirs==4.2.0
plumbum==1.8.2
pooch==1.8.0
prompt-toolkit==3.0.43
psutil==5.9.8
pure-eval==0.2.2
pyaedt==0.7.10
pycparser==2.21
Pygments==2.17.2
pyparsing==3.1.1
PyQt5==5.15.10
PyQt5-Qt5==5.15.2
PyQt5-sip==12.13.0
PySide6==6.6.1
PySide6-Addons==6.6.1
PySide6-Essentials==6.6.1
python-dateutil==2.8.2
pyvista==0.43.2
pywin32==306
referencing==0.33.0
requests==2.31.0
rpds-py==0.17.1
rpyc==5.3.1
scooby==0.9.2
shiboken6==6.6.1
six==1.16.0
stack-data==0.6.3
traitlets==5.14.1
trame==3.5.2
trame-client==2.15.0
trame-server==2.16.0
trame-vtk==2.8.0
urllib3==2.2.0
vtk==9.3.0
wcwidth==0.2.13
widgetsnbextension==4.0.9
wslink==1.12.4
yarl==1.9.4

gmalinve, 2024-02-12:
Hi @Nanaaaaaaa19,

I'm sorry but I couldn't reproduce your issue. 
Please try the snippet below: 

`import pyaedt

app_q3d = pyaedt.Q3d(specified_version="2024.1")

setup1 = app_q3d.create_setup(setupname="Setup1")
setup1.props["SaveFields"] = True

sweep1 = app_q3d.create_frequency_sweep(setupname=setup1.name, units="GHz", freqstart=0.5, freqstop=1.5, sweepname="Sweep1")

app_q3d.release_desktop(False, False)`

Hope this helps,

Giulia

Nanaaaaaaa19, 2024-02-13:
Hi @gmalinve ! Yes it does work this way but then the distribution will be in linear step and not linear count but if you put: 
`sweep1 = app_q3d.create_frequency_sweep(setupname=setup1.name, units="GHz", freqstart=0.5, freqstop=1.5, freqstep=401, sweepname="Sweep1")` it creates the sweep with wrong values, since it works in HFSS solver and not in Q3D I think it can be a bug no ? Keep me posted maybe there's a mistake/something on my side I may have overlooked

gmalinve, 2024-02-14:
Hi @Nanaaaaaaa19,

Could you please try to create a frequency weep through the setup object? like that: 
`sweep1_1 = setup1.create_frequency_sweep(unit="GHz", freqstart=0.5, freqstop=1.5, sweepname="Sweep1_1")`
and specify here your sweep type.

Thanks

Giulia

Nanaaaaaaa19, 2024-02-15:
[image]
[image]
Here are the results of this code line

gmalinve, 2024-02-15:
@Nanaaaaaaa19 ok so it seems it works, right?

Nanaaaaaaa19, 2024-02-15:
`sweep1_2 = setup1.create_frequency_sweep(unit="GHz", freqstart=0.5, freqstop=1.5, sweep_type='Interpolating', sweepname="Sweep1_2")` However if i try it like that the count set automatically to 401 points so it works with this line but if you want to choose the `num_of_freq_points` parameter then a bug appear

gmalinve, 2024-02-15:
@Nanaaaaaaa19 what bug?

Nanaaaaaaa19, 2024-02-15:
Try the function with `num_of_freq_points' inside like that: `sweep1_2 = setup1.create_frequency_sweep(unit="GHz", freqstart=0.5, freqstop=1.5, num_of_freq_points=401, sweep_type='Interpolating', sweepname="Sweep1_2")
`

Nanaaaaaaa19, 2024-02-15:
Try the function with `num_of_freq_points' inside like that: `sweep1_2 = setup1.create_frequency_sweep(unit="GHz", freqstart=0.5, freqstop=1.5, num_of_freq_points=401, sweep_type='Interpolating', sweepname="Sweep1_2")`

Nanaaaaaaa19, 2024-02-15:
`sweep1_2 = setup1.create_frequency_sweep(unit="GHz", freqstart=0.5, freqstop=1.5, num_of_freq_points=401, sweep_type='Interpolating', sweepname="Sweep1_2")`

Nanaaaaaaa19, 2024-02-15:
[image]
[image]

Nanaaaaaaa19, 2024-02-15:
[image]
[image]

gmalinve, 2024-02-15:
@Nanaaaaaaa19 for the moment, if possible, use the default value of 401 when sweep_type is interpolating. Someone fill fix this.
Thanks!

Nanaaaaaaa19, 2024-02-15:
Yes thanks !

closed 2024-02-19 by gmalinve