ansys/pyaedt #4217: Set units on Data Table

giuseppe1692, 2024-02-12:
### Description of the feature

Hi to all. 
Is there a way to modify the units of a Data Table on Ansys EDT using PyAEDT?
Starting from my data table i want to modify the unit in GHz and the impedance in mOhm and after i want extract a .csv file. 
Thank you in advance.

gmalinve, 2024-02-12:
Hi @giuseppe1692,

For Data Table reports I'd suggest to have a look at this Maxwell2D example: https://aedt.docs.pyansys.com/version/stable/examples/03-Maxwell/Maxwell2D_DCConduction.html
We create a Data Table report and then we manipulate it as a SolutionData object by using the `get_solution_data()` method.
If you'd like to export it as a csv then you can do so by using: `export_data_to_csv()`.

Hope this helps!

Giulia

gmalinve, 2024-02-14:
Hi @giuseppe1692,

Does this answer your question?
If yes please do let me know and I'll proceed by closing the issue.

Thanks,

Giulia

giuseppe1692, 2024-02-14:
Hi @gmalinve,

Unfortunately I was unable to change the units in the Data Table report. 
Below I pasted the function that I made and works fine, the row of interest is: 

`myreport_table = aedt_app.post.create_report(expressions= lista_tracce, setup_sweep_name = "Tabella",domain="Sweep",plot_type="Data Table", plotname = "TABELLA_IMPEDENZA")`

I also add an img that describe the units I want to change. I would like to change GHz in MHz and GOhm in mOhm
[image]

`
def SECONDARY_function(str_prj_name, sweep_str, str_table):
    try:
        if (file_path != None):
            documento_da_leggere = openpyxl.load_workbook(file_path)
    except: 
        percorso = filedialog.askopenfilename()
        documento_da_leggere = openpyxl.load_workbook(percorso)
    foglio_spec = documento_da_leggere["Spec"]
    foglio_aghi = documento_da_leggere["Needle vs Z Target"]
    index_dut = 23
    index_start = 24
    index_progetto = 3
    numero_site = foglio_spec["C" + str(index_dut)].value
    start_site = foglio_spec["C" + str(index_start)].value
    nome_progetto_from_file = foglio_spec["C" + str(index_progetto)].value
    global lista_aghi_completa
    lista_aghi_completa = ""
    fine = False
    indice_start_needle = 4
    j = 0
    global index 
    index = j = 0
    pos_ago = 2
    global needle_name, needle_res
    needle_name = foglio_spec["C26"].value
    needle_res = foglio_aghi["B" + str(pos_ago)].value * 1000
    global net_needle
    lista_net_needle = []
    lista_pad_needle = []
    lista_res_needle = []
    lista_ind_needle = []
    fine = False
    while(fine != True):
        net_needle = foglio_aghi["A" + str(j+indice_start_needle)].value
        if (net_needle == "GND"):
            fine = True
        else:
            lista_net_needle.insert(j, foglio_aghi["A" + str(j+indice_start_needle)].value)
            lista_pad_needle.insert(j, foglio_aghi["B" + str(j+indice_start_needle)].value)
            lista_res_needle.insert(j, round((foglio_aghi["C" + str(j+indice_start_needle)].value)* (1000),2))
            lista_ind_needle.insert(j, round((foglio_aghi["D" + str(j+indice_start_needle)].value)*(1*10**9),2))
            j = j + 1
    index = j
    non_graphical = False
    new_thread = True
    desktop_version = "2023.2" 
    if (str_prj_name == "default"):
        nome_progetto = nome_progetto_from_file
    else:     
        nome_progetto = str_prj_name 
    if stato_checkbutton_samsung == True:
        nome_design = "PH_" + data_formattata
    elif stato_checkbutton_samsung == False:
        nome_design = "PH_MLO_"+ data_formattata
    desktop = pyaedt.launch_desktop(desktop_version, non_graphical, new_thread)
    aedt_app = pyaedt.Circuit(projectname=nome_progetto, designname=nome_design)
    aedt_app.modeler.schematic.schematic_units = "mil"
    setup_ac = aedt_app.create_setup("AnalisiAC")
    setup_ac.props["SweepDefinition"]["Data"] = sweep_str
    setup_tab = aedt_app.create_setup("Tabella")
    setup_tab.props["SweepDefinition"]["Data"] = str_table
    offset_y = 0
    offset_x = 0
    for site in range(start_site, numero_site):  #0 , 2
        lista_expression = []
        for i in range (len(lista_net_needle)):
            port = aedt_app.modeler.components.create_interface_port(name=str(site) + "_" + lista_net_needle[i], location=[offset_x, offset_y] )
            if(stato_checkbutton_samsung == False):
                inductor = aedt_app.modeler.schematic.create_inductor(value=str(lista_ind_needle[i]) + "n/" + str(lista_pad_needle[i]), location=[offset_x + 500 , offset_y])
                resistor = aedt_app.modeler.schematic.create_resistor(value=str(lista_res_needle[i])+ "m/"+ str(lista_pad_needle[i]), location=[offset_x + 1000, offset_y])
                pins_resistor = resistor.pins
                port.pins[0].connect_to_component(component_pin=inductor.pins[0], use_wire=True)
                inductor.pins[1].connect_to_component(component_pin=resistor.pins[1], use_wire=True)
            offset_y = offset_y - 100
        lista_tracce = aedt_app.get_traces_for_plot(get_self_terms= True, get_mutual_terms = False, category = "mag(Z")
    for k in range (len(lista_net_needle)):
        expression_new = ""
        for j in range (0, numero_site+1):
            expression_new = expression_new + "mag(Z(" + str(j) + "_" + lista_net_needle[k] + "," + str(j) + "_" + lista_net_needle[k] + "))" + "; "
        expression_nuova = expression_new[:-2]
        lista_expression = expression_nuova.split(';')
        myreport = aedt_app.post.create_report(expressions= lista_expression, setup_sweep_name = "AnalisiAC",domain="Sweep",plot_type="Rectangular Plot", plotname = lista_net_needle[k])
        myreport.edit_general_settings(background_color=(255, 255, 255),plot_color=(255, 255, 255),enable_y_stripes=True,field_width=4,precision=4,use_scientific_notation=False)
        myreport.edit_x_axis_scaling(linear_scaling=False, min_scale="0.001MHz", max_scale="1000MHz", minor_tick_divs=10, min_spacing="5", units= "MHz")
        myreport.edit_y_axis_scaling(linear_scaling=False, min_scale="1mOhm", max_scale="100000mOhm", minor_tick_divs=10, min_spacing="5", units= "mOhm")
        myreport.edit_y_axis(label = "Z impedance", font="Arial", font_size=9, italic=False, bold=False, color=(0, 0, 0))
        myreport.edit_x_axis(label = "Freq", font="Arial", font_size=9, italic=False, bold=False, color=(0, 0, 0))
    myreport_table = aedt_app.post.create_report(expressions= lista_tracce, setup_sweep_name = "Tabella",domain="Sweep",plot_type="Data Table", plotname = "TABELLA_IMPEDENZA")
    aedt_app.analyze_setup("AnalisiAC")
    aedt_app.analyze_setup("Tabella")
    aedt_app.release_desktop(close_desktop=False,  close_projects=False)

giuseppe1692, 2024-02-14:
Hi @gmalinve 

Unfortunately I was unable to change the axis units on the Data Table.
I need to modify the highlighted units from GHz to MHz and from ohm to mOhm 

You can see an example on the img below: 

[image]

The line of code that i use to create this type of report is on the below function: 

```python
`myreport_table = aedt_app.post.create_report(expressions= lista_tracce, setup_sweep_name = "Tabella",domain="Sweep",plot_type="Data Table", plotname = "TABELLA_IMPEDENZA")`
```

```python

def SECONDARY_function(str_prj_name, sweep_str, str_table):
    try:
        if (file_path != None):
            documento_da_leggere = openpyxl.load_workbook(file_path)
    except: 
        percorso = filedialog.askopenfilename()
        documento_da_leggere = openpyxl.load_workbook(percorso)
    foglio_spec = documento_da_leggere["Spec"]
    foglio_aghi = documento_da_leggere["Needle vs Z Target"]
    index_dut = 23
    index_start = 24
    index_progetto = 3
    numero_site = foglio_spec["C" + str(index_dut)].value
    start_site = foglio_spec["C" + str(index_start)].value
    nome_progetto_from_file = foglio_spec["C" + str(index_progetto)].value
    global lista_aghi_completa
    lista_aghi_completa = ""
    fine = False
    indice_start_needle = 4
    j = 0
    global index 
    index = j = 0
    pos_ago = 2
    global needle_name, needle_res
    needle_name = foglio_spec["C26"].value
    needle_res = foglio_aghi["B" + str(pos_ago)].value * 1000
    global net_needle
    lista_net_needle = []
    lista_pad_needle = []
    lista_res_needle = []
    lista_ind_needle = []
    fine = False
    while(fine != True):
        net_needle = foglio_aghi["A" + str(j+indice_start_needle)].value
        if (net_needle == "GND"):
            fine = True
        else:
            lista_net_needle.insert(j, foglio_aghi["A" + str(j+indice_start_needle)].value)
            lista_pad_needle.insert(j, foglio_aghi["B" + str(j+indice_start_needle)].value)
            lista_res_needle.insert(j, round((foglio_aghi["C" + str(j+indice_start_needle)].value)* (1000),2))
            lista_ind_needle.insert(j, round((foglio_aghi["D" + str(j+indice_start_needle)].value)*(1*10**9),2))
            j = j + 1
    index = j
    non_graphical = False
    new_thread = True
    desktop_version = "2023.2" 
    if (str_prj_name == "default"):
        nome_progetto = nome_progetto_from_file
    else:     
        nome_progetto = str_prj_name 
    if stato_checkbutton_samsung == True:
        nome_design = "PH_" + data_formattata
    elif stato_checkbutton_samsung == False:
        nome_design = "PH_MLO_"+ data_formattata
    desktop = pyaedt.launch_desktop(desktop_version, non_graphical, new_thread)
    aedt_app = pyaedt.Circuit(projectname=nome_progetto, designname=nome_design)
    aedt_app.modeler.schematic.schematic_units = "mil"
    setup_ac = aedt_app.create_setup("AnalisiAC")
    setup_ac.props["SweepDefinition"]["Data"] = sweep_str
    setup_tab = aedt_app.create_setup("Tabella")
    setup_tab.props["SweepDefinition"]["Data"] = str_table
    offset_y = 0
    offset_x = 0
    for site in range(start_site, numero_site):  #0 , 2
        lista_expression = []
        for i in range (len(lista_net_needle)):
            port = aedt_app.modeler.components.create_interface_port(name=str(site) + "_" + lista_net_needle[i], location=[offset_x, offset_y] )
            if(stato_checkbutton_samsung == False):
                inductor = aedt_app.modeler.schematic.create_inductor(value=str(lista_ind_needle[i]) + "n/" + str(lista_pad_needle[i]), location=[offset_x + 500 , offset_y])
                resistor = aedt_app.modeler.schematic.create_resistor(value=str(lista_res_needle[i])+ "m/"+ str(lista_pad_needle[i]), location=[offset_x + 1000, offset_y])
                pins_resistor = resistor.pins
                port.pins[0].connect_to_component(component_pin=inductor.pins[0], use_wire=True)
                inductor.pins[1].connect_to_component(component_pin=resistor.pins[1], use_wire=True)
            offset_y = offset_y - 100
        lista_tracce = aedt_app.get_traces_for_plot(get_self_terms= True, get_mutual_terms = False, category = "mag(Z")
    for k in range (len(lista_net_needle)):
        expression_new = ""
        for j in range (0, numero_site+1):
            expression_new = expression_new + "mag(Z(" + str(j) + "_" + lista_net_needle[k] + "," + str(j) + "_" + lista_net_needle[k] + "))" + "; "
        expression_nuova = expression_new[:-2]
        lista_expression = expression_nuova.split(';')
        myreport = aedt_app.post.create_report(expressions= lista_expression, setup_sweep_name = "AnalisiAC",domain="Sweep",plot_type="Rectangular Plot", plotname = lista_net_needle[k])
        myreport.edit_general_settings(background_color=(255, 255, 255),plot_color=(255, 255, 255),enable_y_stripes=True,field_width=4,precision=4,use_scientific_notation=False)
        myreport.edit_x_axis_scaling(linear_scaling=False, min_scale="0.001MHz", max_scale="1000MHz", minor_tick_divs=10, min_spacing="5", units= "MHz")
        myreport.edit_y_axis_scaling(linear_scaling=False, min_scale="1mOhm", max_scale="100000mOhm", minor_tick_divs=10, min_spacing="5", units= "mOhm")
        myreport.edit_y_axis(label = "Z impedance", font="Arial", font_size=9, italic=False, bold=False, color=(0, 0, 0))
        myreport.edit_x_axis(label = "Freq", font="Arial", font_size=9, italic=False, bold=False, color=(0, 0, 0))
    myreport_table = aedt_app.post.create_report(expressions= lista_tracce, setup_sweep_name = "Tabella",domain="Sweep",plot_type="Data Table", plotname = "TABELLA_IMPEDENZA")
    aedt_app.analyze_setup("AnalisiAC")
    aedt_app.analyze_setup("Tabella")
    aedt_app.release_desktop(close_desktop=False,  close_projects=False)

```

gmalinve, 2024-02-15:
Hi @giuseppe1692,

There's no way to do it in PyAEDT so I'd suggest you to do it through the object oriented language.
You can access your report object by doing: 
`report = app.get_oo_object(app.odesign, "Results\\{}".format(data_table_name))`
Then you can access the primary sweep value (in your case Freq.):
`report.GetChildObject('PrimarySweepDrawing').GetPropValue("Units")
report.GetChildObject('PrimarySweepDrawing').SetPropValue("Units", "kHz")`
and change each unit curve by curve: 
`report.GetChildObject('C(Bar1,Bar2)').GetChildObject("Curve1").GetPropValue("Units")
report.GetChildObject('C(Bar1,Bar2)').GetChildObject("Curve1").SetPropValue("Units", "mOhm")`

Hope this helps!

Giulia

gmalinve, 2024-02-16:
@giuseppe1692 if this reply to your question please do let me know.

Thanks,

Giulia

closed 2024-02-16 by gmalinve

giuseppe1692, 2024-02-21:
Hi @gmalinve, 
sorry for the delay in replying.
I tried to use your code but maybe I make some mistakes in using the variables in the right place.
For now, I used only the line that are useful to modify the frequency, but they don't work and I get errors. 
I wrote the code in this way:

```python
myreport_table = aedt_app.post.create_report(expressions= lista_tracce, setup_sweep_name = "Tabella",domain="Sweep",plot_type="Data Table", plotname = "TABELLA_IMPEDENZA")
    myreport_table = aedt_app.get_oo_object(aedt_app.odesign, "Results\\{}".format("TABELLA IMPEDENZA"))
    myreport_table.GetChildObject('Freq').GetPropValue("Units") 
    myreport_table.GetChildObject('Freq').SetPropValue("Units", "kHz")
``` 

Could you help me? Sorry to bother you

Thank you so much!

giuseppe1692, 2024-02-21:
Hi @gmalinve,
I solved it by trying to change some things in my code (attached below).
Thank you so much for your support. 
I will definitely bother you with other issues :-)

```python
myreport_table = aedt_app.post.create_report(expressions= lista_tracce, setup_sweep_name = "Tabella",domain="Sweep",plot_type="Data Table", plotname = "TABELLA_IMPEDENZA")
    myreport_table = aedt_app.get_oo_object(aedt_app.odesign, "Results\\{}".format("TABELLA_IMPEDENZA"))
    myreport_table.GetChildObject('PrimarySweepDrawing').GetPropValue("Units")
    myreport_table.GetChildObject('PrimarySweepDrawing').SetPropValue("Units", "MHz")
    for k in range(len(lista_tracce)):
        myreport_table.GetChildObject(lista_tracce[k]).GetChildObject("Curve1").GetPropValue("Units") 
        myreport_table.GetChildObject(lista_tracce[k]).GetChildObject("Curve1").SetPropValue("Units", "mOhm")
```