ansys/pyaedt #4222: Warnings in unit tests

kbhagat-ansys, 2024-02-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

All runs of unit tests show warnings. When these are run in an ADO pipeline they cause a lot of noise. 
e.g. [go to any successful run of the unit_tests.yml Run](https://github.com/ansys/pyaedt/actions/workflows/unit_tests.yml) and then expand the "Unit testing" task and see the TFS warnings. 

[image]

### Steps To Reproduce

[Go to any successful run of the unit_tests.yml Run](https://github.com/ansys/pyaedt/actions/workflows/unit_tests.yml) and then expand the "Unit testing" task and see the TFS warnings. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

N/A

Samuelopez-ansys, 2024-02-14:
@SMoraisAnsys I found this warning also some months ago. Do you have any idea how could he hide it?

SMoraisAnsys, 2024-02-14:
> @SMoraisAnsys I found this warning also some months ago. Do you have any idea how could he hide it?

Dunno where that comes from yet. I'll look into it next week if it's fine with you.

kbhagat-ansys, 2024-02-14:
> > @SMoraisAnsys I found this warning also some months ago. Do you have any idea how could he hide it?
> 
> Dunno where that comes from yet. I'll look into it next week if it's fine with you.

@SMoraisAnsys  I don't think we should hide it. We should work on making the code fix to avoid seeing it. Note these are only shown when it is run through an ADO/Github environment and not when its run via PyCharm. 
A subset of the warnings can be reproduced by running `test_01_GeometryOperators.py` unit test file. I will see if I can narrow it down to 1-2 tests so we don't have to run all the unit tests to triage.

cc @Samuelopez-ansys

kbhagat-ansys, 2024-02-15:
> @SMoraisAnsys I don't think we should hide it. We should work on making the code fix to avoid seeing it. Note these are only shown when it is run through an ADO/Github environment and not when its [...]

We can reproduce many warnings by running `test_unit_converter` inside `_unittest\test_01_GeometryOperators.py`
Thus most of these are coming from the usage of `warnings.warn()`

SMoraisAnsys, 2024-02-16:
@kbhagat-ansys Thanks for pointing that out !

SMoraisAnsys, 2024-02-24:
@kbhagat-ansys Sry for not being able to work on that yet, this week has been heavily loaded. I should be able to find time to work on this issue next week. Ping me if you don't get any feedback !

PR #4297 (2024-03-01): TESTS: Refact warnings filtering
Remove pytest.ini file to put filterwarnings in pyproject.toml file.
Add UserWarning to avoid unwanted warnings when testing.

Closes #4222
closed 2024-03-01 by Samuelopez-ansys