ansys/pyaedt #4227: Bug located in Maxwell3D assign_impedance: Impedance is assigned to a *body*, but it should be assigned to a *face*

IreneWoyna, 2024-02-13:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Impedance BC should be assigned to faces of an object, but it is assigned to bodies.

### Steps To Reproduce

This doesn't work, although it's the correct usage of the function:

shield_faces = m3d.modeler.select_allfaces_fromobjects(["Shield_1st", "Shield_2nd"])
m3d.assign_impedance(
    geometry_selection=[shield_faces ],
    material_name="aluminum",
    impedance_name="ShieldImpedance",
)

This works, but is incorrect:

m3d.assign_impedance(
    geometry_selection=["Shield_1st", "Shield_2nd"],
    material_name="aluminum",
    impedance_name="ShieldImpedance",
)

An idea how to fix the issue: The function should select_allfaces_fromobjects from the objects given as an input and assign the impedance BC to the faces.

This is an example IronPython code which gives the correct output: 

oModule.AssignImpedance(
	[
		"NAME:Impedance1",
		"Faces:="		, [151,152,96,99,153,97,95,154,126,155,156,157,158,128,246,247,190,192,248,194,191,249,221,250,251,252,253,223],
		"UseMaterial:="		, True,
		"MaterialName:="	, "aluminum",
		"IsPermeabilityNonlinear:=", False,
		"Conductivity:="	, "38000000"
	])

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

anyio==3.6.2
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.3
asttokens==2.2.1
async-lru==2.0.4
attrs==23.1.0
Babel==2.13.1
backcall==0.2.0
beautifulsoup4==4.12.2
bleach==6.0.0
certifi==2022.5.18.1
cffi==1.15.1
charset-normalizer==2.0.12
colorama==0.4.6
comm==0.1.3
debugpy==1.6.7
decorator==5.1.1
defusedxml==0.7.1
docutils==0.20.1
executing==1.2.0
fastjsonschema==2.16.3
fqdn==1.5.1
idna==3.3
importlib-metadata==6.6.0
importlib-resources==5.12.0
ipykernel==6.23.0
ipython==8.12.2
ipython-genutils==0.2.0
isoduration==20.11.0
jedi==0.18.2
Jinja2==3.1.2
joblib==1.3.1
json5==0.9.14
jsonpointer==2.3
jsonschema==4.19.2
jsonschema-specifications==2023.7.1
jupyter-events==0.6.3
jupyter-lsp==2.2.0
jupyter_client==8.2.0
jupyter_core==5.3.0
jupyter_server==2.5.0
jupyter_server_terminals==0.4.4
jupyterlab==4.0.8
jupyterlab-pygments==0.2.2
jupyterlab_server==2.25.1
jupyterlab_theme_solarized_dark==3.0.0
jupytext==1.15.2
markdown-it-py==3.0.0
MarkupSafe==2.1.2
matplotlib-inline==0.1.6
mdit-py-plugins==0.4.0
mdurl==0.1.2
mistune==2.0.5
nbclassic==1.0.0
nbclient==0.7.4
nbconvert==7.4.0
nbformat==5.8.0
nest-asyncio==1.5.6
notebook==6.5.4
notebook_shim==0.2.3
numpy==1.22.4
p2j==1.3.2
packaging==23.1
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
pkgutil_resolve_name==1.3.10
platformdirs==3.5.1
prometheus-client==0.16.0
prompt-toolkit==3.0.38
psutil==5.9.1
pure-eval==0.2.2
pycparser==2.21
Pygments==2.15.1
# Editable install with no version control (PyMotorCAD==0.0.1)
-e c:\ansys_motor-cad\15_1_3\python\rpc
pyrsistent==0.19.3
python-dateutil==2.8.2
python-json-logger==2.0.7
pytz==2023.3.post1
pywin32==304
pywinpty==2.0.10
PyYAML==6.0
pyzmq==25.0.2
readme-renderer==36.0
referencing==0.30.2
requests==2.31.0
restview==3.0.1
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.12.0
scikit-learn==1.3.0
scipy==1.8.1
Send2Trash==1.8.2
six==1.16.0
sniffio==1.3.0
soupsieve==2.4.1
stack-data==0.6.2
terminado==0.17.1
threadpoolctl==3.1.0
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tornado==6.3.1
traitlets==5.9.0
typing_extensions==4.5.0
uri-template==1.2.0
urllib3==1.26.9
wcwidth==0.2.6
webcolors==1.13
webencodings==0.5.1
websocket-client==1.5.1
zipp==3.15.0

Samuelopez-ansys, 2024-02-14:
@IreneWoyna I think the method just needs a refactoring like the assign_insulating method which allows both Bodies and Faces. We could test if it works similarly to assign_insulating.

https://github.com/ansys/pyaedt/blob/main/pyaedt/maxwell.py#L2118

Do you want me to take a look? Or do you want to try?

IreneWoyna, 2024-02-14:
Thanks for the hint @Samuelopez-ansys, I'll try myself :)

closed 2024-02-14 by Samuelopez-ansys