ansys/pyaedt #4240: (AO) Import Layer Stackup XML

aohta2021, 2024-02-16:
### Description of the feature

HFSS3D Layout GUI can import Layer Stackup XML.
It has material definition and layer construction.
Pintopinsetup command can support limited import Layer Stackup XML.
It could not support Thermal material and Frequency depended Dielectric material definition.
User want to import Layer Stackup XML without GUI.

aohta2021, 2024-03-19:
It is related TFS#1006882:(AO) Seting Layer Stackup is not correct by pintopinsetup.exe

Samuelopez-ansys, 2024-03-19:
@svandenb-dev @ring630 Is this possible with PyEBD right?

ring630, 2024-04-05:
@Samuelopez-ansys it is supported by PyEDB. @aohta2021, could you provide an example xml file I can test?

aohta2021, 2024-11-22:
I have confirmed PyEDB can support it.
This issue is closed.

closed 2024-11-22 by aohta2021