ansys/pyaedt #4248: Add the ability to assign boundaries to a specific face of an object

OscarCapac, 2024-02-19:
### Description of the feature

Add the ability to assign boundaries to a specific face or a collection of faces instead of to an entire object.

### Useful links and references

This is how I used to do this in IronPython for a single face : 

```
oBoundary.AssignAnisotropicImpedance(
     [
         "NAME:{}".format(boundary_name),
         "Faces:="       , [int(face_id)],
         "UseInfiniteGroundPlane:=", False,
         "CoordSystem:="     , "Global",
         "HasExternalLink:=" , False,
         "ZxxResistance:="   , str(R_xx),
         "ZxxReactance:="    , str(X_xx),
         "ZxyResistance:="   , str(R_xy),
         "ZxyReactance:="    , str(X_xy),
         "ZyxResistance:="   , str(R_yx),
         "ZyxReactance:="    , str(X_yx),
         "ZyyResistance:="   , str(R_yy),
         "ZyyReactance:="    , str(X_yy)
    ])
```

gmalinve, 2024-02-20:
Hi @OscarCapac,

Thanks for the feedback!
This enhancement has recently been implemented in this PR: #4229 and be available in the next PyAEDT release.
@IreneWoyna 

Giulia

closed 2024-02-20 by gmalinve

OscarCapac, 2024-02-20:
Hi @gmalinve,

Thank for the response. I looked at #4229 and it seems to only affect the Maxwell application, while I need it for HFSS.
Also it seems there is no support at the time for anisotropic impedance?

I managed to code what i needed in the HFSS class.
Maybe I need to do a pull request when I have the time?

gmalinve, 2024-02-20:
Hi @OscarCapac,

sure you can create a PR! PyAEDT is an open repository where everyone can contribute!

Thanks,

Giulia